raspberrypi/libcamera #114: RPI4B OV5647 Camera black screen!

ahmad4011, 2024-02-23:
Hi i recenlty bought a camera for my RPI4B but for some reason it's not working, whenever i enter libcamera-hello or rpicam-vid i get this error 

```
    new@raspberrypi:~/rpicam-apps $ rpicam-vid
    [0:13:59.700932659] [3073]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
    [0:13:59.719660797] [3074]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
    [0:13:59.722425283] [3074]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media4
    [0:13:59.722557469] [3074]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
    Preview window unavailable
    Mode selection for 640:480:12:P
        SGBRG10_CSI2P,640x480/0 - Score: 1000
        SGBRG10_CSI2P,1296x972/0 - Score: 1287
        SGBRG10_CSI2P,1920x1080/0 - Score: 1636.67
        SGBRG10_CSI2P,2592x1944/0 - Score: 1854
    Stream configuration adjusted
    [0:13:59.724585227] [3073]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 640x480-SGBRG10_CSI2P
    [0:13:59.725015990] [3074]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
    [0:14:00.843386065] [3074]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[12:cap]: Dequeue timer of 1000000.00us has expired!
    [0:14:00.843549492] [3074] ERROR RPI pipeline_base.cpp:1376 Camera frontend has timed out!
    [0:14:00.843592566] [3074] ERROR RPI pipeline_base.cpp:1377 Please check that your camera sensor connector is attached securely.
    [0:14:00.843636141] [3074] ERROR RPI pipeline_base.cpp:1378 Alternatively, try another cable and/or sensor.
    ERROR: Device timeout detected, attempting a restart!!!
```

please i need help less than 3 weeks i have to finish a capstone project

note that i tried different other commands, i have followed the complete guide on this [https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-rpicam-apps][1] website and downloaded everything. still same error 

heres my os 

```

    PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
    NAME="Debian GNU/Linux"
    VERSION_ID="11"
    VERSION="11 (bullseye)"
    VERSION_CODENAME=bullseye
    ID=debian
    HOME_URL="https://www.debian.org/"
    SUPPORT_URL="https://www.debian.org/support"
    BUG_REPORT_URL="https://bugs.debian.org/"
```

  [1]: https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-rpicam-apps

naushir, 2024-02-26:
> [0:14:00.843549492] [3074] ERROR RPI pipeline_base.cpp:1376 Camera frontend has timed out!
>     [0:14:00.843592566] [3074] ERROR RPI pipeline_base.cpp:1377 Please check that your camera sensor [...]

This may be a camera hardware related issue.  Have you tested this camera on any other device?

xax111a, 2024-02-26:
> This may be a camera hardware related issue. Have you tested this camera on any other device?

i hope, honestly i tried everything. different os's. different raspberrys. im working on my capstone project. and i have to finish my project within 2 weeks

xax111a, 2024-02-26:
> > Can you please post the output of `uname -a` command?
> 
> ![image](https://private-user-images.githubusercontent.com/125976429/307899152-93d7566b-f4a0-4112-9471-e1e5c51d791a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5NzA2NTAsIm5iZiI6MTcwODk3MDM1MCwicGF0aCI6Ii8xMjU5NzY0MjkvMzA3ODk5MTUyLTkzZDc1NjZiLWY0YTAtNDExMi05NDcxLWUxZTVjNTFkNzkxYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNlQxNzU5MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZDRjN2M2YTk1ZTI2ZWFkNzUzMGRkOTZlM2I1NGVkOTNlNTA4MjQ1YTUyZGYxMDE0YzU0OWE5MGJmNWU1NDJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ywmShb21jD-Z85Q4rzXjsJEPYoK6GkK-3RFnJIyGFao)

I know that is an old. But i tried different os’s old and new different raspberries. Still same error

naushir, 2024-02-27:
Please try again with a fully updated Bullseye OS image.  If you still get the problem, the only thing I can suggest is to return the camera from your supplier and get a new one.

naushir, 2024-03-06:
Closing due to inactivity.

closed 2024-03-06 by naushir