raspberrypi/libcamera #116: Can't start gstreamer pipeline after update

svettuff, 2024-03-01:
I got this after last update:

`dev@raspberrypi:~ $ gst-launch-1.0 libcamerasrc ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
[0:23:51.222922052] [43358]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:23:51.247154371] [43363]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 29-02-2024 (23:05:29)
[0:23:51.294768979] [43363]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[0:23:51.296230876] [43366] ERROR Stream stream.cpp:215 Range format is ambiguous
[0:23:51.296300263] [43366] ERROR Stream stream.cpp:215 Range format is ambiguous
[0:23:51.296315244] [43366] ERROR Stream stream.cpp:215 Range format is ambiguous
[0:23:51.296327688] [43366] ERROR Stream stream.cpp:215 Range format is ambiguous
[0:23:51.372244299] [43366] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Failed to configure camera: Invalid argument
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(446): gst_libcamera_src_negotiate (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
Camera::configure() failed with error code -22
Execution ended after 0:00:00.077091444
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Internal data stream error.
Setting pipeline to NULL ...
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(645): gst_libcamera_src_task_enter (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
streaming stopped, reason not-negotiated (-4)
Freeing pipeline ...
dev@raspberrypi:~ $`

What should I do? Thank you!

svettuff, 2024-03-01:
Can I get somehow previous version of gstreamer1.0-libcamera 0.2.0 but not 0.0.3?

naushir, 2024-03-01:
I'm afraid I don't know anything about the gstreamer plugin.  You can try an earlier build of libcamera with Pi 5 support from the release package here: https://github.com/raspberrypi/libcamera/releases/tag/v0.1.0%2Brpt20231122

svettuff, 2024-03-01:
@naushir It works! Thank you very much)

naushir, 2024-03-01:
Great, resolving this now.

closed 2024-03-01 by naushir