raspberrypi/libcamera #122: False colours from global shutter camera on RPi5

snaggas, 2024-03-21:
With Bookworm freshly installed on a Raspberry Pi 5, the global shutter camera (IMX296) gives false colours. The attached picture of a colour wheel was created with `libcamera-still -o test.jpg` and shows that red and blue are exchanged, which suggests an issue with colour encoding, e.g. RGB exchanged for BGR. Surprisingly, I have not seen this issue reported so perhaps it's caused by a recent update. Does anyone else see this behaviour? Thanks in advance!
[image]

6by9, 2024-03-21:
Should be already resolved via https://github.com/raspberrypi/linux/pull/6027. Use `sudo rpi-update` to get the latest kernel to confirm (normal warnings about not doing this on a critical Pi, or taking backups first).

snaggas, 2024-03-21:
Yes, that resolves it. Thanks. Sorry if that was obvious - it didn't turn up in my searches.

closed 2024-03-21 by naushir