raspberrypi/libcamera #126: RPI5 - Omnivision 9281 8 Bit Mode

automaton123456, 2024-04-02:
Hi,

I have been getting issues when trying to get the output of the camera working in 8 bit mode at 640:400 resolution, both through Python and the command line. Everything works fine with 10bit, but as soon as I try rpicam-raw 640:400:8:P or set the bit_depth as 8bit in Python I start getting Broken Pipe errors. I get no other errors in my logs

Here are my listed modes

rpicam-hello --list-cameras
Available cameras
-----------------
0 : ov9281 [1280x800 10-bit MONO] (/base/axi/pcie@120000/rp1/i2c@88000/ov9281@60)
    Modes: 'R8' : 640x400 [309.79 fps - (0, 0)/1280x800 crop]
                  1280x720 [171.79 fps - (0, 0)/1280x720 crop]
                  1280x800 [143.66 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [247.83 fps - (0, 0)/1280x800 crop]
                         1280x720 [137.42 fps - (0, 0)/1280x720 crop]
                         1280x800 [114.93 fps - (0, 0)/1280x800 crop]

Am I miss-understanding how this all works. Does setting bit depth to 8 correspond to the R8 mode? In this example what mode is the app actually selecting?  R10_CSI2P,640x400/0?

rpicam-hello --camera 0 --mode 640:480:8:P
[0:07:13.057859596] [2369]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:07:13.074079445] [2370]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 04-03-2024 (18:25:28)
[0:07:13.075956320] [2370]  WARN CameraSensorProperties camera_sensor_properties.cpp:274 No static properties available for 'ov9281'
[0:07:13.076385548] [2370]  WARN CameraSensorProperties camera_sensor_properties.cpp:276 Please consider updating the camera sensor properties database
[0:07:13.080948308] [2370]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 640:400:12:P
    R8,640x400/0 - Score: 2000
    R8,1280x720/0 - Score: 2306.67
    R8,1280x800/0 - Score: 2260
    R10_CSI2P,640x400/0 - Score: 1000
    R10_CSI2P,1280x720/0 - Score: 1306.67
    R10_CSI2P,1280x800/0 - Score: 1260
Stream configuration adjusted
[0:07:13.372661787] [2369]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x400-YUV420 (1) 640x400-MONO_PISP_COMP1
[0:07:13.372806308] [2370]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 - Selected sensor format: 640x400-Y10_1X10 - Selected CFE format: 640x400-PC1M
[0:07:13.373938174] [2374]  WARN IPARPI ipa_base.cpp:1055 Could not set SHARPNESS - no sharpen algorithm

automaton123456, 2024-04-20:
Thanks for the response,

I have tried a few different combos but still the same result. With your command I get the following output from the terminal

rpicam-hello --camera 0 --viewfinder-mode 640:400:8:U
[0:14:13.312375547] [3529]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+47-46b868fe
[0:14:13.323087932] [3530]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 04-03-2024 (18:25:28)
[0:14:13.324124742] [3530]  WARN CameraSensorProperties camera_sensor_properties.cpp:274 No static properties available for 'ov9281'
[0:14:13.324139779] [3530]  WARN CameraSensorProperties camera_sensor_properties.cpp:276 Please consider updating the camera sensor properties database
[0:14:13.331387992] [3530]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:14:13.331452844] [3530]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 04-03-2024 (18:25:28)
[0:14:13.332136397] [3530]  WARN CameraSensorProperties camera_sensor_properties.cpp:274 No static properties available for 'ov9281'
[0:14:13.332149360] [3530]  WARN CameraSensorProperties camera_sensor_properties.cpp:276 Please consider updating the camera sensor properties database
[0:14:13.333825038] [3530]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/ov9281@60 to CFE device /dev/media1 and ISP device /dev/media3 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 640:400:8:U
    R8,640x400/0 - Score: 0
    R8,1280x720/0 - Score: 306.667
    R8,1280x800/0 - Score: 260
    R10_CSI2P,640x400/0 - Score: 1000
    R10_CSI2P,1280x720/0 - Score: 1306.67
    R10_CSI2P,1280x800/0 - Score: 1260
Stream configuration adjusted
[0:14:13.693232908] [3529]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x400-YUV420 (1) 640x400-R16
[0:14:13.693370093] [3530]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 - Selected sensor format: 640x400-Y8_1X8 - Selected CFE format: 640x400-Y16 
[0:14:13.694656199] [3535]  WARN IPARPI ipa_base.cpp:1055 Could not set SHARPNESS - no sharpen algorithm
[0:14:13.701096249] [3530] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video12[16:cap]: Failed to start streaming: Broken pipe
ERROR: *** failed to start camera ***

I have a bad feeling I am going to have to play with kernel driver for the camera

I noticed that the latest builds of the kernel are using a different driver to the past
/linux/blob/rpi-6.6.y/drivers/media/i2c/ov9282.c

6by9, 2024-04-22:
The kernel is throwing an error
```
[   22.006338] rp1-cfe 1f00110000.csi: Failed to start media pipeline: -32
```
Extra debug
```
rp1-cfe 1f00110000.csi: Link 'csi2':4 -> 'pisp-fe':0 failed validation: -32
```
media-ctl after this error:
```
pi@bookworm64:~/v4l-utils $ ./build/utils/media-ctl/media-ctl -p -d 3 
Media controller API version 6.6.28

Media device information
------------------------
driver          rp1-cfe
model           rp1-cfe
serial          
bus info        platform:1f00110000.csi
hw revision     0x114666
driver version  6.6.28

Device topology
- entity 1: csi2 (8 pads, 8 links, 0 routes)
            type V4L2 subdev subtype Unknown flags 0
            device node name /dev/v4l-subdev0
	pad0: SINK
		[stream:0 fmt:Y8_1X8/640x400 field:none colorspace:raw]
		<- "ov9281 6-0060":0 [ENABLED,IMMUTABLE]
	pad1: SINK
		[stream:0 fmt:unknown/16384x1 field:none]
	pad2: SINK
		[stream:0 fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
	pad3: SINK
		[stream:0 fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
	pad4: SOURCE
		[stream:0 fmt:Y8_1X8/640x400 field:none colorspace:raw]
		-> "rp1-cfe-csi2_ch0":0 []
		-> "pisp-fe":0 [ENABLED]
	pad5: SOURCE
		[stream:0 fmt:unknown/16384x1 field:none]
		-> "rp1-cfe-embedded":0 []
	pad6: SOURCE
		[stream:0 fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
		-> "rp1-cfe-csi2_ch2":0 []
		-> "pisp-fe":0 []
	pad7: SOURCE
		[stream:0 fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
		-> "rp1-cfe-csi2_ch3":0 []
		-> "pisp-fe":0 []

- entity 10: pisp-fe (5 pads, 7 links, 0 routes)
             type V4L2 subdev subtype Unknown flags 0
             device node name /dev/v4l-subdev1
	pad0: SINK,MUST_CONNECT
		[stream:0 fmt:Y16_1X16/640x400 field:none colorspace:raw]
		<- "csi2":4 [ENABLED]
		<- "csi2":6 []
		<- "csi2":7 []
	pad1: SINK
		[stream:0 fmt:FIXED/16384x1 field:none]
		<- "rp1-cfe-fe_config":0 [ENABLED]
	pad2: SOURCE
		[stream:0 fmt:Y16_1X16/640x400 field:none colorspace:raw]
		-> "rp1-cfe-fe_image0":0 [ENABLED]
	pad3: SOURCE
		[stream:0 fmt:Y16_1X16/640x400 field:none colorspace:raw]
		-> "rp1-cfe-fe_image1":0 []
	pad4: SOURCE
		[stream:0 fmt:FIXED/16384x1 field:none]
		-> "rp1-cfe-fe_stats":0 [ENABLED]

- entity 16: ov9281 6-0060 (1 pad, 1 link, 0 routes)
             type V4L2 subdev subtype Sensor flags 0
             device node name /dev/v4l-subdev2
	pad0: SOURCE
		[stream:0 fmt:Y8_1X8/640x400 field:none colorspace:raw xfer:none
		 crop.bounds:(8,8)/1280x800
		 crop:(8,8)/1280x800]
		-> "csi2":0 [ENABLED,IMMUTABLE]

- entity 18: rp1-cfe-csi2_ch0 (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video0
	pad0: SINK,MUST_CONNECT
		<- "csi2":4 []

- entity 22: rp1-cfe-embedded (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video1
	pad0: SINK,MUST_CONNECT
		<- "csi2":5 []

- entity 26: rp1-cfe-csi2_ch2 (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video2
	pad0: SINK,MUST_CONNECT
		<- "csi2":6 []

- entity 30: rp1-cfe-csi2_ch3 (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video3
	pad0: SINK,MUST_CONNECT
		<- "csi2":7 []

- entity 34: rp1-cfe-fe_image0 (1 pad, 1 link)
             type Node subtype V4L flags 1
             device node name /dev/video4
	pad0: SINK,MUST_CONNECT
		<- "pisp-fe":2 [ENABLED]

- entity 38: rp1-cfe-fe_image1 (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video5
	pad0: SINK,MUST_CONNECT
		<- "pisp-fe":3 []

- entity 42: rp1-cfe-fe_stats (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video6
	pad0: SINK,MUST_CONNECT
		<- "pisp-fe":4 [ENABLED]

- entity 46: rp1-cfe-fe_config (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video7
	pad0: SOURCE,MUST_CONNECT
		-> "pisp-fe":1 [ENABLED]
```
Note that you need to build the latest media-ctl. The standard Raspberry Pi OS version doesn't know `MEDIA_BUS_FMT_Y16_1X16`

csi2:4 would appear to be the mismatch as it reads
```
	pad4: SOURCE
		[stream:0 fmt:Y8_1X8/640x400 field:none colorspace:raw]
		-> "rp1-cfe-csi2_ch0":0 []
		-> "pisp-fe":0 [ENABLED]
```
when for all other modes it will read `Y16_1X16/640x400`. Indeed trying to set it to `Y8_1X8` it "corrects" the format to `Y16_1X16`.

It looks like there is a missing `remap` entry at https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/media/platform/raspberrypi/rp1_cfe/cfe_fmts.h#L258, meaning we can't remap the Y8 to Y16 to feed it into the FE.
```
diff --git a/drivers/media/platform/raspberrypi/rp1_cfe/cfe_fmts.h b/drivers/media/platform/raspberrypi/rp1_cfe/cfe_fmts.h
index 72516c93c5c5..ec68b4326c90 100644
--- a/drivers/media/platform/raspberrypi/rp1_cfe/cfe_fmts.h
+++ b/drivers/media/platform/raspberrypi/rp1_cfe/cfe_fmts.h
@@ -256,6 +256,7 @@ static const struct cfe_fmt formats[] = {
                .code = MEDIA_BUS_FMT_Y8_1X8,
                .depth = 8,
                .csi_dt = MIPI_CSI2_DT_RAW8,
+               .remap = { V4L2_PIX_FMT_Y16, V4L2_PIX_FMT_PISP_COMP1_MONO },
        },
        {
                .fourcc = V4L2_PIX_FMT_Y10P,
```
gets it working for me, but I'll leave it up to @naushir to confirm the change and create the PR.

naushir, 2024-04-23:
PR is available at https://github.com/raspberrypi/linux/pull/6122

automaton123456, 2024-04-23:
Cheers Guys,

Pulled changes, rebuilt kernel and my tests are working

closed 2024-04-23 by automaton123456

6by9, 2024-04-24:
FYI we have CI running on all kernel PRs, so you can use `sudo rpi-update pulls/6122` (substitute the appropriate PR number in there) to pull the relevant CI build instead of having to build the kernel yourself.

To answer your other comment over using ov9282, yes I upstreamed all my extra functionality from the downstream ov9281 driver into the mainline ov9282 driver that Intel had originated. We switched in 6.2 via #5304.
OV9281 and OV9282 are the same sensor but with different CRA (Chief Ray Angle) in the optics.