raspberrypi/libcamera #135: libcamerify not working rpi5 bookworm. Not enough buffers

gentoosys, 2024-05-09:
Can't get ffmpeg working through libcamerify. 
Tried different options in config.txt. Legacy stack is not available. 
Tried to disable gpu_mem line, forcing legacy media_controller, nothing helped. 

CmaTotal:         524288 kB
CmaFree:          507808 kB

Please help

us1@raspberrypi:~ $ libcamerify -d ffplay /dev/video0
ffplay version 5.1.4-0+rpt3+deb12u1 Copyright (c) 2003-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --extra-version=0+rpt3+deb12u1 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
[0:00:22.347352023] [2077]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:00:22.356121597] [2078]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 25-04-2024 (11:31:53)
[0:00:22.373334005] [2078]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:00:22.373413264] [2077] DEBUG V4L2Compat v4l2_compat_manager.cpp:80 Started camera manager
[0:00:22.373494801] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:50 [] open()
[0:00:22.373657338] [2077]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[0:00:22.373842227] [2077] DEBUG V4L2Compat v4l2_compat_manager.cpp:185 Opened /dev/video0 -> fd 12
[0:00:22.373853764] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:251 [/dev/video0 (fd:12)] vidioc_querycap()
[0:00:22.373860060] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:450 [/dev/video0 (fd:12)] vidioc_g_input()
[0:00:22.373865301] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:432 [/dev/video0 (fd:12)] vidioc_enuminput()
[0:00:22.373872116] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:12)] vidioc_g_fmt()
[0:00:22.373878005] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:351 [/dev/video0 (fd:12)] vidioc_s_fmt()
[0:00:22.373956690] [2077]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[0:00:22.374090301] [2077] DEBUG V4L2Compat v4l2_camera.cpp:125 Configuration adjusted
[0:00:22.374095412] [2077] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-YUV420
[0:00:22.374136634] [2077]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x600-YUV420
[0:00:22.374273005] [2078]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[video4linux2,v4l2 @ 0x7fff78000c20] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x7fff78000c20] Time per frame unknown
[0:00:22.375801134] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:478 [/dev/video0 (fd:12)] vidioc_reqbufs()
[0:00:22.375808856] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:485 256 buffers requested 
[0:00:22.375851801] [2077] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-YUV420
[0:00:22.375891782] [2077]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x600-YUV420
[0:00:22.375951819] [2078]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[0:00:22.387021505] [2078] ERROR V4L2 v4l2_videodevice.cpp:1248 /dev/video25[32:cap]: Not enough buffers provided by V4L2VideoDevice
[video4linux2,v4l2 @ 0x7fff78000c20] ioctl(VIDIOC_REQBUFS): Cannot allocate memory
[0:00:22.388631597] [2077] DEBUG V4L2Compat v4l2_camera_proxy.cpp:85 [/dev/video0 (fd:12)] close()
/dev/video0: Cannot allocate memory

camera_auto_detect=0
dtoverlay=imx708,media-controller=0
gpu_mem=256
dtoverlay=vc4-kms-v3d,cma-512

#gpu_mem=128
#dtoverlay=imx708,cam1

naushir, 2024-05-09:
Can you expand on what you are trying to do here?  ``rpicam-apps`` has ffmpeg/libav integration, so you may not need to run ffmpeg with libcamerify.

gentoosys, 2024-05-11:
I need to make perspective transform. 
something like this
ffmpeg -i /dev/video0 -vf 'perspective=W*.1:0:W*.9:0:-W*.1:H:W*1.1:H'
or https://www.educba.com/opencv-perspectivetransform

naushir, 2024-05-13:
Can you try piping from, say, ``rpicam-vid`` into ffplay instead of getting ffmepg to access ``/dev/videoX`` through libcamerify?

``rpicam-vid -t 0 -o - --width 1920 --height 1080 | ffplay ...``

gentoosys, 2024-05-13:
Tried that. There is 0.5sec delay

naushir, 2024-05-13:
If latency is a problem, perhaps it might be best to consider using the OpenCV postprocessing pipeline to perform your perspective correction directly in rpicam-apps and avoid using ffmpeg altogether.  

You might also want to consider doing this in Picamera 2 instead of rpicam-apps for quick prototyping.

gentoosys, 2024-05-13:
but why just not to fix libcamerify? it is not working at all for any program. Why it was created? It ever worked?

naushir, 2024-05-13:
``libcamerify`` wraps applications which use the v4l2 API to access simple cameras - V4L2 was only ever designed for simple webcam style camera systems.  The more modern libcamera API supports significantly more complex camera architectures with ISP  HW support.

``libcamerify`` does work with some application that use the simple v4l2 API.  However, IIRC ffmpeg as attempting to allocate 32 framebuffers through the simple v4l2 API.  When you use this in a more complex camera architecture with a hardware ISP through ``libcamerify``,  this is not possible due to memory limitations, and cannot be fixed/circumvented by ``libcamerrify``.

closed 2024-05-17 by naushir