raspberrypi/libcamera #136: Range format is ambiguous

yazzr, 2024-05-17:
Hi,

When I run this command on the raspberry pi 5 (cam v3)

`
gst-launch-1.0 libcamerasrc camera-name="/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" ! queue ! glimagesin
`

I get the following error (_libcamera v0.2.0+120-eb00c13d_)

```
Setting pipeline to PAUSED ...
[2:41:24.011162729] [8107]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[2:41:24.019007908] [8115]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[2:41:24.029278813] [8115]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Pipeline is live and does not need PREROLL ...
Got context from element 'sink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayWayland\)\ gldisplaywayland0";
[2:41:24.029972185] [8119]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[2:41:24.030187185] [8119] ERROR Stream stream.cpp:215 Range format is ambiguous
[2:41:24.030222908] [8119] ERROR Stream stream.cpp:215 Range format is ambiguous
[2:41:24.030229685] [8119] ERROR Stream stream.cpp:215 Range format is ambiguous
[2:41:24.030235963] [8119] ERROR Stream stream.cpp:215 Range format is ambiguous
[2:41:24.701001221] [8119] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Failed to configure camera: Invalid argument
Additional debug info:
```

I stumbled on this link: https://forums.raspberrypi.com/viewtopic.php?p=2221605#p2221605 which talks about a patch (see the second to last post) and I set about building libcamera from source using the patch in this link https://lists.libcamera.org/pipermail/libcamera-devel/2024-April/041642.html

The build goes successfully and I am pretty much following this guide: https://www.raspberrypi.com/documentation/computers/camera_software.html#build-libcamera-and-rpicam-apps

As per the documentation I then run `meson devenv -C build`

now when I run 

`gst-launch-1.0 libcamerasrc camera-name="/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" ! queue ! glimagesin`

I have the following error (_libcamera v0.2.0+120-eb00c13d-dirty_):

```
[7:56:14.802207345] [69931]  INFO IPAManager ipa_manager.cpp:143 libcamera is not installed. Adding '/home/pi/dev/libcamera/build/src/ipa' to the IPA search path
[7:56:14.808821310] [69931]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d-dirty (2024-05-15T17:53:05+02:00)
[7:56:14.819750277] [69932]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 15-05-2024 (17:07:49)
[7:56:14.839157505] [69932]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/pi/dev/libcamera/src/ipa/rpi/pisp/data'
[7:56:14.874367350] [69932]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[7:56:14.877433518] [69931]  INFO IPAManager ipa_manager.cpp:143 libcamera is not installed. Adding '/home/pi/dev/libcamera/build/src/ipa' to the IPA search path
[7:56:14.883866520] [69931]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d-dirty (2024-05-15T17:53:05+02:00)
[7:56:14.893606079] [69935]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 15-05-2024 (17:07:49)
[7:56:14.906603953] [69935]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/pi/dev/libcamera/src/ipa/rpi/pisp/data'
[7:56:14.932367906] [69935]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[7:56:14.933221592] [69939]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[7:56:14.961574268] [69939] ERROR Camera camera.cpp:1171 Can't configure camera with invalid configuration
2024-05-15 16:18:52.116 PROFILE createRtcCertificate(): [Certificate creation time] Time taken: 0 ms
```

I feel like I'm missing something very obvious - any help will be very much appreciated

naushir, 2024-05-20:
Did you run ``sudo meson -C build install`` to install libcamera to your system?   Another thing to check is run

```bash
ldd `which rpicam-hello` | grep libcamera
```

to see which libcamera libraries are in use by rpicam-apps.

yazzr, 2024-05-20:
Do you mean : `sudo jinja -C build install` ? if so then yes.

**running:**
```
ldd `which rpicam-hello` | grep libcamera
```

**results in**

```
root@raspberrypi:# ldd `which rpicam-hello` | grep libcamera
        libcamera.so.0.2 => /usr/local/lib/aarch64-linux-gnu/libcamera.so.0.2 (0x00007fff6f0e0000)
        libcamera-base.so.0.2 => /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0.2 (0x00007fff6f090000)
```

let me go through a reinstall:

`
sudo ninja -C build uninstall
`

**output:** 

```
ninja: Entering directory `build'
[0/1] /usr/bin/meson --internal uninstall
Deleted: /usr/local/include/libcamera/libcamera/ipa/core_ipa_interface.h
Deleted: /usr/local/include/libcamera/libcamera/ipa/raspberrypi_ipa_interface.h
Deleted: /usr/local/include/libcamera/libcamera/control_ids.h
Deleted: /usr/local/include/libcamera/libcamera/property_ids.h
Deleted: /usr/local/include/libcamera/libcamera/formats.h
Deleted: /usr/local/include/libcamera/libcamera/libcamera.h
Deleted: /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0.2.0
Deleted: /usr/local/lib/libyaml.a
Deleted: /usr/local/lib/aarch64-linux-gnu/libcamera.so.0.2.0
Deleted: /usr/local/libexec/libcamera/raspberrypi_ipa_proxy
Deleted: /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so
Deleted: /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so
Deleted: /usr/local/bin/lc-compliance
Deleted: /usr/local/bin/cam
Deleted: /usr/local/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
Deleted: /usr/local/lib/aarch64-linux-gnu/python3.11/site-packages/libcamera/_libcamera.so
Deleted: /usr/local/libexec/libcamera/v4l2-compat.so
Deleted: /usr/local/include/libcamera/libcamera/base/bound_method.h
Deleted: /usr/local/include/libcamera/libcamera/base/class.h
Deleted: /usr/local/include/libcamera/libcamera/base/compiler.h
Deleted: /usr/local/include/libcamera/libcamera/base/flags.h
Deleted: /usr/local/include/libcamera/libcamera/base/object.h
Deleted: /usr/local/include/libcamera/libcamera/base/shared_fd.h
Deleted: /usr/local/include/libcamera/libcamera/base/signal.h
Deleted: /usr/local/include/libcamera/libcamera/base/span.h
Deleted: /usr/local/include/libcamera/libcamera/base/unique_fd.h
Deleted: /usr/local/include/libcamera/libcamera/ipa/ipa_controls.h
Deleted: /usr/local/include/libcamera/libcamera/ipa/ipa_interface.h
Deleted: /usr/local/include/libcamera/libcamera/ipa/ipa_module_info.h
Deleted: /usr/local/include/libcamera/libcamera/camera.h
Deleted: /usr/local/include/libcamera/libcamera/camera_manager.h
Deleted: /usr/local/include/libcamera/libcamera/color_space.h
Deleted: /usr/local/include/libcamera/libcamera/controls.h
Deleted: /usr/local/include/libcamera/libcamera/fence.h
Deleted: /usr/local/include/libcamera/libcamera/framebuffer.h
Deleted: /usr/local/include/libcamera/libcamera/framebuffer_allocator.h
Deleted: /usr/local/include/libcamera/libcamera/geometry.h
Deleted: /usr/local/include/libcamera/libcamera/logging.h
Deleted: /usr/local/include/libcamera/libcamera/orientation.h
Deleted: /usr/local/include/libcamera/libcamera/pixel_format.h
Deleted: /usr/local/include/libcamera/libcamera/request.h
Deleted: /usr/local/include/libcamera/libcamera/stream.h
Deleted: /usr/local/include/libcamera/libcamera/transform.h
Deleted: /usr/local/include/libcamera/libcamera/version.h
[... 376 lines cut ...]
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(446): gst_libcamera_src_negotiate (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
Camera::configure() failed with error code -22
Execution ended after 0:00:01.287573342
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Internal data stream error.
Setting pipeline to NULL ...
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(645): gst_libcamera_src_task_enter (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
streaming stopped, reason not-negotiated (-4)
Freeing pipeline ...
```

and 

```
ldd `which rpicam-hello` | grep libcamera
```

still comes back with

```
[libcamera] pi@raspberrypi:~/dev/libcamera/build $ ldd `which rpicam-hello` | grep libcamera
        libcamera.so.0.2 => /home/pi/dev/libcamera/build/src/libcamera/libcamera.so.0.2 (0x00007ffed5360000)
        libcamera-base.so.0.2 => /home/pi/dev/libcamera/build/src/libcamera/base/libcamera-base.so.0.2 (0x00007ffed5310000)
```

naushir, 2024-05-20:
> ``meson devenv -C build``

I'm not entirely familiar with what this does.  Can you try running ``rpicam-hello -t 0`` without issuing the ``meson devenv`` command to start with, and let me know if that starts a simple preview window?

yazzr, 2024-05-20:
`rpicam-hello -t 0`  creates preview and this is the output:

```
1:26:12.838390762] [26683]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d-dirty (2024-05-20T13:11:33+02:00)
[1:26:12.850501136] [26684]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 15-05-2024 (17:07:49)
[1:26:12.966958405] [26684]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made X/EGL preview window
[1:26:13.604999850] [26683]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[1:26:13.605233850] [26683]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR_PISP_COMP1
[1:26:13.605368257] [26684]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
```

> > `meson devenv -C build`
> 
> I'm not entirely familiar with what this does. Can you try running `rpicam-hello -t 0` without issuing the `meson devenv` command to start with, and let me know if that starts a simple preview window?

I got the `meson devenv -C build` command from https://github.com/raspberrypi/libcamera?tab=readme-ov-file#using-gstreamer-plugin which says "To use the GStreamer plugin from the source tree, use the meson devenv command. This will create a new shell instance with the GST_PLUGIN_PATH"

without  command `meson devenv -C build`, running `gst-launch-1.0 libcamerasrc camera-name="/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" ! queue ! glimagesink`

results in:

```
WARNING: erroneous pipeline: no element "libcamerasrc"
```

Sure enough the `GST_PLUGIN_PATH` is being set to `/home/pi/dev/libcamera/build/src/gstreamer`

naushir, 2024-05-20:
Got it.  I'm able to reproduce your error as well.  I'll try to debug it and see what could be going wrong.

naushir, 2024-05-20:
Would you be able to try the following pipeline and see if it works:

```
gst-launch-1.0 libcamerasrc camera-name="/base/axi/pcie@120000/rp1/i2c@88000/imx708@1a" ! 'video/x-raw,format=NV12' ! queue ! glimagesink
```

It seems that you must setup a stream format otherwise the gst component (or libcamera) gets quite upset.

yazzr, 2024-05-21:
Thank you very much, that did it - would be great to default to one (and document that) - I should have run the streaming example also which includes the stream format . Maybe we can update the documentation (https://github.com/raspberrypi/libcamera?tab=readme-ov-file#using-gstreamer-plugin)? 

gst-launch-1.0 libcamerasrc ! 'video/x-raw,**_format=NV12_**,width=1280,height=720' ! \
     queue ! glimagesink

otherwise happy to close it

naushir, 2024-05-21:
This document comes from the upstream libcamera repo.  I will talk to them about updating the documentation, thanks!