raspberrypi/libcamera #137: Installation issue

pradyxdev, 2024-05-18:
**OS :** Dietpi 64x
**Device :** Raspberry pi 4

I have already tried reinstalling jinja2 via pip

**Command :** 
ninja -C build install

**Error :** 

ninja: Entering directory `build'                                                              [2/3] Generating Documentation/documentation with a custom command                             FAILED: Documentation/html                                                                     /usr/bin/sphinx-build -D release=v0.2.0+150-2031e2f2 -q -W -b html /home/dietpi/libcamera/Documentation Documentation/html                                                                    Traceback (most recent call last):                                                               File "/usr/bin/sphinx-build", line 33, in <module>                                               sys.exit(load_entry_point('Sphinx==3.4.3', 'console_scripts', 'sphinx-build')())             File "/usr/bin/sphinx-build", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/sphinx/cmd/build.py", line 25, in <module>
    from sphinx.application import Sphinx
  File "/usr/lib/python3/dist-packages/sphinx/application.py", line 43, in <module>
  File "/usr/lib/python3/dist-packages/sphinx/application.py", line 43, in <module>
    from sphinx.registry import SphinxComponentRegistry
  File "/usr/lib/python3/dist-packages/sphinx/registry.py", line 24, in <module>
    from sphinx.builders import Builder
  File "/usr/lib/python3/dist-packages/sphinx/builders/__init__.py", line 26, in <module>
    from sphinx.util import import_object, logging, progress_message, rst, status_iterator
  File "/usr/lib/python3/dist-packages/sphinx/util/rst.py", line 21, in <module>
    from jinja2 import Environment, environmentfilter
ImportError: cannot import name 'environmentfilter' from 'jinja2' (/usr/local/lib/python3.9/dist-packages/jinja2/__init__.py)
ninja: build stopped: subcommand failed.

**Output from mesa :**

The Meson build system                                                                                                                                                 Version: 1.0.0                                                                                                                                                         Source dir: /home/dietpi/libcamera                                                                                                                                     Build dir: /home/dietpi/libcamera/build                                                                                                                                Build type: native build                                                                                                                                               Project name: libcamera                                                                                                                                                Project version: 0.2.0                                                                                                                                                 C compiler for the host machine: cc (gcc 10.2.1 "cc (Debian 10.2.1-6) 10.2.1 20210110")                                                                                C linker for the host machine: cc ld.bfd 2.35.2                                                                                                                        C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Debian 10.2.1-6) 10.2.1 20210110")                                                                            C++ linker for the host machine: c++ ld.bfd 2.35.2                                                                                                                     Host machine cpu family: aarch64                                                                                                                                       Host machine cpu: aarch64                                                                                                                                              Header "unistd.h" has symbol "issetugid" : NO (cached)                                                                                                                 Header "locale.h" has symbol "locale_t" : YES (cached)                                                                                                                 Header "sys/mman.h" has symbol "memfd_create" : YES (cached)                                                                                                           Header "stdlib.h" has symbol "secure_getenv" : YES (cached)                                                                                                            Compiler for C supports arguments -Wno-c99-designator: NO (cached)                                                                                                     Found pkg-config: /usr/bin/pkg-config (0.29.2)                                                                                                                         Found CMake: /usr/bin/cmake (3.18.4)                                                                                                                                   Run-time dependency lttng-ust found: NO (tried pkgconfig and cmake)                                                                                                    Program ./parser.py found: YES (/home/dietpi/libcamera/utils/ipc/./parser.py)                                                                                          Program ./generate.py found: YES (/home/dietpi/libcamera/utils/ipc/./generate.py)                                                                                      Program ./extract-docs.py found: YES (/home/dietpi/libcamera/utils/ipc/./extract-docs.py)                                                                              Program ./gen-tp-header.py found: YES (/home/dietpi/libcamera/utils/tracepoints/./gen-tp-header.py)                                                                    Configuring version.h using configuration                                                                                                                              Program openssl found: YES (/usr/bin/openssl)                                                                                                                          Library atomic found: YES                                                                                                                                              Dependency threads found: YES unknown (cached)
Dependency libdw found: YES 0.183 (cached)
Dependency libunwind found: YES 1.3.2 (cached)
Header "execinfo.h" has symbol "backtrace" : YES (cached)                                                                                                              Checking for function "dlopen" : NO (cached)
Library dl found: YES
Dependency libudev found: YES 247 (cached)
Dependency yaml-0.1 found: YES 0.2.2 (cached)
Run-time dependency gnutls found: NO (tried pkgconfig and cmake)                                                                                                       Dependency libcrypto found: YES 1.1.1w (cached)
Dependency libexif skipped: feature android disabled                                                                                                                   Dependency libjpeg skipped: feature android disabled                                                                                                                   Dependency libevent_pthreads found: YES 2.1.12-stable (cached)                                                                     [... cut ...]         Hotplug support          : YES
    Tracing support          : NO
    Android support          : NO
    GStreamer support        : YES
    Python bindings          : YES
    V4L2 emulation support   : NO
    cam application          : YES
    qcam application         : YES
    lc-compliance application: YES
    Unit tests               : NO

  Subprojects
    gtest                    : YES

  User defined options
    pycamera                 : enabled

Found ninja-1.10.1 at /usr/bin/ninja

naushir, 2024-05-20:
It seems you have a problem with the documentation dependencies for the libcamera build.  Can you try disabling documentation (``-Ddocumentation=disabled``) in your meson config and try again?

naushir, 2024-06-05:
Any updates on this?  If not I'll close this shortly.

closed 2024-06-10 by naushir