raspberrypi/libcamera #141: libcamerasrc produces buffers with invalid size on Raspberry Pi 3 with official RPi Cam v2.1 attached

RSATom, 2024-06-04:
and it leads to "green" picture.

```
pi@rpi3:~ $ LIBCAMERA_LOG_LEVELS=7 gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1280,height=720' ! queue ! autovideosink
Setting pipeline to PAUSED ...
[0:20:12.784582040] [1178]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:20:12.833118956] [1181]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:20:12.837026456] [1181]  WARN RPI vc4.cpp:392 Mismatch between Unicam and CamHelper for embedded data usage!
[0:20:12.838046348] [1181]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[0:20:12.847490082] [1185]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-NV21
[0:20:12.848495183] [1181]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
Redistribute latency...
handling interrupt.:99.
Interrupt: Stopping pipeline ...
Execution ended after 0:01:52.063719828
Setting pipeline to NULL ...
Freeing pipeline ...
```

If I'm using `gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1920,height=1080' ! queue ! autovideosink` it works as expected.
Also, there is no issue on Raspberry Pi 4 with the same camera module.

On both devices installed the latest Raspbian Bookwork x64.

RSATom, 2024-06-05:
It looks like I found the point when it breaks. Right after install fresh Raspberry Pi Os and install all required packages (but without `sudo apt update && sudo apt upgrade`) it works. But it breaks after  `sudo apt update && sudo apt upgrade`. So I have a feeling it's something related to kernel drivers. Going to to `rpi-update`...

RSATom, 2024-06-05:
No, `rpi-update` didn't help...

RSATom, 2024-06-05:
on RPi 4 after `rpi-update` even `gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1920,height=1080' ! queue ! autovideosink` stops working.... I see video with very distorted colors (mostly yellow)

naushir, 2024-06-05:
```
gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1920,height=1080' ! queue ! autovideosink
``` 
on  a Pi 4 seems to work as expected for me.  However, note that I am budling/running libcamera from the ``next`` branch in this repo.  Can you share an image showing what goes wrong for you?

RSATom, 2024-06-05:
@naushir 
Just to avoid possible side effects I've did everything from scratch :
1. flash Raspberry Pi OS Lite (64-bit) Released 2024-03-15
2. `sudo apt update && sudo apt upgrade`
3. `sudo reboot`
4. `sudo apt install gstreamer1.0-plugins-{base,good,bad,ugly} gstreamer1.0-tools gstreamer1.0-libcamera -y`
5. `gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1920,height=1080' ! queue ! autovideosink`
[image]
6. `gst-launch-1.0 libcamerasrc ! queue ! autovideosink`
[image]

naushir, 2024-06-05:
Hmm, with

```
gst-launch-1.0 libcamerasrc ! queue ! autovideosink
```

I see the green corruption like you, but with

```
gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1920,height=1080' ! queue ! autovideosink
```

everything works fine.  I suspect it may be something in the libcamerasrc gst component not setting up a default format and getting confused.  I have to admit, I don't know anything about gstreamer, so it's only a guess.

And just to confirm, ``rpicam-hello`` gives you a reasonable looking image?

RSATom, 2024-06-05:
> And just to confirm, rpicam-hello gives you a reasonable looking image?

yes, it works fine.

naushir, 2024-06-05:
Are you able to build and deploy libcamera yourself for testing purposes.  If so, can you try the ``next`` branch on this repo, which is what I am using, and see if it works?

RSATom, 2024-06-05:
yes I can. I'll try.

RSATom, 2024-06-05:
@naushir tried `libcamera` from `next` branch (did fresh Raspbian install on SD card) - result is the same as before for both `gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1920,height=1080' ! queue ! autovideosink`  and `gst-launch-1.0 libcamerasrc ! queue ! autovideosink`...

naushir, 2024-06-06:
And presumably this is with an up-to-date kernel through ``rpi-update``?  I can't really see what could be going wrong :(

RSATom, 2024-06-06:
Tried after `rpi-update` - didn't help...

naushir, 2024-06-06:
I think this is the main difference:
gstreamer:
[0:08:33.160903645] [1029]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-NV21

rpicam-hello
[0:13:38.211593356] [1070]  INFO Camera camera.cpp:1183 configuring streams: (0) 1640x922-YUV420 (1) 1920x1080-SBGGR10_CSI2P

gstreamer is selecting an NV12 format, while rpicam-hello is selecting a planar YUV240 format.  The hardware supports NV21 output (and it does seem to work on my system), so this difference should be inconsequential.

RSATom, 2024-06-06:
If I remember correctly I've tried to use `video/x-raw,width=1920,height=1080,format=I420` - it didn't help. I'll check one more time and will provide output.

RSATom, 2024-06-06:
> @RSATom, if you set the resolution to 1920x1080 as per my command above, do you get a correct looking image?

Some time ago I did. I'm sure 100%.  But now I didn't. I don't now what changed....

naushir, 2024-06-06:
```
INFO RPI pipeline_base.cpp:346  stream res 1280x1080-NV21 bpl 1920
```

This is the key line in the failure case.   Assuming if you try 1920x1080 and it works, then we can narrow the problem down to the 1280x1080 framebuffer stride being incorrectly used by gstreamer in one of 2 ways:

- The stride is actually 1920, but gstreamer is treating it as if it was 1280
- The stride is 1280, but gstreamer is treating it as 1920

I think the correct "fix", like what was suggested in https://github.com/raspberrypi/libcamera/issues/138 is to ensure gstreamer clears the stride value in the StreamConfiguration structure when it configures libcamera.  That way the stride reported back is guaranteed to be correct for downstream use.

Unfortunately I know next to nothing about the libcameragst component.  Perhaps @kbingham might be able to help here?

naushir, 2024-06-06:
I'm now considering not setting a stride in the pipeline handler generateConfiguration() call.  The following patch:
```patch
diff --git a/src/libcamera/pipeline/rpi/common/pipeline_base.cpp b/src/libcamera/pipeline/rpi/common/pipeline_base.cpp
index 0034e06d5ec2..ab123b22248e 100644
--- a/src/libcamera/pipeline/rpi/common/pipeline_base.cpp
+++ b/src/libcamera/pipeline/rpi/common/pipeline_base.cpp
@@ -490,14 +490,13 @@ PipelineHandlerBase::generateConfiguration(Camera *camera, Span<const StreamRole
                StreamFormats formats(deviceFormats);
                StreamConfiguration cfg(formats);
                cfg.size = size;
+               cfg.stride = 0;
                cfg.pixelFormat = pixelFormat;
                cfg.colorSpace = colorSpace;
                cfg.bufferCount = bufferCount;
                config->addConfiguration(cfg);
        }

-       config->validate();
-
        return config;
 }
```

fixes the ``1280x1080`` case for me.

RSATom, 2024-06-06:
Unfortunately will not be able to try it today, but I promise I'll do it tomorrow. Thanks!

kbingham, 2024-06-06:
> I think the correct "fix", like what was suggested in #138 is to ensure gstreamer clears the stride value in the StreamConfiguration structure when it configures libcamera. That way the stride [...]

I haven't got all context here yet, but also looking up - I see a patch going to the generateConfiguration/validate() calls. That's where this should be fixed I think - as it's the pipeline handlers responsibility to make sure the configurations are valid.

Applications shouldn't have to know in advance to reset the stride - but they should be able to 'set' it explicitly if they are queueing buffers in from an external source I think ?

Sounds like review topics to handle when Naush posts his patch to the lists though.

RSATom, 2024-06-06:
@naushir your patch helped partially. Now size of video is correct no matter what video size I requesting. But colors are still distorted.

RSATom, 2024-06-06:
@naushir please ignore my last complain. Broken colors - it's something wrong with `glimagesink` without X11 running...

RSATom, 2024-06-07:
@naushir now I can confirm your patch definitely fixes issue. I've tested it with [app where I initially found issue](https://snapcraft.io/webrtsp-camera-streamer) - and it works great now. The only remaining issue is compatibility with RPi 5 - but it's totally different story. Thank you!

naushir, 2024-06-10:
Patch has been posed, so I'll resolve this issue now.

closed 2024-06-10 by naushir