raspberrypi/libcamera #145: On RPi5:  Unable to acquire a Unicam instance

jzb-hpe, 2024-06-14:
I tried a recent libcamera build from source on github on an imx477, and today's github on an imx296 (GS, official RPI), and I get the same unsatisfactory results:  no camera detected (at least not entirely):

# LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[0:06:25.674125102] [3275] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /usr/local/lib64/libcamera/ipa_rkisp1.so is signed
[0:06:25.674376399] [3275] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib64/libcamera/ipa_rkisp1.so'
[0:06:25.676115274] [3275] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib64/libcamera/ipa_rpi_vc4.so is signed
[0:06:25.676193830] [3275] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib64/libcamera/ipa_rpi_vc4.so'
[0:06:25.676852072] [3275] DEBUG IPAModule ipa_module.cpp:334 ipa_soft_simple.so: IPA module /usr/local/lib64/libcamera/ipa_soft_simple.so is signed
[0:06:25.676894868] [3275] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib64/libcamera/ipa_soft_simple.so'
[0:06:25.677126813] [3275]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[0:06:25.677305129] [3276] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[0:06:25.684867537] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media2
[0:06:25.684954444] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rp1-cfe
[0:06:25.685305390] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[0:06:25.685445112] [3276] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[0:06:25.685832521] [3276] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:06:25.685852743] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[0:06:25.686162559] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[0:06:25.686227892] [3276] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 9 missing dependencies
[0:06:25.686466522] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:06:25.686525097] [3276] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 9 missing dependencies
[0:06:25.688035526] [3276] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:06:25.688062878] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[0:06:25.689521345] [3276] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:06:25.689558475] [3276] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:06:25.689898031] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'imx8-isi'
[0:06:25.690201828] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'mali-c55'
[0:06:25.690292921] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rkisp1'
[0:06:25.690458199] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rpi/vc4'
[0:06:25.690579033] [3276] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
[0:06:25.690597218] [3276] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
[0:06:25.690614070] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'simple'
[0:06:25.690718718] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'uvcvideo'
Available cameras:

So no cameras, and the only clue that I can see is the "Unable to acquire a Unicam instance" message.

Not sure what the problem is.  Kernel seems to detect stuff, but I don't know if its all of it since I've never seen a success, but here's what I got:

# uname -a
Linux styx 6.6.28_p20240423-raspberrypi-linux-gnu #1 SMP PREEMPT Mon Jun  3 19:23:49 CDT 2024 aarch64 GNU/Linux

# lsmod | grep imx
imx296                 49152  0
regmap_i2c             49152  1 imx296

# dmesg | grep imx
[    0.576908] platform 1f00110000.csi: Fixed dependency cycle(s) with /axi/pcie@120000/rp1/i2c@88000/imx296@1a
[    0.621621] rp1-cfe 1f00110000.csi: found subdevice /axi/pcie@120000/rp1/i2c@88000/imx296@1a
[    7.784384] imx296 6-001a: found IMX296LQ (23.7C)
[    7.785186] rp1-cfe 1f00110000.csi: Using sensor imx296 6-001a for capture

# dmesg | grep rpi
[    0.394377] rpi-rtc soc:rpi_rtc: registered as rtc0
[    0.395920] rpi-rtc soc:rpi_rtc: setting system clock to 2024-06-13T20:22:27 UTC (1718310147)
[    6.891672] rpi-gpiomem 107d508500.gpiomem: window base 0x107d508500 size 0x00000040
[    6.892388] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    6.892942] rpi-gpiomem 107d508500.gpiomem: initialised 1 regions as /dev/gpiomem1
[    6.893052] rpi-gpiomem 107d517c00.gpiomem: window base 0x107d517c00 size 0x00000040
[    6.893153] rpi-gpiomem 107d517c00.gpiomem: initialised 1 regions as /dev/gpiomem2
[    6.893216] rpi-gpiomem 107d504100.gpiomem: window base 0x107d504100 size 0x00000020
[    6.893270] rpi-gpiomem 107d504100.gpiomem: initialised 1 regions as /dev/gpiomem3
[    6.893320] rpi-gpiomem 107d510700.gpiomem: window base 0x107d510700 size 0x00000020
[    6.893362] rpi-gpiomem 107d510700.gpiomem: initialised 1 regions as /dev/gpiomem4
[    6.893429] rpivid 1000800000.codec: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    6.893562] rpi-gpiomem 1f000d0000.gpiomem: window base 0x1f000d0000 size 0x00030000
[    6.893627] rpi-gpiomem 1f000d0000.gpiomem: initialised 1 regions as /dev/gpiomem0
[    6.895242] rpivid 1000800000.codec: Device registered as /dev/video19

Either config.txt with:

camera_auto_detect=1

or

camera_auto_detect=0
dtoverlay=imx296,cam0

produces identical results as above.

Any ideas?  Since this is my first camera setup, I could be doing something incorrect or incomplete.  This is on a Raspberry Pi5.  Thanks for any help...

6by9, 2024-06-14:
Unicam is the Pi0-4 CSI-2 receiver. Pi5 uses pisp.
```
[0:06:25.689898031] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'imx8-isi'
[0:06:25.690201828] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'mali-c55'
[0:06:25.690292921] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rkisp1'
[0:06:25.690458199] [3276] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rpi/vc4'
```
pisp isn't listed as a pipeline handler that you've built into libcamera.

Did you build libcamera from this repo, or from the upstream git.linuxtv.org? pisp support isn't upstream yet.

naushir, 2024-06-14:
Yup, you don't have pisp support built into libcamera.  You need to change your meson configure command like detailed [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera) to enable both vc4 and pisp support in libcamera.

jzb-hpe, 2024-06-14:
Thanks for your help.  Pulled it from the wrong repo:  https://git.libcamera.org/libcamera/libcamera.git.  Works from here: https://github.com/raspberrypi/libcamera.git.

closed 2024-06-14 by jzb-hpe