raspberrypi/libcamera #147: licamera on Jassy: Preview window unavailable

kartheekmedathati, 2024-06-19:
Preview window unavailable
c:~/libcamera# libcamera-hello 
[17:20:06.918303697] [4272]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[17:20:06.948164232] [4273]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[17:20:06.952087769] [4273]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[17:20:06.952182582] [4273]  INFO RPI pipeline_base.cpp:1104 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900

naushir, 2024-06-19:
Do you mean Debian Jessie?

naushir, 2024-06-19:
You also seem to be building locally instead of installing a package. Please ensure you have all dependencies available for drm and elf before your meson configure step.

naushir, 2024-06-26:
Any updates on this please?  If not, I'll close this shortly.

closed 2024-06-26 by naushir