raspberrypi/libcamera #148: cam: How to actually view the captured images

bryangd34, 2024-06-20:
I am curious about how to actually view the captures with cam, for example I run:

`cam -c 2 --capture=10 --file`
[0:20:56.708360625] [2937]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[0:20:56.737611142] [2940]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 19-06-2024 (18:49:22)
[0:20:56.750133839] [2940]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:20:56.750208432] [2940]  INFO RPI pisp.cpp:695 libpisp version v1.0.6 b567f0455680 19-06-2024 (18:49:22)
[0:20:56.759870684] [2940]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media1 and ISP device /dev/media4 using PiSP variant BCM2712_C0
[0:20:56.760175594] [2937]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
Camera configuration adjusted
Using camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 as cam0
[0:20:56.760413984] [2937]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x600-XRGB8888
[0:20:56.760558708] [2940]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
cam0: Capture 10 frames
1257.171040 (0.00 fps) cam0-stream0 seq: 000007 bytesused: 1920000
1257.231347 (16.58 fps) cam0-stream0 seq: 000008 bytesused: 1920000
1257.291403 (16.65 fps) cam0-stream0 seq: 000009 bytesused: 1920000
1257.351308 (16.69 fps) cam0-stream0 seq: 000010 bytesused: 1920000
1257.411546 (16.60 fps) cam0-stream0 seq: 000011 bytesused: 1920000
1257.471527 (16.67 fps) cam0-stream0 seq: 000012 bytesused: 1920000
1257.531574 (16.65 fps) cam0-stream0 seq: 000013 bytesused: 1920000
1257.591758 (16.62 fps) cam0-stream0 seq: 000014 bytesused: 1920000
1257.651189 (16.83 fps) cam0-stream0 seq: 000015 bytesused: 1920000
1257.711728 (16.52 fps) cam0-stream0 seq: 000016 bytesused: 1920000

This creates 10 .bin files. I don't know how to use these files and have had errors when trying to convert to jpeg using other software. Any help would be much appreciated. 
Thanks.

6by9, 2024-06-20:
It's told you the output format as
```
configuring streams: (0) 800x600-XRGB8888
```
800 * 600 * 32bit/pixel = 1920000 bytes.
It's raw RGB data, so something like [Vooya](https://www.offminor.de/) should happily display them once you've told it the geometry and pixel format.

naushir, 2024-06-20:
Alternatively, you can use [``rpicam-still``](https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-still) to capture directly to a JPEG (and other) format.

pinchartl, 2024-06-20:
raw2rgbpnm (https://git.retiisi.eu/?p=~sailus/raw2rgbpnm.git;a=summary) is a free software command line tool that supports converting a wide variety of raw binary formats to PPM. For some Bayer formats, bayer2rgb (https://github.com/jdthomas/bayer2rgb) may work better.

bryangd34, 2024-06-20:
Thanks @6by9 , @naushir and @pinchartl 
I was able to get an image ultimately using **rpicam-still** (had to run it with sudo to get around _dmaHeap allocation failure for rpicam-apps0_ error).
Appreciate the swift help. Feel free to close this issue if you so please.

naushir, 2024-06-20:
Thanks for confirming.  You should not need to run ``rpicam-still`` with ``sudo`` in a standard Raspberry Pi OS environment.  This is likely down to group permissions for your username.

closed 2024-06-20 by naushir