raspberrypi/libcamera #151: FAILED: pipeline_rpi_pisp_pisp.cpp.o ninja build

Jai-GAY, 2024-06-26:
From https://github.com/raspberrypi/libcamera
 * branch              main       -> FETCH_HEAD
Already up to date.

The Meson build system
Version: 0.64.0
Source dir: /home/pi/libcamera
Build dir: /home/pi/libcamera/build
Build type: native build
Project name: libcamera
Project version: 0.3.0
```
  Configuration
    SoftISP support          : False
    IPA modules signed with  : gnutls
    Enabled pipelines        : rpi/vc4
                               rpi/pisp
    Enabled IPA modules      : rpi/vc4
                               rpi/pisp
    Controls files           : control_ids_draft.yaml
                               control_ids_core.yaml
                               control_ids_rpi.yaml
    Properties files         : property_ids_draft.yaml
                               property_ids_core.yaml
    Hotplug support          : YES
    Tracing support          : NO
    Android support          : NO
    GStreamer support        : YES
    Python bindings          : YES
    V4L2 emulation support   : YES
    cam application          : NO
    qcam application         : NO
    lc-compliance application: NO
    Unit tests               : NO

  User defined options
    buildtype                : release
    cam                      : disabled
    documentation            : disabled
    gstreamer                : enabled
    ipas                     : rpi/vc4,rpi/pisp
    lc-compliance            : disabled
    pipelines                : rpi/vc4,rpi/pisp
    pycamera                 : enabled
    qcam                     : disabled
    test                     : false
    v4l2                     : true
```
```
FAILED: src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o 
c++ -Isrc/libcamera/libcamera.so.0.3.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/local/include/libpisp -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/pi/libcamera/build/config.h -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -MF src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o.d -o src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -c ../src/libcamera/pipeline/rpi/pisp/pisp.cpp
In file included from /usr/include/c++/11/bits/move.h:57,
                 from /usr/include/c++/11/bits/stl_pair.h:59,
                 from /usr/include/c++/11/utility:70,
                 from /usr/include/c++/11/algorithm:60,
                 from ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:8:
/usr/include/c++/11/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = void]’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:26:   required from here
/usr/include/c++/11/type_traits:2579:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2579 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:33: error: template argument 2 is invalid
  742 |         SharedMemObject<FrontEnd> fe_;
      |                                 ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:743:32: error: template argument 2 is invalid
  743 |         SharedMemObject<BackEnd> be_;
      |                                ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual bool libcamera::PipelineHandlerPiSP::match(libcamera::DeviceEnumerator*)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:909:61: error: template argument 2 is invalid
  909 |                         pisp->fe_ = SharedMemObject<FrontEnd>
[... 163 lines cut ...]
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2170:46: error: invalid type argument of unary ‘*’ (have ‘int’)
 2170 |                 std::scoped_lock<FrontEnd> l(*fe_);
      |                                              ^~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2172:20: error: base operand of ‘->’ is not a pointer
 2172 |                 fe_->Prepare(reinterpret_cast<pisp_fe_config *>(configBuffer.data()));
      |                    ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘void libcamera::PiSPCameraData::prepareBe(uint32_t, bool)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2210:12: error: base operand of ‘->’ is not a pointer
 2210 |         be_->Prepare(configBuffer);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2226:34: error: base operand of ‘->’ is not a pointer
 2226 |                         of << be_->GetJsonConfig(configBuffer);
      |                                  ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: At global scope:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:668:14: error: ‘unsigned int libcamera::{anonymous}::calculateSwDownscale(const libcamera::V4L2DeviceFormat&, unsigned int, unsigned int)’ defined but not used [-Werror=unused-function]
  668 | unsigned int calculateSwDownscale(const V4L2DeviceFormat &format, unsigned int largestWidth,
      |              ^~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
[102/202] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/sensor_camera_sensor_properties.cpp.o
[103/202] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/sensor_camera_sensor.cpp.o
[104/202] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_common_pipeline_base.cpp.o
[105/202] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/v4l2_subdevice.cpp.o
[106/202] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_vc4_vc4.cpp.o
ninja: build stopped: subcommand failed.
```

naushir, 2024-06-26:
Did you install libpisp manually?  If so, you will need to update the repo and rebuild/install.  If not, run ``meson subprojects update`` in the root of your libcamera directory and rebuild libcamera.

Jai-GAY, 2024-06-26:
> Did you install libpisp manually?

install manually? no, via apt-get or meson.
I built libcamera once successfully previously, version 0.1.0

> `meson subprojects update` in the root of your libcamera directory and rebuild libcamera.

```
meson subprojects update
Progress: 0 / 5 (libyuv)
[KProgress: 0 / 5 (libyuv, libpisp)
[K
Updating libyuv...
  -> Not used.
Progress: 1 / 5 (libpisp)
[KProgress: 1 / 5 (libpisp, nlohmann_json)
[K
Updating nlohmann_json...
  -> Subproject has not changed, or the new source/patch needs to be extracted on the same location.
     Pass --reset option to delete directory and redownload.
Progress: 2 / 5 (libpisp)
[KProgress: 2 / 5 (libpisp, libyaml)
[KProgress: 2 / 5 (gtest, libpisp, libyaml)
[K
Updating libyaml...
  -> Not used.
Progress: 3 / 5 (gtest, libpisp)
[K
Updating gtest...
  -> Not used.
Progress: 4 / 5 (libpisp)
[K
Updating libpisp...
  -> b567f04 (HEAD, tag: v1.0.6) build: Bump release version to 1.0.6 [Naushir Patuck]
Progress: 5 / 5 ()
[K
WARNING: Please check logs above as command failed in some subprojects which could have been left in conflict state: nlohmann_json
ninja: Entering directory `build'
[1/99] Generating src/libcamera/version.cpp with a custom command
[2/99] Generating src/libcamera/proxy/worker/raspberrypi_proxy_worker with a custom command
[3/99] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/meson-generated_.._version.cpp.o
[4/99] Compiling C++ object src/ipa/libipa/libipa.a.p/fc_queue.cpp.o
[5/99] Compiling C++ object src/ipa/libipa/libipa.a.p/algorithm.cpp.o
[6/99] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/device_enumerator_udev.cpp.o
[7/99] Compiling C++ object src/ipa/libipa/libipa.a.p/camera_sensor_helper.cpp.o
[8/99] Compiling C++ object src/ipa/libipa/libipa.a.p/histogram.cpp.o
[9/99] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o
FAILED: src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o 
c++ -Isrc/libcamera/libcamera.so.0.3.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/local/include/libpisp -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/pi/libcamera/build/config.h -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -MF src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o.d -o src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -c ../src/libcamera/pipeline/rpi/pisp/pisp.cpp
In file included from /usr/include/c++/11/bits/move.h:57,
                 from /usr/include/c++/11/bits/stl_pair.h:59,
                 from /usr/include/c++/11/utility:70,
                 from /usr/include/c++/11/algorithm:60,
                 from ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:8:
/usr/include/c++/11/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = void]’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:26:   required from here
/usr/include/c++/11/type_traits:2579:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2579 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:33: error: template argument 2 is invalid
  742 |         SharedMemObject<FrontEnd> fe_;
      |                                 ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:743:32: error: template argument 2 is invalid
  743 |         SharedMemObject<BackEnd> be_;
      |                                ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual bool libcamera::PipelineHandlerPiSP::match(libcamera::DeviceEnumerator*)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:909:61: error: template argument 2 is invalid
  909 |                         pisp->fe_ = SharedMemObject<FrontEnd>
      |                                                             ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:911:60: error: template argument 2 is invalid
[... 161 lines cut ...]
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2170:46: error: invalid type argument of unary ‘*’ (have ‘int’)
 2170 |                 std::scoped_lock<FrontEnd> l(*fe_);
      |                                              ^~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2172:20: error: base operand of ‘->’ is not a pointer
 2172 |                 fe_->Prepare(reinterpret_cast<pisp_fe_config *>(configBuffer.data()));
      |                    ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘void libcamera::PiSPCameraData::prepareBe(uint32_t, bool)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2210:12: error: base operand of ‘->’ is not a pointer
 2210 |         be_->Prepare(configBuffer);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2226:34: error: base operand of ‘->’ is not a pointer
 2226 |                         of << be_->GetJsonConfig(configBuffer);
      |                                  ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: At global scope:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:668:14: error: ‘unsigned int libcamera::{anonymous}::calculateSwDownscale(const libcamera::V4L2DeviceFormat&, unsigned int, unsigned int)’ defined but not used [... cut ...]
cc1plus: all warnings being treated as errors
[10/99] Compiling C++ object src/ipa/libipa/libipa.a.p/module.cpp.o
[11/99] Compiling C++ object src/ipa/libipa/libipa.a.p/agc_mean_luminance.cpp.o
[12/99] Compiling C++ object src/ipa/libipa/libipa.a.p/pwl.cpp.o
[13/99] Compiling C++ object src/ipa/libipa/libipa.a.p/vector.cpp.o
[14/99] Compiling C++ object src/libcamera/proxy/worker/raspberrypi_ipa_proxy.p/meson-generated_.._raspberrypi_ipa_proxy_worker.cpp.o
ninja: build stopped: subcommand failed.
```

naushir, 2024-06-26:
If you are ok with messing around with local builds, I can suggest you build libpisp manually as well?

- First thing, ensure any system install of libpisp is removed: ``sudo apt purge libpisp``
- Clone/build/install libpisp.  Instructions can be found at https://github.com/raspberrypi/libpisp.
- Reconfigure/rebuild/install libcamera as you normally do.

naushir, 2024-07-02:
Any updates on this?  If not, I'll close shortly.

Jai-GAY, 2024-07-02:
No, I do not want to mess around and prefer the majority of users' usage patterns so I can get a solution to problems I face in the future journey.

naushir, 2024-07-02:
Ok, then the issue is that Canonical have not updated their packages for our latest libpisp release (https://github.com/raspberrypi/libpisp/releases/tag/v1.0.6).  Our libcamera release (https://github.com/raspberrypi/libcamera/releases/tag/v0.3.0%2Brpt20240617) relies on libpisp v1.0.6.

You will probably have to request Canonical to update the package, and in the meantime, use the older release of libcamera  (https://github.com/raspberrypi/libcamera/releases/tag/v0.2.0%2Brpt20240418).

Jai-GAY, 2024-07-03:
```
meson subprojects update
Progress: 0 / 5 (libyuv)
�[KProgress: 0 / 5 (libyuv, libpisp)
�[K
Updating libyuv...
  -> Not used.
Progress: 1 / 5 (libpisp)
�[KProgress: 1 / 5 (libpisp, nlohmann_json)
�[K
Updating nlohmann_json...
  -> Subproject has not changed, or the new source/patch needs to be extracted on the same location.
     Pass --reset option to delete directory and redownload.
Progress: 2 / 5 (libpisp)
�[KProgress: 2 / 5 (libpisp, libyaml)
�[KProgress: 2 / 5 (gtest, libpisp, libyaml)
�[K
Updating libyaml...
  -> Not used.
Progress: 3 / 5 (gtest, libpisp)
�[K
Updating gtest...
  -> Not used.
Progress: 4 / 5 (libpisp)
�[K
Updating libpisp...
  -> b567f04 (HEAD, tag: v1.0.6) build: Bump release version to 1.0.6 [Naushir Patuck]
Progress: 5 / 5 ()
```

naushir, 2024-07-03:
Have you tried a ``meson setup <builddir> --wipe`` and then rebuilt?

Jai-GAY, 2024-07-03:
> Have you tried a `meson setup <builddir> --wipe` and then rebuilt?

previous 0.64.0 does not work, meson 1.4.1 wipe works, but ninja install failed.

```
$ ninja -C build
ninja: Entering directory `build'
[1/224] Generating include/libcamera/internal/tp_header with a custom command
[2/224] Generating include/libcamera/ipa/core_mojom_module with a custom command

[223/224] Compiling C++ object src/py/libcamera/_libcamera.so.p/py_main.cpp.o
[224/224] Linking target src/py/libcamera/_libcamera.so

$ sudo ninja -C build install
ninja: Entering directory `build'
[1/18] Generating src/libcamera/version.cpp with a custom command
[2/18] Generating subprojects/libpisp/src/version.cpp with a custom command
[3/18] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.0.6.p/meson-generated_.._version.cpp.o
[4/18] Linking target subprojects/libpisp/src/libpisp.so.1.0.6
[5/18] Generating symbol file subprojects/libpisp/src/libpisp.so.1.0.6.p/libpisp.so.1.0.6.symbols
[6/18] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/meson-generated_.._version.cpp.o
[7/18] Linking target src/libcamera/libcamera.so.0.3.0
[8/18] Generating symbol file src/libcamera/libcamera.so.0.3.0.p/libcamera.so.0.3.0.symbols
[8/9] Installing files.

ERROR: Build directory has been generated with Meson version 1.4.1, which is incompatible with the current version 0.64.0.
FAILED: meson-internal__install
/home/pi/.local/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.
```

without sudo install got problem
```
$ ninja -C build install 2>&1 |tee -a ~/libcamera/libcamera_build.txt
ninja: Entering directory `build'
[1/18] Generating src/libcamera/version.cpp with a custom command
[2/18] Generating subprojects/libpisp/src/version.cpp with a custom command
[3/18] Compiling C++ object subprojects/libpisp/src/libpisp.so.1.0.6.p/meson-generated_.._version.cpp.o
[4/18] Linking target subprojects/libpisp/src/libpisp.so.1.0.6
[5/18] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/meson-generated_.._version.cpp.o
[6/18] Generating symbol file subprojects/libpisp/src/libpisp.so.1.0.6.p/libpisp.so.1.0.6.symbols
[7/18] Linking target src/libcamera/libcamera.so.0.3.0
[8/18] Generating symbol file src/libcamera/libcamera.so.0.3.0.p/libcamera.so.0.3.0.symbols
[8/9] Installing files.
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.10/site-packages/mesonbuild/mesonmain.py", line 186, in run
    return options.run_func(options)
  File "/home/pi/.local/lib/python3.10/site-packages/mesonbuild/minstall.py", line 863, in run
    installer.do_install(datafilename)
  File "/home/pi/.local/lib/python3.10/site-packages/mesonbuild/minstall.py", line 543, in do_install
    self.install_targets(d, dm, destdir, fullprefix)
  File "/home/pi/.local/lib/python3.10/site-packages/mesonbuild/minstall.py", line 740, in install_targets
    file_copied = self.do_copyfile(fname, outname, makedirs=(dm, outdir))
  File "/home/pi/.local/lib/python3.10/site-packages/mesonbuild/minstall.py", line 420, in do_copyfile
    self.copy2(from_file, to_file)
  File "/home/pi/.local/lib/python3.10/site-packages/mesonbuild/minstall.py", line 317, in copy2
    shutil.copy2(*args, **kwargs)
  File "/usr/lib/python3.10/shutil.py", line 434, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.10/shutil.py", line 256, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: '/usr/local/include/libcamera/libcamera/ipa/core_ipa_interface.h'
Installing include/libcamera/ipa/core_ipa_interface.h to /usr/local/include/libcamera/libcamera/ipa

ERROR: Unhandled python OSError. This is probably not a Meson bug, but an issue with your build environment.
FAILED: meson-internal__install
/home/pi/.local/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.
```

naushir, 2024-07-05:
As with https://github.com/raspberrypi/rpicam-apps/issues/692, I suggest you get in touch with Canonical support to get the latest libcamera 0.3 packages updated for Ubuntu on Raspberry Pi.

closed 2024-07-05 by naushir