raspberrypi/libcamera #153: All my pi cameras stopped working after latest raspbian update libcamera0.2->libcamera0.3

voidmain99, 2024-06-30:
I have 6 Raspberry Pis (Pi zero W and Pi zero 2 W) running bookworm. They have the Pi cameras installed (ribbon cable) and I use the "motion" software to autodetect motion and save images/videos. I usually update these devices every few weeks using "apt update; apt upgrade". When I updated them today they all stopped working (I know I should have updated one and tested before updating them all).  It doesn't seem to detect the camera. I suspect this could be either a libcamera issue with the new packages or an issue with the updated kernel. I'm not quite sure how to revert back to the older versions. Here's what I am seeing.

On the motion interface I get the gray image that says "UNABLE TO OPEN VIDEO DEVICE".

$ vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0

$ libcamera-hello list-cameras
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.

One thing I noticed is the libcamera stack got updated from libcamera0.2 to libcamera0.3. Here's an excerpt from my dpkg log where you can see the version that was installed and the version that was removed.

dpkg.log:2024-06-29 09:34:53 install libcamera0.3:armhf <none> 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:34:53 status half-installed libcamera0.3:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:34:54 status unpacked libcamera0.3:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:34:54 upgrade libcamera-ipa:armhf 0.2.0+rpt20240418-1 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:34:54 status half-configured libcamera-ipa:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:34:54 status unpacked libcamera-ipa:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:34:54 status half-installed libcamera-ipa:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:06 status unpacked libcamera-ipa:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:35:07 upgrade libcamera-tools:armhf 0.2.0+rpt20240418-1 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:35:07 status half-configured libcamera-tools:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:07 status unpacked libcamera-tools:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:07 status half-installed libcamera-tools:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:09 status unpacked libcamera-tools:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:35:10 upgrade libcamera-v4l2:armhf 0.2.0+rpt20240418-1 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:35:10 status half-configured libcamera-v4l2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:10 status unpacked libcamera-v4l2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:10 status half-installed libcamera-v4l2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:10 status unpacked libcamera-v4l2:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:35:11 upgrade python3-libcamera:armhf 0.2.0+rpt20240418-1 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:35:11 status half-configured python3-libcamera:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:15 status unpacked python3-libcamera:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:15 status half-installed python3-libcamera:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 09:35:16 status unpacked python3-libcamera:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:11 configure libcamera-ipa:armhf 0.3.0+rpt20240617-1 <none>
dpkg.log:2024-06-29 09:38:11 status unpacked libcamera-ipa:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:11 status half-configured libcamera-ipa:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:11 status installed libcamera-ipa:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:11 configure libcamera0.3:armhf 0.3.0+rpt20240617-1 <none>
dpkg.log:2024-06-29 09:38:11 status unpacked libcamera0.3:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:11 status half-configured libcamera0.3:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:11 status installed libcamera0.3:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:12 configure libcamera-v4l2:armhf 0.3.0+rpt20240617-1 <none>
dpkg.log:2024-06-29 09:38:12 status unpacked libcamera-v4l2:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:12 status half-configured libcamera-v4l2:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:12 status installed libcamera-v4l2:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:12 configure python3-libcamera:armhf 0.3.0+rpt20240617-1 <none>
dpkg.log:2024-06-29 09:38:12 status unpacked python3-libcamera:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:12 status half-configured python3-libcamera:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:17 status installed python3-libcamera:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:18 configure libcamera-tools:armhf 0.3.0+rpt20240617-1 <none>
dpkg.log:2024-06-29 09:38:18 status unpacked libcamera-tools:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:18 status half-configured libcamera-tools:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 09:38:18 status installed libcamera-tools:armhf 0.3.0+rpt20240617-1
dpkg.log:2024-06-29 10:07:34 status installed libcamera0.2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 10:07:40 remove libcamera0.2:armhf 0.2.0+rpt20240418-1 <none>
dpkg.log:2024-06-29 10:07:40 status half-configured libcamera0.2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 10:07:40 status half-installed libcamera0.2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 10:07:40 status config-files libcamera0.2:armhf 0.2.0+rpt20240418-1
dpkg.log:2024-06-29 10:07:40 status not-installed libcamera0.2:armhf <none>

Any pointers would be greatly appreciated!

naushir, 2024-10-07:
Is this still an issue?  If so, can you try updating all your software again ``sudo apt update && sudo apt full-upgrade -y`` and see if this fixes your problem?

voidmain99, 2024-10-10:
> Is this still an issue? If so, can you try updating all your software again `sudo apt update && sudo apt full-upgrade -y` and see if this fixes your problem?

This is no longer an issue. My problem was a bad config change that I pushed out to all my pis. For some reason I got locked out of github so I could not update the issue and close it. It's letting me in now.  Sorry!

closed 2024-10-11 by naushir