raspberrypi/libcamera #154: libcamerify cheese with libcamera's V4L2 compatibility layer.

Jai-GAY, 2024-07-02:
does anyone face the same issue when using libcamerify cheese on Pi4B Ubuntu 22.04 64bit?

In my experience, it ran once for a few seconds on Pi 4B then an error was logged, subsequent run failed.

[0:12:47.623382997] [2868] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:12:47.623461891] [2868] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request

Failed to allocate a buffer: ../sys/v4l2/gstv4l2src.c(1146): gst_v4l2src_create (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1

```
$ cat /boot/firmware/config.txt
camera_auto_detect=0
#start_x=1
dtoverlay=ov5647
```
```
$ sudo libcamerify cheese 
[sudo] password for pi: 
[0:12:37.083754826] [2784]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:12:37.174030864] [2785]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:37.180399833] [2785]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media1
[0:12:39.967233621] [2734]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:12:40.035550847] [2856]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:40.040479835] [2856]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media1
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.704: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.719: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.734: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.753: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.753: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:32.754: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:12:40.633513263] [2870]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.634882343] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:12:40.639818899] [2870]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.640634106] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.436: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.474: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:12:40.938770931] [2869]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.940419326] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:12:40.944527472] [2869]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:12:40.945345938] [2856]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.533: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.628: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.671: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:2734): GLib-GObject-CRITICAL **: 12:11:33.705: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:12:47.623382997] [2868] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:12:47.623461891] [2868] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
(cheese:2734): cheese-WARNING **: 12:11:40.319: Failed to allocate a buffer: ../sys/v4l2/gstv4l2src.c(1146): gst_v4l2src_create (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1
```

naushir, 2024-07-02:
Can I ask why you are running ``cheese``? The ov5647 is natively supported by ``libcamera`` and ``rpicam-apps``.  It would be much better to use in most cases.

Jai-GAY, 2024-07-03:
> Can I ask why you are running `cheese`?

I need to confirm it works for others for example [camera_ros](https://github.com/christianrauch/camera_ros).

naushir, 2024-07-22:
This is not actively being worked on, so closing.

closed 2024-07-22 by naushir

Jai-GAY, 2024-09-26:
finally, I got [the legacy camera working with Ubuntu 22.04 over OpenCV 4.x](https://www.youtube.com/watch?v=va7o7wzhEE4).