raspberrypi/libcamera #157: libcamera build error on rpi5

mmxuxp, 2024-07-16:
pi@raspberrypi:$ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

```
pi@raspberrypi:~/libcamera/libcamera-0.3.0-rpt20240617 $ sudo ninja -C build install
ninja: Entering directory `build'
[8/99] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o
FAILED: src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o 
c++ -Isrc/libcamera/libcamera.so.0.3.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/local/include/libpisp -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/pi/libcamera/libcamera-0.3.0-rpt20240617/build/config.h -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -MF src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o.d -o src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -c ../src/libcamera/pipeline/rpi/pisp/pisp.cpp
In file included from /usr/include/c++/12/bits/stl_pair.h:60,
                 from /usr/include/c++/12/bits/stl_algobase.h:64,
                 from /usr/include/c++/12/algorithm:60,
                 from ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:8:
/usr/include/c++/12/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = void]’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:26:   required from here
/usr/include/c++/12/type_traits:2614:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2614 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:33: error: template argument 2 is invalid
  742 |         SharedMemObject<FrontEnd> fe_;
      |                                 ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:743:32: error: template argument 2 is invalid
  743 |         SharedMemObject<BackEnd> be_;
      |                                ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual bool libcamera::PipelineHandlerPiSP::match(libcamera::DeviceEnumerator*)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:909:61: error: template argument 2 is invalid
  909 |                         pisp->fe_ = SharedMemObject<FrontEnd>
      |                                                             ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:911:60: error: template argument 2 is invalid
  911 |                         pisp->be_ = SharedMemObject<BackEnd>
      |                                                            ^
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:914:40: error: request for member ‘fd’ in ‘pisp->libcamera::PiSPCameraData::fe_’, which is of non-class type ‘int’
  914 |                         if (!pisp->fe_.fd().isValid() || !pisp->be_.fd().isValid()) {
      |                                        ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:914:69: error: request for member ‘fd’ in ‘pisp->libcamera::PiSPCameraData::be_’, which is of non-class type ‘int’
  914 |                         if (!pisp->fe_.fd().isValid() || !pisp->be_.fd().isValid()) {
      |                                                                     ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual int libcamera::PipelineHandlerPiSP::prepareBuffers(libcamera::Camera*)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1011:18: error: base operand of ‘->’ is not a pointer
 1011 |         data->be_->GetTdnOutputFormat(tdn);
      |                  ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual libcamera::CameraConfiguration::Status libcamera::PiSPCameraData::platformValidate(libcamera::RPi::RPiCameraConfiguration*) const’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1294:49: error: base operand of ‘->’ is not a pointer
 1294 |                                              be_->GetMaxDownscale());
      |                                                 ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual int libcamera::PiSPCameraData::platformConfigure(const libcamera::RPi::RPiCameraConfiguration*)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1500:68: error: base operand of ‘->’ is not a pointer
 1500 |                                                                 be_->GetMaxDownscale());
      |                                                                    ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1537:12: error: base operand of ‘->’ is not a pointer
 1537 |         be_->GetGlobal(global);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1540:12: error: base operand of ‘->’ is not a pointer
 1540 |         be_->SetGlobal(global);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureCfe()’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1843:38: error: invalid type argument of unary ‘*’ (have ‘int’)
 1843 |         std::scoped_lock<FrontEnd> l(*fe_);
      |                                      ^~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1846:12: error: base operand of ‘->’ is not a pointer
 1846 |         fe_->GetGlobal(global);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1865:20: error: base operand of ‘->’ is not a pointer
 1865 |                 fe_->SetCompress(0, compress);
      |                    ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1871:12: error: base operand of ‘->’ is not a pointer
 1871 |         fe_->SetGlobal(global);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1872:12: error: base operand of ‘->’ is not a pointer
 1872 |         fe_->SetInput(input);
[... 116 lines cut ...]
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2092:25: error: request for member ‘fd’ in ‘((libcamera::PiSPCameraData*)this)->libcamera::PiSPCameraData::be_’, which is of non-class type ‘int’
 2092 |         params.be = be_.fd();
      |                         ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘void libcamera::PiSPCameraData::prepareCfe()’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2170:46: error: invalid type argument of unary ‘*’ (have ‘int’)
 2170 |                 std::scoped_lock<FrontEnd> l(*fe_);
      |                                              ^~~~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2172:20: error: base operand of ‘->’ is not a pointer
 2172 |                 fe_->Prepare(reinterpret_cast<pisp_fe_config *>(configBuffer.data()));
      |                    ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘void libcamera::PiSPCameraData::prepareBe(uint32_t, bool)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2210:12: error: base operand of ‘->’ is not a pointer
 2210 |         be_->Prepare(configBuffer);
      |            ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2226:34: error: base operand of ‘->’ is not a pointer
 2226 |                         of << be_->GetJsonConfig(configBuffer);
      |                                  ^~
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: At global scope:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:668:14: error: ‘unsigned int libcamera::{anonymous}::calculateSwDownscale(const libcamera::V4L2DeviceFormat&, unsigned int, unsigned int)’ defined but not used [-Werror=unused-function]
  668 | unsigned int calculateSwDownscale(const V4L2DeviceFormat &format, unsigned int largestWidth,
      |              ^~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
[13/99] Compiling C++ object src/libcamera/proxy/worker/raspberrypi_ipa_proxy.p/meson-generated_.._raspberrypi_ipa_proxy_worker.cpp.o
ninja: build stopped: subcommand failed.
```

naushir, 2024-07-16:
You need to rebuild the latest version of [libpisp ](https://github.com/raspberrypi/libpisp) in order to get libcamera building and working correctly.

closed 2024-07-16 by naushir

mmxuxp, 2024-07-16:
@naushir Thanks for your quick response. But I encourage a new issue.  
```
[106/223] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o
FAILED: src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o 
c++ -Isrc/libcamera/libcamera.so.0.3.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/local/include/libpisp -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/pi/libcamera/backup/libcamera-0.3.0-rpt20240617/build/config.h -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -MF src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o.d -o src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -c ../src/libcamera/pipeline/rpi/pisp/pisp.cpp
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureCfe()’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1859:13: error: ‘PISP_IMAGE_FORMAT_compressed’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_COMPRESSE’?
 1859 |         if (PISP_IMAGE_FORMAT_compressed(image.format)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             PISP_IMAGE_FORMAT_COMPRESSED
../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureBe(const std::optional<libcamera::ColorSpace>&)’:
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1938:13: error: ‘PISP_IMAGE_FORMAT_compressed’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_COMPRESSE’?
 1938 |         if (PISP_IMAGE_FORMAT_compressed(inputFormat.format)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             PISP_IMAGE_FORMAT_COMPRESSED
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2002:13: error: ‘PISP_IMAGE_FORMAT_compressed’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_COMPRESSE’?
 2002 |         if (PISP_IMAGE_FORMAT_compressed(tdnFormat.format)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             PISP_IMAGE_FORMAT_COMPRESSED
../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2021:13: error: ‘PISP_IMAGE_FORMAT_compressed’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_COMPRESSE’?
 2021 |         if (PISP_IMAGE_FORMAT_compressed(stitchFormat.format)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             PISP_IMAGE_FORMAT_COMPRESSED
[111/223] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_vc4_vc4.cpp.o
ninja: build stopped: subcommand failed.
```

naushir, 2024-07-16:
This suggests you may not be using the ``next`` branch of libcamera (https://github.com/raspberrypi/libcamera/tree/next)?

The ``main`` branch is the sable release, and may not compile correctly (as in this case) with the latest changes in dependency repos.