raspberrypi/libcamera #159: [HOW TO] Access Metadata Libcamera-still

Bra1nsen, 2024-07-25:
Dear developer, sorry for opening this issue, could you please briefly describe how one can request metadata infos such as gain, and sensor temperature using app command libcamera-still?

e.g.:

```
    def capture_with_metadata(self, output_file):
        log_file = self.LOG_DIR / "capture.log"
        self.LOG_DIR.mkdir(parents=True, exist_ok=True)  # Ensure log directory exists
        result = subprocess.run(
            [
                "libcamera-still",
                "-o", str(output_file),
                "--tuning-file", "/usr/share/libcamera/ipa/rpi/pisp/imx477_noir.json",
                "--ev", "-2",
                "--post-process-file", "/home/pi/skycam/scripts/hdr.json",
                "--immediate",
                "--width", "3280",
                "--height", "2464",
                "-n",
                "--verbose", "3"
            ],
            capture_output=True,
            text=True
        )
```

How can I print metadata?

Thanks in advance!

naushir, 2024-07-26:
This might be possible with the ``--info-text`` command line argument.  What parameters are you looking to print out?

naushir, 2024-07-26:
Or you might be able to use the ``--metadata`` argument to save the metadata to a file.

Bra1nsen, 2024-07-26:
Hi naushir, with Picamera2 i can get the following metadata:

`{'SensorTimestamp': 691371990000, 'ExposureTime': 36309, 'FocusFoM': 536, 'AnalogueGain': 2.0, 'AeLocked': True, 'ColourCorrectionMatrix': (1.9748594760894775, -0.7156795263290405, -0.2591899633407593, -0.43473267555236816, 2.0987930297851562, -0.6640604138374329, -0.04917403683066368, -0.8441071510314941, 1.8932812213897705), 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'FrameDuration': 36642, 'Lux': 84.6844711303711, 'ColourTemperature': 2445, 'SensorTemperature': 30.0, 'DigitalGain': 1.000342607498169, 'ColourGains': (2.1123898029327393, 3.8502354621887207), 'ScalerCrop': (2, 0, 4052, 3040)}`

with 

```
#!/usr/bin/python3

import time
from picamera2 import Picamera2

# Initialize the camera
picam2 = Picamera2()

# Create and configure the preview configuration
preview_config = picam2.create_preview_configuration()
picam2.configure(preview_config)

# Start the camera
picam2.start()
time.sleep(2)

# Capture and print the metadata
print(picam2.capture_metadata())
```

the flaggs you recommend to not work for me:

```
(asi_venv) pi@raspberrypi:~/asi_driver $ libcamera-still -o /home/pi/skycam/images/image.jpg --metadata
ERROR: *** the required argument for option '--metadata' is missing ***
(asi_venv) pi@raspberrypi:~/asi_driver $ libcamera-still -o /home/pi/skycam/images/image.jpg --info-text
ERROR: *** the required argument for option '--info-text' is missing ***
```

Bra1nsen, 2024-07-26:
Ah okay I got it now with:

`pi@raspberrypi:~ $ libcamera-still -o /home/pi/skycam/images/image.jpg --metadata /home/pi/skycam/images/metadata.txt`

```
pi@raspberrypi:~ $ cat /home/pi/skycam/images/metadata.txt
{
    "ExposureTime": 36428,
    "SensorTimestamp": 868547887000,
    "ScalerCrop": "(0, 0)/4056x3040",
    "ColourCorrectionMatrix": [ 1.974859, -0.715680, -0.259190, -0.434733, 2.098793, -0.664060, -0.049174, -0.844107, 1.893281 ],
    "FocusFoM": 939,
    "ColourTemperature": 2445,
    "ColourGains": [ 2.112390, 3.850235 ],
    "DigitalGain": 1.000296,
    "AeLocked": true,
    "SensorTemperature": 33.000000,
    "Lux": 83.396866,
    "FrameDuration": 100000,
    "SensorBlackLevels": [ 4096, 4096, 4096, 4096 ],
    "AnalogueGain": 2.000000
}pi@raspberrypi:~ $ 
```

Thanks!

closed 2024-07-26 by Bra1nsen