raspberrypi/libcamera #160: strange behavior with library version > 0.1.0

lowPerformer, 2024-07-31:
Hi everyone,

I am programming a Qt application that is based on the libcamera qcam application and I see a strange behavior as soon as I use a version of the library greater than 0.1.0:

For photo capture I increase the resolution of the streams just before the capture. To do this, I stop the camera, set the new parameters and start the camera again. This works very well with version 0.5.0 on bullseye, for example. 
However, as soon as I use a newer version greater than 0.1.0 (f.i. 0.3.0), the app starts correctly, the **first snapshot works**, but as soon as I start another one, I get the following error message:

```
[0:57:28.911464239] [1215]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x400-NV12 (1) 2304x1296-XBGR8888
[0:57:28.919905128] [1231] ERROR Camera camera.cpp:587 Removing camera while still in use
```

The code never returns from Camera::configure() in my applications startCapture() and seems to crash somewhere in libcameras camera.cpp.

Someone any ideas? I don't know where to dig any further!

Niklas

My Setup:

- Raspberry CM4
- Raspberry CAM 3.0
- Qt 5.15, cross-compile setup
- PiOS Bullseye libcamera stock 0.5.0 or home-baked 0.3.0. But same problems on Bookworm with stock libcamera 0.3.0.

lowPerformer, 2024-07-31:
Hi, 
the CameraManager is only initialized once at startup.

I do this for switching to "full resolution" mode and back to "normal" mode instead:

1)
camera->stop();
camera->requestCompleted.disconnect(this);
... clearing buffer, queues, deleting FrameBufferAllocator ...

2)
Setting new stream sizes in CameraConfiguration

3)
CameraConfiguration::validate()
Camera::configure() where code never returns if you take more than one snapshot.
... preparing buffer, queues, FrameBufferAllocator ...

This is a part of the logfile where the app successfully switches back from full resolution to normal resolution ((0) 640x400-NV12 (1) 2304x1296-XBGR8888) after first snapshot:

```
[0:19:18.541454067] [2142] DEBUG Camera camera.cpp:1385 Stopping capture
[0:19:18.558077901] [2142] DEBUG Camera camera.cpp:1118 streams configuration: (0) 800x600-XRGB8888 (1) 4608x2592-YUV420
[0:19:18.558629864] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 1536x864 fmt SRGGB10 Score: 3400 (best 3400)
[0:19:18.558695419] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 2304x1296 fmt SRGGB10 Score: 1000 (best 1000)
[0:19:18.558733641] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 4608x2592 fmt SRGGB10 Score: 1900 (best 1000)
[0:19:18.558776345] [2142] DEBUG RPI pipeline_base.cpp:284 Try color space sYCC
[0:19:18.558823660] [2142] DEBUG RPI pipeline_base.cpp:339 Color space changed from sRGB to sRGB
[0:19:18.558854790] [2142] DEBUG RPI pipeline_base.cpp:284 Try color space sYCC
[0:19:18.559142178] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 1536x864 fmt SRGGB10 Score: 3400 (best 3400)
[0:19:18.559197216] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 2304x1296 fmt SRGGB10 Score: 1000 (best 1000)
[0:19:18.559234290] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 4608x2592 fmt SRGGB10 Score: 1900 (best 1000)
[0:19:18.559275012] [2142] DEBUG RPI pipeline_base.cpp:284 Try color space sRGB
[0:19:18.559315641] [2142] DEBUG RPI pipeline_base.cpp:284 Try color space sYCC
[0:19:18.559386475] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 1536x864 fmt SRGGB10 Score: 3400 (best 3400)
[0:19:18.559424475] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 2304x1296 fmt SRGGB10 Score: 1000 (best 1000)
[0:19:18.559459882] [2142] DEBUG RPI pipeline_base.cpp:957 Format: 4608x2592 fmt SRGGB10 Score: 1900 (best 1000)
[0:19:18.559497530] [2142] DEBUG RPI pipeline_base.cpp:284 Try color space sRGB
[0:19:18.559531604] [2142] DEBUG RPI pipeline_base.cpp:284 Try color space sYCC
[0:19:18.559578178] [2142]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x400-NV12 (1) 2304x1296-XBGR8888
[0:19:18.559749790] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video0[35:cap]: Releasing buffers
[0:19:18.559818549] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video0[35:cap]: 0 buffers requested.
[0:19:18.564224734] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video1[36:cap]: Releasing buffers
[0:19:18.564418512] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video1[36:cap]: 0 buffers requested.
[0:19:18.564686678] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video13[37:out]: Releasing buffers
[0:19:18.564806438] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video13[37:out]: 0 buffers requested.
[0:19:18.564871512] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video14[38:cap]: Releasing buffers
[0:19:18.565033919] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video14[38:cap]: 0 buffers requested.
[0:19:18.571589104] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video15[39:cap]: Releasing buffers
[0:19:18.571745160] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video15[39:cap]: 0 buffers requested.
[0:19:18.571875364] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video16[40:cap]: Releasing buffers
[0:19:18.571980771] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video16[40:cap]: 0 buffers requested.
[0:19:18.572614160] [2140]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@0/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[0:19:18.572694308] [2140] DEBUG RPI vc4.cpp:636 Setting ISP Output0 to 2304x1296-XB24
[0:19:18.572951678] [2140] DEBUG RPI vc4.cpp:643 Stream ISP Output0 has color space sRGB
[0:19:18.573013790] [2140] DEBUG RPI vc4.cpp:636 Setting ISP Output1 to 640x400-NV12
[0:19:18.573255234] [2140] DEBUG RPI vc4.cpp:643 Stream ISP Output1 has color space sYCC
[0:19:18.573506234] [2140] DEBUG RPI vc4.cpp:744 Setting embedded data format 0x0-SENS
[0:19:18.575216049] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video15[39:cap]: 1 buffers requested.
[0:19:18.575348215] [2140] DEBUG Buffer framebuffer.cpp:346 Buffer is contiguous
[0:19:18.575420938] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video15[39:cap]: 0 buffers requested.
[0:19:18.584367123] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video14[38:cap]: 1 buffers requested.
[0:19:18.584503012] [2140] DEBUG Buffer framebuffer.cpp:346 Buffer is contiguous
[0:19:18.584547067] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video14[38:cap]: 0 buffers requested.
[0:19:18.584872215] [2142] DEBUG Request request.cpp:361 Created request - cookie: 0
[0:19:18.584928623] [2142] DEBUG Camera camera.cpp:1341 Starting capture
```

If I do another snapshot, code never returns from Camera::configure() when switching back to normal resolution:

[... 9 lines cut ...]
[0:19:19.921349659] [2142] DEBUG RPI pipeline_base.cpp:339 Color space changed from sRGB to sRGB
[... cut ...]
[0:19:19.930570455] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video15[39:cap]: Releasing buffers
[0:19:19.930796937] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video15[39:cap]: 0 buffers requested.
[0:19:19.930899215] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video16[40:cap]: Releasing buffers
[0:19:19.930978622] [2140] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video16[40:cap]: 0 buffers requested.
[0:19:19.932235011] [2140] ERROR Camera camera.cpp:587 Removing camera while still in use
```

ShevaDavid, 2024-08-16:
any updates on this issue?

lowPerformer, 2024-09-10:
Since release 0.3.1, issue seems to be gone!

naushir, 2024-09-10:
Good news.  Feel free to close this once you have fully tested.

closed 2024-09-10 by lowPerformer