raspberrypi/libcamera #162: Raspberry PI Zero 2W Bookworm

magni-optitog, 2024-08-07:
Does libcamera support Raspberry PI Zero 2W running Bookworm? I have followed the conventional guides and cannot get the camera to appear when running `libcamera-vid --list-cameras`. Any advice is welcome.

naushir, 2024-08-07:
Yes it ought to work fine.  Please follow the instruction here:
https://forums.raspberrypi.com/viewtopic.php?t=362707

magni-optitog, 2024-08-07:
I followed `How to check if an official Raspberry Pi camera is working` and the raspberry pi still outputs that no camera is detected. The same is true of `rpicam-hello`, I am using an OV2311 camera https://docs.arducam.com/Raspberry-Pi-Camera/Pivariety-Camera/2MP-OV2311/. I have no idea where to look for inspiration although I have tried many things. The last step I took was to do a clean installation and see if the camera is detected, to no avail.

naushir, 2024-08-07:
As per the following post:
https://forums.raspberrypi.com/viewtopic.php?t=372520

please get in touch with Arducam Support for help on their products.

closed 2024-08-07 by naushir