raspberrypi/libcamera #165: Error compiling version 0.3.0

nickoppen, 2024-08-18:
I'm trying to compile version 0.3.0 on a pi5 running

> PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
> NAME="Debian GNU/Linux"
> VERSION_ID="12"
> VERSION="12 (bookworm)"
> VERSION_CODENAME=bookworm
 
I've downloaded the prerequisites and tried with the packaged release and the source code downloaded using **git clone https://github.com/raspberrypi/libcamera.git**

For both the output from **ninja -C build install** is:

> Found ninja-1.11.1 at /usr/bin/ninja
> ninja: Entering directory `build'
> [19/211] Generating src/ipa-priv-key with a custom command
> .............+.....+.+......+...+.....+....+............+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*........+.........+..+............+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*............+......+.+...+...+............+.....+............+...+...+...+.+...+...........+......+....+.....+......+....+.........+........+...+.+........+.+..+.......+......+......+.....+...+.+..+.......+...+..+...+....+..................+...+...+......+........+......+.+......+.....+.........+.........+.......+...+......+.................................+..+.+.....+.......+........+...................+...+..+.........+.+..............+.......+..+......+.......+...+..+.......+..+.+..............................+...+.....+..........+......+......+..+...+...+....+........+..................+.......+.....+....+........................+.....+.+...+.................+..................+..........+..+.+...+.....+......+......+.......+......+..+....+...+.................+...+.+.....+.........+...............+...+...+.+...+...+...+.....+....+...+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
> ........+.......+..+.+......+.....+...+...+.......+.....+.+..+...+.+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*...........+.+.........+.....+.......+..+..................+.......+...+..+.+......+...+.................+......+...+...................+..+......+.........+.+..+...+....+........+...............+...+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*.................+...+....+...+...............+......+.........+.................+...+......+......+......+.......+..+....+..+...+....+..+..........+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
> [45/211] Generating src/libcamera/ipa_pub_key_cpp with a custom command
> writing RSA key
> [110/211] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o
> FAILED: src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o 
> c++ -Isrc/libcamera/libcamera.so.0.3.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -I../subprojects/libyaml/include -Isubprojects/libyaml/__CMake_build/include -I../subprojects/libyaml/__CMake_build/include -Isubprojects/libyaml/__CMake_build -I../subprojects/libyaml/__CMake_build -Isubprojects/libyaml -I../subprojects/libyaml -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/local/include/libpisp -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/user0/Downloads/libcamera-0.3.0-rpt20240617/build/config.h -fPIC -DYAML_DECLARE_STATIC -D_CRT_SECURE_NO_WARNINGS -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -MF src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o.d -o src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -c ../src/libcamera/pipeline/rpi/pisp/pisp.cpp
> In file included from /usr/include/c++/12/bits/stl_pair.h:60,
>                  from /usr/include/c++/12/bits/stl_algobase.h:64,
>                  from /usr/include/c++/12/algorithm:60,
>                  from ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:8:
> /usr/include/c++/12/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = void]’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:26:   required from here
> /usr/include/c++/12/type_traits:2614:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
>  2614 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
>       |           ^~~~~~~~~~~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:742:33: error: template argument 2 is invalid
>   742 |         SharedMemObject<FrontEnd> fe_;
>       |                                 ^
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:743:32: error: template argument 2 is invalid
>   743 |         SharedMemObject<BackEnd> be_;
>       |                                ^
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual bool libcamera::PipelineHandlerPiSP::match(libcamera::DeviceEnumerator*)’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:909:61: error: template argument 2 is invalid
>   909 |                         pisp->fe_ = SharedMemObject<FrontEnd>
>       |                                                             ^
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:911:60: error: template argument 2 is invalid
>   911 |                         pisp->be_ = SharedMemObject<BackEnd>
>       |                                                            ^
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:914:40: error: request for member ‘fd’ in ‘pisp->libcamera::PiSPCameraData::fe_’, which is of non-class type ‘int’
>   914 |                         if (!pisp->fe_.fd().isValid() || !pisp->be_.fd().isValid()) {
>       |                                        ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:914:69: error: request for member ‘fd’ in ‘pisp->libcamera::PiSPCameraData::be_’, which is of non-class type ‘int’
>   914 |                         if (!pisp->fe_.fd().isValid() || !pisp->be_.fd().isValid()) {
>       |                                                                     ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual int libcamera::PipelineHandlerPiSP::prepareBuffers(libcamera::Camera*)’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1011:18: error: base operand of ‘->’ is not a pointer
>  1011 |         data->be_->GetTdnOutputFormat(tdn);
>       |                  ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual libcamera::CameraConfiguration::Status libcamera::PiSPCameraData::platformValidate(libcamera::RPi::RPiCameraConfiguration*) const’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1294:49: error: base operand of ‘->’ is not a pointer
>  1294 |                                              be_->GetMaxDownscale());
>       |                                                 ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘virtual int libcamera::PiSPCameraData::platformConfigure(const libcamera::RPi::RPiCameraConfiguration*)’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1500:68: error: base operand of ‘->’ is not a pointer
>  1500 |                                                                 be_->GetMaxDownscale());
>       |                                                                    ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1537:12: error: base operand of ‘->’ is not a pointer
>  1537 |         be_->GetGlobal(global);
>       |            ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1540:12: error: base operand of ‘->’ is not a pointer
>  1540 |         be_->SetGlobal(global);
>       |            ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureCfe()’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1843:38: error: invalid type argument of unary ‘*’ (have ‘int’)
[... 128 lines cut ...]
>       |                         ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2092:25: error: request for member ‘fd’ in ‘((libcamera::PiSPCameraData*)this)->libcamera::PiSPCameraData::be_’, which is of non-class type ‘int’
>  2092 |         params.be = be_.fd();
>       |                         ^~
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘void libcamera::PiSPCameraData::prepareCfe()’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:2170:46: error: invalid type argument of unary ‘*’ (have ‘int’)
>  2170 |                 std::scoped_lock<FrontEnd> l(*fe_);
>       |                                              ^~~~
[... cut ...]
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:668:14: error: ‘unsigned int libcamera::{anonymous}::calculateSwDownscale(const libcamera::V4L2DeviceFormat&, unsigned int, unsigned int)’ defined but not used [-Werror=unused-function]
>   668 | unsigned int calculateSwDownscale(const V4L2DeviceFormat &format, unsigned int largestWidth,
>       |              ^~~~~~~~~~~~~~~~~~~~
> cc1plus: all warnings being treated as errors
> [115/211] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_vc4_vc4.cpp.o
> ninja: build stopped: subcommand failed.

davidplowman, 2024-08-19:
Hi, I've just given this a go. I checked out libcamera from the Raspberry Pi fork (this repo), followed the instructions on the README.md page (replacing `libqt5widgets` by `libqt5widgets5`... oops!) and it seemed to work OK.

Ah, perhaps you're not building the `main` branch? I tried the `next` branch as well and indeed that didn't compile. Though I didn't get the same errors as you, so don't really understand what's going on. Anyway, in my case I also needed to check out https://github.com/raspberrypi/libpisp.git and build that (`meson setup build`, `ninja -C build` and `sudo ninja -C build install`), and that seems to have fixed things for me.

Perhaps you could try that? If that's still not working, maybe let us know exactly what code you're building and whether there are any other differences in your environment, and we could try that too.

nickoppen, 2024-08-20:
Thanks David,

I check that I had libqt5widgets5 and that was good.

I downloaded the latest libpisp from the link you provided and that went in without problems.

Getting back to libcamera I cloned it using **gh repo clone raspberrypi/libcamera** which, I assume clones the master branch.

Using **ninja -C build** I got a different error:

> user0@pi5:~/projects/libcamera $ ninja -C build
> ninja: Entering directory `build'
> [19/210] Generating src/ipa-priv-key with a custom command
> .....+....+......+...+.....+.+..+....+.....+..........+.........+..+..................+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*....+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..+................+..+.......+...........+...+............+.............+..+.......+........+....+...+..+...+......................+...+..+...+......+.......+...+..+.........+.+..+...+............+......+.........+.+.....+...............+.......+...+..+...+....+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
> ......+..+..........+.....+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..+...+.......+......+.....+............+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*.......+.+.........+.....+......+....+........+.+..+...+......+.+..+............+.+..+..........+........+....+..............+.......+.........+...+..+...+...............+......+......+....+.....................+...+.....+......+...................+..+.+...+......+...+.....+...+...+.........+.+.....+....+...+..............+.+............+...+..+......+......+.+...+.................+.........................+..+......+....+............+.....+.+..+.+.....+.+.....+.+.........+...+......+.........+...+.....+.............+..+.......+...+............+..+.........+.+........+.+..+...+......+......+....+.....+......+.......+.....+.+..+......................+...+..+...+.......+...........+.+.....+.+..+.............+.........+...+...+...+...........+.............+..+..........+........+.......+...+...+........+......+.............+..+...+...+....+......+...............+..+.+..+.......+...............+...+.........+...+...........+......+.+...+.....+.........+.............+..............+....+..+....+.....+.........+.+........+......+.+.....+.+.........+........+......+.+...+.................+.+...+..+...+....+.....+...+..................+....+...+.....+...+.+.....+...+.+......+...........+...+.+...........+...+......+......+.......+...+.........+...+......+..+....+......+.....+.......+............+.....+......+...+....+...+..+............+.+...+..+......................+.....+.......+...+....................+.+..............+...+............+.......+...+...+.....+...+......+..........+..................+............+......+...+...........+......+.........+...+..........+..+...............+.+.........+...+.....+......+.+...........+....+........+.......+.....+........................+....+..+.+...+.....+.......+.................+.............+..+......+.+...+.................+.+.....+...+....+.....+...........................+...+.+......+...+......+..+....+..............+......+.+..+................+..+.........+.+.....+....+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
> [47/210] Generating src/libcamera/ipa_pub_key_cpp with a custom command
> writing RSA key
> [109/210] Compiling C++ object src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o
> FAILED: src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o 
> c++ -Isrc/libcamera/libcamera.so.0.3.0.p -Isrc/libcamera -I../src/libcamera -Iinclude -I../include -I../subprojects/libyaml/include -Isubprojects/libyaml/__CMake_build/include -I../subprojects/libyaml/__CMake_build/include -Isubprojects/libyaml/__CMake_build -I../subprojects/libyaml/__CMake_build -Isubprojects/libyaml -I../subprojects/libyaml -Iinclude/libcamera -Iinclude/libcamera/ipa -Iinclude/libcamera/internal -Isrc/libcamera/proxy -I/usr/local/include/libpisp -I/usr/include/p11-kit-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/user0/projects/libcamera/build/config.h -fPIC -DYAML_DECLARE_STATIC -D_CRT_SECURE_NO_WARNINGS -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -MF src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o.d -o src/libcamera/libcamera.so.0.3.0.p/pipeline_rpi_pisp_pisp.cpp.o -c ../src/libcamera/pipeline/rpi/pisp/pisp.cpp
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureCfe()’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1859:13: error: ‘PISP_IMAGE_FORMAT_compressed’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_COMPRESSED’?
>  1859 |         if (PISP_IMAGE_FORMAT_compressed(image.format)) {
>       |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |             PISP_IMAGE_FORMAT_COMPRESSED
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp: In member function ‘int libcamera::PiSPCameraData::configureBe(const std::optional<libcamera::ColorSpace>&)’:
> ../src/libcamera/pipeline/rpi/pisp/pisp.cpp:1938:13: error: ‘PISP_IMAGE_FORMAT_compressed’ was not declared in this scope; did you mean ‘PISP_IMAGE_FORMAT_COMPRESSED’?
>  1938 |         if (PISP_IMAGE_FORMAT_compressed(inputFormat.format)) {
>       |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |             PISP_IMAGE_FORMAT_COMPRESSED
[... cut ...]
I'm upgrading to 0.3.0 based on naushir's forum post [https://forums.raspberrypi.com/viewtopic.php?t=374993](url). That post was about the 16:9 image ending up being a stretched version of the 4:3 image given that the cropping parameters were shared. I noticed the same issue with setting the controls for different requests. I want to take three images of the same scene changing the ExposureValue control (-1, 0, 1) and I find that they all end up with the same exposure value. I thought that 0.3.0 may have fixed that as well and that I should give it a try before posting another issue.

nickoppen, 2024-08-20:
Given that the errors reported in my last post had a suggested solution I thought I'd give it a try. I replaced _compressed with _COMPRESSED and ran ninja -C build again.

(I also ran sudo apt update and sudo apt upgrade and they both showed that my system is up-to-date)

This produced the following error:

> user0@pi5:~/projects/libcamera $ sudo ninja -C build install
> ninja: Entering directory `build'
> [67/100] Compiling C++ object src/ipa/rpi/pisp/ipa_rpi_pisp.so.p/pisp.cpp.o
> FAILED: src/ipa/rpi/pisp/ipa_rpi_pisp.so.p/pisp.cpp.o 
> c++ -Isrc/ipa/rpi/pisp/ipa_rpi_pisp.so.p -Isrc/ipa/rpi/pisp -I../src/ipa/rpi/pisp -Iinclude -I../include -Isrc/ipa -I../src/ipa -Isrc/ipa/rpi -I../src/ipa/rpi -Iinclude/libcamera/ipa -Iinclude/libcamera -I/usr/local/include/libpisp -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wshadow -include /home/user0/projects/libcamera/build/config.h -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/ipa/rpi/pisp/ipa_rpi_pisp.so.p/pisp.cpp.o -MF src/ipa/rpi/pisp/ipa_rpi_pisp.so.p/pisp.cpp.o.d -o src/ipa/rpi/pisp/ipa_rpi_pisp.so.p/pisp.cpp.o -c ../src/ipa/rpi/pisp/pisp.cpp
> ../src/ipa/rpi/pisp/pisp.cpp: In member function ‘void libcamera::ipa::RPi::IpaPiSP::applyContrast(const ContrastStatus*, pisp_be_global_config&)’:
> ../src/ipa/rpi/pisp/pisp.cpp:569:60: error: taking address of packed member of ‘pisp_be_gamma_config’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
>   569 |         if (!generateLut(contrastStatus->gammaCurve, gamma.lut, PISP_BE_GAMMA_LUT_SIZE)) {
>       |                                                      ~~~~~~^~~
> ../src/ipa/rpi/pisp/pisp.cpp: In member function ‘void libcamera::ipa::RPi::IpaPiSP::applyLensShading(const AlscStatus*, pisp_be_global_config&)’:
> ../src/ipa/rpi/pisp/pisp.cpp:642:24: error: taking address of packed member of ‘pisp_be_lsc_config’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
>   642 |         packLscLut(lsc.lut_packed, rgb);
>       |                    ~~~~^~~~~~~~~~
> ../src/ipa/rpi/pisp/pisp.cpp: In member function ‘void libcamera::ipa::RPi::IpaPiSP::applyTonemap(const TonemapStatus*, pisp_be_global_config&)’:
> ../src/ipa/rpi/pisp/pisp.cpp:877:58: error: taking address of packed member of ‘pisp_be_tonemap_config’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
>   877 |         if (!generateLut(tonemapStatus->tonemap, tonemap.lut, PISP_BE_TONEMAP_LUT_SIZE)) {
>       |                                                  ~~~~~~~~^~~
> cc1plus: all warnings being treated as errors
> [72/100] Compiling C++ object src/gstreamer/libgstlibcamera.so.p/gstlibcameraallocator.cpp.o
> ninja: build stopped: subcommand failed.
> 

I've got no idea what this is about.

davidplowman, 2024-08-20:
I think that `PISP_IMAGE_FORMAT_COMPRESSED` error is what I was getting before I updated libpisp. I found that libcamera took some "persuading" to work after rebuilding and installing libpisp, I think I had to blow away my build directory and start over. Don't know why.

Anyway, for reference, my working setup here has:

libpisp at git version 61d651aef38b5bd263164cd18e1a0f46521184f8 (top of the main branch)

libcamera at version 08eb6758129788cb7e33cc1726aea40c672201b8 (top of the next branch)

nickoppen, 2024-08-20:
Hey! then "next" branch compiled without a hitch.  Thanks.

closed 2024-08-20 by nickoppen

pbaetens, 2024-09-03:
having the same problem at the moment.
commit [48fe316](https://github.com/raspberrypi/libcamera/commit/48fe316f38912ec800b73553e3980e7c3fb40409)
works fine on my pi4 but fails to compile on pi5.
seeing the same error as you @davidplowman PISP_IMAGE_FORMAT_COMPRESSED