raspberrypi/libcamera #170: python bindings when building from scratch

pbaetens, 2024-08-30:
Hi all,

situation:

I build latest libcam and rpicam-apps from scratch.
commands like rpicam-hello work fine.
However, the python binding is not working.

```
python -m venv --system-site-packages venv
source venv/bin/activate
python
>>> import libcamera
>>> ModuleNotFoundError

```
Solution:
create a symbolic link to a place where python can find the .so file
`sudo ln -sf /usr/local/lib/aarch64-linux-gnu/python3.11/site-packages/libcamera /usr/local/lib/python3.11/dist-packages/libcamera
`

OS: Bookworm

naushir, 2024-09-02:
@pbaetens thanks for the info.  I must admit, python and venv is not something I am very familiar with, so I cannot say if there is a better way than what you described.

@davidplowman @XECDesign is this a solution to the venv issue we were talking about last week?

pbaetens, 2024-09-02:
seems related to:
https://patchwork.libcamera.org/patch/19186/

kbingham, 2024-09-02:
I've hit this a few times and I just set the python path which is another "work around". But I would love to see this resolved more globally too.

@will-v-pi there are comments from Tomi for you. Do you have time to reply to the patch please?

will-v-pi, 2024-09-02:
@kbingham I think the decision on that patch was not to merge it, as the behaviour that I was looking for wasn't cleanly compatible with cross-compilation? The patch was getting a bit complicated, and wasn't actually required for my `rpi-libcamera` pip package to compile. Tomi also mentioned that it would still install to an incorrect path on his PC, as that patch would install everything to `site-packages` whereas his PC only searched for `dist-packages`, so it's possibly not the correct solution to this problem?

I have a small downstream patch in [raspberrypi/pylibcamera](https://github.com/raspberrypi/pylibcamera) which I'm using to build the `rpi-libcamera` pip package, but that's specifically for building just the python bindings within that repository, rather than the whole of libcamera.

naushir, 2024-10-07:
From the above comments, I don't think there's anything we are able to do here...

closed 2024-10-07 by naushir