raspberrypi/libcamera #171: RPI4 - munmap_chunk(): invalid pointer

officialluq, 2024-08-31:
Hello Everyone, while using libcamera i have encountered one issue: munmap_chunk(): invalid pointer

[0:00:28.466082389] [406]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+dirty (2024-08-31T15:09:42+00:00)
[0:00:28.505386889] [408]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:00:28.507712259] [408]  INFO RPI vc4.cpp:400 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
[0:00:28.511571537] [411]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x1080-NV21
New clock: GstSystemClock
[0:00:28.514059556] [408]  INFO RPI vc4.cpp:569 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
munmap_chunk(): invalid pointer

Do you know that issue? I tried 2 raspberrypi4 with 2 different cameras supported by libcamera - IMX219 and ov5647
I compiled 3.1, 3.0, 2.0 versions but the issue still exist :(

officialluq, 2024-08-31:
strace: 
mmap(NULL, 8454144, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0x7fb01b4000
mprotect(0x7fb01c4000, 8388608, PROT_READ|PROT_WRITE) = 0
rt_sigprocmask(SIG_BLOCK, ~[], [], 8)   = 0
clone3({flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, child_tid=0x7fb09c31d0, parent_tid=0x7fb09c31d0, exit_signal=0, stack=0x7fb01b4000, stack_size=0x80e9c0, tls=0x7fb09c3840} => {parent_tid=[436]}, 88) = 436
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
write(8, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x7fee9b7698, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANYmunmap_chunk(): invalid pointer
) = ?
+++ killed by SIGABRT (core dumped) +++
Aborted (core dumped)

officialluq, 2024-08-31:
Do you have any hints how to debug this?

Thanks in advance

officialluq, 2024-08-31:
DEBUG LOGS FROM libcamera: 
[0:11:00.664134224] [460] DEBUG RPiAgc agc_channel.cpp:988 Output written, total exposure requested is 0.00us
[0:11:00.664151520] [460] DEBUG RPiAgc agc_channel.cpp:990 Camera exposure update: shutter time 1000.00us analogue gain 1
[0:11:00.664175428] [460] DEBUG RPiAlsc alsc.cpp:335 AWB results found, using 4000
munmap_chunk(): invalid pointer
Aborted (core dumped)

officialluq, 2024-09-01:
> Hello Everyone, while using libcamera i have encountered one issue: munmap_chunk(): invalid pointer
> 
> Kernel: Linux raspberrypi4-64 6.6.22-v8
> 
> [0:00:28.466082389] [406] INFO Camera camera_manager.cpp:313 libcamera v0.3.0+dirty (2024-08-31T15:09:42+00:00) [0:00:28.505386889] [408] WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise [0:00:28.507712259] [408] INFO RPI vc4.cpp:400 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2 Pipeline is live and does not need PREROLL ... Pipeline is PREROLLED ... Setting pipeline to PLAYING ... [0:00:28.511571537] [411] INFO Camera camera.cpp:1183 configuring streams: (0) 1280x1080-NV21 New clock: GstSystemClock [0:00:28.514059556] [408] INFO RPI vc4.cpp:569 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA munmap_chunk(): invalid pointer
> 
> Do you know that issue? I tried 2 raspberrypi4 with 2 different cameras supported by libcamera - IMX219 and ov5647 I compiled 3.1, 3.0, 2.0 versions but the issue still exist :(

It seems that 1.0 version with PIPELINE set to raspberrypi works as expected.

naushir, 2024-09-02:
@officialluq this is not something I've seen myself - nor has been reported by others.  Have you made any sort of modifications to the libcamera source tree?  I see the ``+dirty`` tag on the version string.

Are you able to run under ``gdb`` and get a callstack when thing go wrong?

officialluq, 2024-09-02:
I didn't make any modifications, i just build it from source using yocto(2 attempts - kirkstone & scarthgap). I think additional patches from bitbake recipe are making this tag on the version string. 

I will try to run it under gdb.

alitofresko, 2024-09-26:
I encountered same issue with libcamera while running the following command:
`gdb --args cam -c 1 --capture=10 --file`
The application crashes with a SIGABRT signal, and GDB reports the following error:
`munmap_chunk(): invalid pointer
Thread 5 "cam" received signal SIGABRT, Aborted.`
Here is the complete log from GDB:

```
Starting program: /usr/bin/cam -c 1 --capture=10 --file
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
[12:40:26.582115748] [759]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+dirty (2024-09-25T09:49:55+00:00)
[New Thread 0xb67a83c0 (LWP 761)]
[12:40:26.642895442] [761]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[New Thread 0xb5dff3c0 (LWP 762)]
[New Thread 0xb55fe3c0 (LWP 763)]
[12:40:26.646293579] [761]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
Using camera /base/soc/i2c0mux/i2c@1/ov5647@36 as cam0
[12:40:26.646935444] [759]  INFO Camera camera.cpp:1183 configuring streams: (0) 800x600-XRGB8888
[12:40:26.648187414] [761]  INFO RPI vc4.cpp:559 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[New Thread 0xb46a93c0 (LWP 764)]
munmap_chunk(): invalid pointer
Thread 5 "cam" received signal SIGABRT, Aborted.
[Switching to Thread 0xb46a93c0 (LWP 764)]
0xb6c28a66 in ?? () from /lib/libc.so.6

```
Any advice on how to resolve this issue or further debug it would be greatly appreciated.
Thank you!

naushir, 2024-10-07:
Any updates on this issue?  Does it still happen with the latest libcamera packages?

closed 2024-10-07 by naushir

alitofresko, 2024-10-08:
In my opinion, the issue still affects version 0.2.0, but it has already been resolved in version 0.3.0. Unfortunately, Yocto (scarthgap) is still pointing to version 0.2.0, which might be a problem. However, in this case, an issue should be opened on the meta-openembedded repository.