raspberrypi/libcamera #173: Support for SRGGB8_1x8

Demon000, 2024-09-05:
Hi. I'm adding support for a new camera sensor, the OV5640. Among other issues which I've handled (skipping unsupported/non-bayer formats), I seem to be stuck.

This is the output of running rpicam-hello.
```
[0:04:53.437493699] [2225]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/ov5640@3c - Selected sensor format: 1920x1080-SRGGB8_1X8 - Selected CFE format: 1920x1080-BYR2
[0:04:53.459385633] [2225] ERROR V4L2 v4l2_videodevice.cpp:1995 /dev/video4[16:cap]: Failed to start streaming: Invalid argument
```

Kernel says
```
[  293.241517] rp1-cfe 1f00128000.csi: FOURCC format: 844257602
[  293.241526] rp1-cfe 1f00128000.csi: Failed to start media pipeline: -22
```

Coming from the link validation function:
```
		for (i = 0; i < ARRAY_SIZE(formats); i++) {
			if (formats[i].code == source_fmt->code &&
			    formats[i].fourcc == pix_fmt->pixelformat) {
				fmt = &formats[i];
				break;
			}
		}
		if (!fmt) {
			cfe_err("FOURCC format: %d\n", pix_fmt->pixelformat);
			ret = -EINVAL;
			goto out;
		}
```

Is there a known fix for this?

Demon000, 2024-09-05:
[   30.283946] rp1-cfe 1f00128000.csi: format: code: 0x3020, fourcc: BYR2
This is the format userspace is trying to use.

Demon000, 2024-09-06:
1.
```
analog@raspberrypi:~/rpicam-apps $ rpicam-hello --list-cameras
Available cameras
-----------------
0 : ov5640 [2592x1944] (/base/axi/pcie@120000/rp1/i2c@80000/ov5640@3c)
    Modes: 'SBGGR8' : 1280x720 [104.17 fps - (-16, 236)/2624x1456 crop]
                      1920x1080 [60.01 fps - (320, 420)/1952x1088 crop]
                      2592x1944 [30.02 fps - (-16, -14)/2624x1952 crop]
           'SRGGB8' : 1280x720 [104.17 fps - (-16, 236)/2624x1456 crop]
                      1920x1080 [60.01 fps - (320, 420)/1952x1088 crop]
                      2592x1944 [30.02 fps - (-16, -14)/2624x1952 crop]
           'SGRBG8' : 1280x720 [104.17 fps - (-16, 236)/2624x1456 crop]
                      1920x1080 [60.01 fps - (320, 420)/1952x1088 crop]
                      2592x1944 [30.02 fps - (-16, -14)/2624x1952 crop]
           'SGBRG8' : 1280x720 [104.17 fps - (-16, 236)/2624x1456 crop]
                      1920x1080 [60.01 fps - (320, 420)/1952x1088 crop]
                      2592x1944 [30.02 fps - (-16, -14)/2624x1952 crop]
```

2. Just `rpicam-hello`
3. Yeah I do.

naushir, 2024-09-06:
Your sensor driver is reporting modes with all 4 Bayer orders via the V4L2 pixel formats.  That looks quite wrong.  It may be the cause of the issue.  Typically, you would only ever have one V4L2 pixel format defined.

Can you try running:
``rpicam-hello -t 0 --viewfinder-mode 1280:720:8:U``

and see if it works.

Demon000, 2024-09-06:
It seems like the computed sensor format changes here, in PipelineHandlerBase::configure().
```
	LOG(RPI, Info) << "Applying sensor format: " << *sensorFormat;
	if (rpiConfig->sensorConfig) {
		ret = data->sensor_->applyConfiguration(*rpiConfig->sensorConfig,
							rpiConfig->combinedTransform_,
							sensorFormat);
	} else {
		ret = data->sensor_->setFormat(sensorFormat,
					       rpiConfig->combinedTransform_);
	}
	LOG(RPI, Info) << "Applied sensor format: " << *sensorFormat;
```

```
[0:29:22.542225458] [2671]  INFO Camera camera.cpp:1191 configuring streams: (0) 1296x972-YUV420 (1) 1920x1080-SBGGR16
[0:29:22.542289273] [2672]  INFO RPI pipeline_base.cpp:543 Applying sensor format: 1920x1080-SBGGR8_1X8
[0:29:22.542376218] [2672]  INFO RPI pipeline_base.cpp:552 Applied sensor format: 1920x1080-SRGGB8_1X8
[0:29:22.542387144] [2672]  INFO RPI pisp.cpp:1426 Raw streams: 1920x1080-BYR2
[0:29:22.542419477] [2672]  INFO RPI pisp.cpp:1455 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/ov5640@3c - Selected sensor format: 1920x1080-SRGGB8_1X8 - Selected CFE format: 1920x1080-BYR2
```

From the expected SBGGR to SRGGB.
Causing the sensor format to change to SRGGB8_1X8, but the CFE format to be kept as 1920x1080-BYR2, which is the wrong order (BGGR).

Demon000, 2024-09-06:
> Your sensor driver is reporting modes with all 4 Bayer orders via the V4L2 pixel formats. That looks quite wrong. It may be the cause of the issue. Typically, you would only ever have one V4L2 pixel [...]

Same stuff.

naushir, 2024-09-06:
The sensor only ever reports a single format out to userland.  The formats in the ``imx219_mbus_formats`` list above are used to translate the formats when the sensor is configured to do horizontal and/or vertical flips.  See:

```
/* Get bayer order based on flip setting. */
static u32 imx219_get_format_code(struct imx219 *imx219, u32 code)
{
	unsigned int i;

	for (i = 0; i < ARRAY_SIZE(imx219_mbus_formats); i++)
		if (imx219_mbus_formats[i] == code)
			break;

	if (i >= ARRAY_SIZE(imx219_mbus_formats))
		i = 0;

	i = (i & ~3) | (imx219->vflip->val ? 2 : 0) |
	    (imx219->hflip->val ? 1 : 0);

	return imx219_mbus_formats[i];
}
``

Demon000, 2024-09-06:
So the formats change based on the flip parameters?

naushir, 2024-09-06:
That's correct - the bayer pattern is not h/v symmetric and will change based on the flips configured in the sensor.

Demon000, 2024-09-06:
Should this actually be the standard behavior? Is this how the mainline driver behaves too?

naushir, 2024-09-06:
Yes, all mainline drivers will behave this way.  There is a mainline IMX219 driver that you can refer to to confirm.

naushir, 2024-09-06:
It's likely that our pipeline handler and kernel drivers for CFE/Unicam won't work correctly since we have never encountered this type of operation.  To be honest, I'm not sure what use it would be to have the ability change the sensor bayer order from userland.

I would suggest you only advertise a single bayer format to start with and see if libcamera can stream correctly.

Demon000, 2024-09-06:
I'm already elbow deep in the libcamera code, might as well find a way to fix it.
Also, is libcamera supposed to hit this assert
https://github.com/raspberrypi/libcamera/blob/main/src/libcamera/pipeline/rpi/common/pipeline_base.cpp#L45
just because the sensor supports other formats too? Seems a bit harsh.

naushir, 2024-09-06:
The Raspberry Pi pipeline handler only supports Bayer sensors.  Having the user attempt to use a non-Bayer mode if a sensor supports it is an error condition.

Demon000, 2024-09-06:
> The Raspberry Pi pipeline handler only supports Bayer sensors. Having the user attempt to use a non-Bayer mode if a sensor supports it is an error condition.

So the pipeline shouldn't even try to skip non-bayer modes? The sensor supports bayer modes too...

naushir, 2024-09-06:
Again, we have not previously encountered sensors that does both Bayer and YUV/RGB outputs, so we could not test this scenario and don't have support for it.

Demon000, 2024-09-06:
If I found fixed for all these issues, should these be submitted on the libcamera mailing list or here?

naushir, 2024-09-06:
Please do submit them to the mailing list.

Demon000, 2024-09-06:
A few more questions.

Where's the logic that decides to use vflip and hflip on IMX219? It seems that those two are set no matter what by libcamera, resulting in a proper BGGR format that can be together with BYR2 (I think this is specifier by the rpicam apps?).

naushir, 2024-09-06:
> Where's the logic that decides to use vflip and hflip on IMX219

This is done via V4L2 controls, see ``imx219_set_ctrl()``:
```c
	case V4L2_CID_HFLIP:
	case V4L2_CID_VFLIP:
		cci_write(imx219->regmap, IMX219_REG_ORIENTATION,
			  imx219->hflip->val | imx219->vflip->val << 1, &ret);
		break;
```

> It seems that those two are set no matter what by libcamera, resulting in a proper BGGR format that can be used together with BYR2 (I think this is specified by the rpicam apps?).

Sensors have an ``orientation`` flag in devicetree to specify mounting orientation.  This gets corrected automatically in libcamera to give you an upright image.  In the case of the IMX219, it is mounted 180 degrees round, so libcamera defaults to a H/V flip to get the correct upright orienation.

If you want to change the orientation, you can specify ``--hflip`` or ``--vflip`` on the ``rpicam-apps`` command line.

Demon000, 2024-09-06:
I switched to IMX219 to test the --hflip and --vflip parameters and they don't work.
```
[0:06:17.999358680] [2783]  INFO RPI pisp.cpp:1455 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
```

The selected format is still bggr.

```
analog@raspberrypi:~/rpicam-apps $ v4l2-ctl --all -d /dev/v4l-subdev2
Driver Info:
        Driver version   : 6.6.45
        Capabilities     : 0x00000000

User Controls

                       exposure 0x00980911 (int)    : min=4 max=1759 step=1 default=1600 value=1759
                horizontal_flip 0x00980914 (bool)   : default=0 value=1 flags=modify-layout
                  vertical_flip 0x00980915 (bool)   : default=0 value=1 flags=modify-layout
```

The flips are still set to 1.

Demon000, 2024-09-06:
Indeed, in 8 bit mode it works.
```
[0:10:18.393426878] [2997]  INFO RPI pisp.cpp:1455 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1920x1080-SBGGR8_1X8 - Selected CFE format: 1920x1080-BYR2
```

But only because SBGGR is the same bayer order as 1920x1080-BYR2.

naushir, 2024-09-06:
This is my output which seems to work fine
```
pi@pi5:~/rpicam-apps/build $ rpicam-hello --camera 1
[1:10:59.426613963] [3246]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1
[1:10:59.440468472] [3247]  INFO RPI pisp.cpp:695 libpisp version 000000000000-invalid 05-08-2024 (11:21:14)
[1:10:59.592947743] [3247]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[1:10:59.593176040] [3247]  INFO RPI pisp.cpp:695 libpisp version 000000000000-invalid 05-08-2024 (11:21:14)
[1:10:59.634176733] [3247]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media4 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made QT preview window
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[1:10:59.970671973] [3246]  INFO Camera camera.cpp:1191 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR_PISP_COMP1
[1:10:59.971228271] [3247]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B

pi@pi5:~/rpicam-apps/build $ rpicam-hello --camera 1 --hflip
[1:11:11.366863148] [3261]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1
[1:11:11.380751546] [3262]  INFO RPI pisp.cpp:695 libpisp version 000000000000-invalid 05-08-2024 (11:21:14)
[1:11:11.532806279] [3262]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[1:11:11.533026761] [3262]  INFO RPI pisp.cpp:695 libpisp version 000000000000-invalid 05-08-2024 (11:21:14)
[1:11:11.574004232] [3262]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media4 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made QT preview window
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[1:11:11.834566551] [3261]  INFO Camera camera.cpp:1191 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-GBRG_PISP_COMP1
[1:11:11.835127627] [3262]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SGBRG10_1X10 - Selected CFE format: 1640x1232-PC1g

pi@pi5:~/rpicam-apps/build $ rpicam-hello --camera 1 --vflip
[1:11:16.806678945] [3275]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1
[1:11:16.820322343] [3276]  INFO RPI pisp.cpp:695 libpisp version 000000000000-invalid 05-08-2024 (11:21:14)
[1:11:16.971803759] [3276]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[1:11:16.972041445] [3276]  INFO RPI pisp.cpp:695 libpisp version 000000000000-invalid 05-08-2024 (11:21:14)
[1:11:17.012951101] [3276]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media4 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made QT preview window
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[1:11:17.276487245] [3275]  INFO Camera camera.cpp:1191 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-GRBG_PISP_COMP1
[1:11:17.277199562] [3276]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SGRBG10_1X10 - Selected CFE format: 1640x1232-PC1G
```

When the flips change, you can see the bayer format change

Demon000, 2024-09-06:
Yeah, I see that now, thanks.

closed 2024-09-10 by naushir