raspberrypi/libcamera #174: No more 64MP camera detected on my pi5

Sadrahel, 2024-09-07:
Since some updates and package modifications to my raspberry Pi5, yesterday I can no longer access my arducam 64MP camera.

I've tried a new installation of my OS, but the problem remains the same.

I'm getting errors when installing libcamera-dev:
(...)
the following packages contain unsatisfied dependencies:
 libcamera0.3: Depends on: libcamera-ipa but will not be installed
E: Unable to fix problems, faulty packages are in “keep as is” mode.
(...)
The following packages contain unsatisfied dependencies:
 libcamera-dev: Dependency: libcamera0.3 (= 0.3.0+rpt20240618-1) but 0.3.1+rpt20240829-1 should be installed
E: Unable to fix problems, faulty packages are in “keep as is” mode.
(...)
The following packages will be upgraded to a LOWER VERSION :
  libcamera-ipa
0 updated, 0 newly installed, 1 downgraded, 0 to be removed and 1 not updated.
E: Packages have been downgraded and -y has been used without --allow-downgrades.

result of libcamera-hello command :
[0:33:13.736589554] [5895] INFO Camera camera_manager.cpp:316 libcamera v0.3.1+49-48fe316f
[0:33:13.744429231] [5898] INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:33:13.745385108] [5898] WARN CameraSensorProperties camera_sensor_properties.cpp:295 No static properties available for 'arducam_64mp'
[0:33:13.745406497] [5898] WARN CameraSensorProperties camera_sensor_properties.cpp:297 Please consider updating the camera sensor properties database
[0:33:13.758679004] [5898] ERROR IPAProxy ipa_proxy.cpp:155 Configuration file 'arducam_64mp.json' not found for IPA module 'rpi/pisp'
[0:33:13.758743152] [5898] ERROR IPARPI ipa_base.cpp:126 Could not create camera helper for arducam_64mp
[0:33:13.758757078] [5898] ERROR RPI pipeline_base.cpp:812 Failed to load a suitable IPA library
[0:33:13.758768338] [5898] ERROR RPI pisp.cpp:922 Failed to register camera arducam_64mp 4-001a: -22
Made X/EGL preview window
ERROR: *** no cameras available ***

Thanx for your help !!!

naushir, 2024-09-07:
Arducam have never upstreamed support for this sensor in our libcamera tree.  You have likely installed some Adrucam custom packages to have this sernsor working - which have now been overriden by the Raspberry Pi packages during an update.

Please speak to Arducam to enable this sensor again with our updated packages.

closed 2024-09-07 by naushir

kbingham, 2024-09-08:
We have https://git.libcamera.org/libcamera/libcamera.git/tree/src/ipa/rpi/vc4/data/ov64a40.json in the mainline linux kernel and libcamera, but because Pi5 isn't yet 'upstream' ... I think the platform specific tuning file has got lost (or was never supported?)

@jmondi have we ever supported the OV64A40 on Pi5 ?

Have Arducam provided a tuning file for the Pi5 for this module?

kbingham, 2024-09-08:
It looks like this was potentially supported downstream here at https://github.com/raspberrypi/libcamera/pull/84#issuecomment-1963620630

naushir, 2024-09-08:
I don't think this is the same sensor here.  The log show the IPA looking for ``arducam_64mp.json`` which presumably is the older Sony 64MP sensor.  The newer OV 64MP sensor does have a tuning file in this tree and ought to be supported.  But good find about the file missing in ``meson.build``, that will need to be fixed.

kbingham, 2024-09-08:
Oh - sorry - completely right. 'arducam_64mp' ... isn't supported, except only by Arducam directly.

naushir, 2024-09-09:
> But good find about the file missing in meson.build, that will need to be fixed.

Seems this change is already there, so OV64a40 should already have full support on Pi 5.

Sadrahel, 2024-09-09:
Ok thanx, i'am waiting for arducam answer, it seems that i'm not the only one with this issue...