raspberrypi/libcamera #175: rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol:

axeljerabek, 2024-09-09:
Since the updates from Friday, the rpicam apps are not working anymore and produce this output:

root@pi:/home/axel# rpicam-hello
rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp7BackEnd7SetCropEjRK19pisp_be_crop_config

Also the update from Saturday did not fix this. All apt packages on a standard Raspberry Pi OS 64bit bookworm.

naushir, 2024-09-09:
I cannot seem to reproduce this, but I have seen one other report of the same error.  Can you provide the output of the following commands please?

1) ``rpicam-hello --version``
2) ``apt-cache policy libpisp-dev``

Have you manually installed any camera packages in the past (rpicam-apps/libcamera/libpisp)?

axeljerabek, 2024-09-09:
As said, the rpicam commands are throwing this error:

root@pi:/home/axel# rpicam-hello --version
rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp7BackEnd7SetCropEjRK19pisp_be_crop_config
root@pi:/home/axel# apt-cache policy libpisp-dev
libpisp-dev:
  Installiert:           (keine)
  Installationskandidat: 1.0.7-1
  Versionstabelle:
     1.0.7-1 500
        500 http://archive.raspberrypi.org/debian bookworm/main arm64 Packages

naushir, 2024-09-09:
What about ``apt-cache policy libcamera-dev``?

axeljerabek, 2024-09-09:
root@pi:/home/axel# apt-cache policy libcamera-dev
libcamera-dev:
  Installiert:           0.3.1+rpt20240906-1
  Installationskandidat: 0.3.1+rpt20240906-1
  Versionstabelle:
 *** 0.3.1+rpt20240906-1 500
        500 http://archive.raspberrypi.org/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     0.0.3-6 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages

naushir, 2024-09-09:
All the versioning information seems correct to me, so I'm a bit unsure what could be going wrong.

naushir, 2024-09-09:
Are you able to use a new SD card with a fresh image and see if that works?

naushir, 2024-09-09:
One other thing, can you send me the output of ``ldd /usr/lib/aarch64-linux-gnu/libcamera.so.0.3.1`` please?

axeljerabek, 2024-09-09:
Here you go:

root@pi:/home/axel# ldd /usr/lib/aarch64-linux-gnu/libcamera.so.0.3.1
        linux-vdso.so.1 (0x0000007f80bb9000)
        libcamera-base.so.0.3 => /lib/aarch64-linux-gnu/libcamera-base.so.0.3 (0x0000007f80910000)
        libpisp.so.1 => /usr/local/lib/aarch64-linux-gnu/libpisp.so.1 (0x0000007f80880000)
        libgnutls.so.30 => /lib/aarch64-linux-gnu/libgnutls.so.30 (0x0000007f80630000)
        liblttng-ust.so.1 => /lib/aarch64-linux-gnu/liblttng-ust.so.1 (0x0000007f80590000)
        libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007f80540000)
        libyaml-0.so.2 => /lib/aarch64-linux-gnu/libyaml-0.so.2 (0x0000007f80500000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f802e0000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f80240000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f80200000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f80050000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f80b7c000)
        libdw.so.1 => /lib/aarch64-linux-gnu/libdw.so.1 (0x0000007f7ff90000)
        libunwind.so.8 => /lib/aarch64-linux-gnu/libunwind.so.8 (0x0000007f7ff40000)
        libp11-kit.so.0 => /lib/aarch64-linux-gnu/libp11-kit.so.0 (0x0000007f7fdf0000)
        libidn2.so.0 => /lib/aarch64-linux-gnu/libidn2.so.0 (0x0000007f7fda0000)
        libunistring.so.2 => /lib/aarch64-linux-gnu/libunistring.so.2 (0x0000007f7fbd0000)
        libtasn1.so.6 => /lib/aarch64-linux-gnu/libtasn1.so.6 (0x0000007f7fb90000)
        libnettle.so.8 => /lib/aarch64-linux-gnu/libnettle.so.8 (0x0000007f7fb20000)
        libhogweed.so.6 => /lib/aarch64-linux-gnu/libhogweed.so.6 (0x0000007f7fab0000)
        libgmp.so.10 => /lib/aarch64-linux-gnu/libgmp.so.10 (0x0000007f7fa10000)
        libnuma.so.1 => /lib/aarch64-linux-gnu/libnuma.so.1 (0x0000007f7f9e0000)
        liblttng-ust-common.so.1 => /lib/aarch64-linux-gnu/liblttng-ust-common.so.1 (0x0000007f7f9b0000)
        liblttng-ust-tracepoint.so.1 => /lib/aarch64-linux-gnu/liblttng-ust-tracepoint.so.1 (0x0000007f7f970000)
        libelf.so.1 => /lib/aarch64-linux-gnu/libelf.so.1 (0x0000007f7f930000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f7f8f0000)
        liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f7f8a0000)
        libbz2.so.1.0 => /lib/aarch64-linux-gnu/libbz2.so.1.0 (0x0000007f7f870000)
        libffi.so.8 => /lib/aarch64-linux-gnu/libffi.so.8 (0x0000007f7f840000)

axeljerabek, 2024-09-09:
Cannot use a new SD, but checked it and it seems ok. It is not older than 3 month actually.

naushir, 2024-09-09:
>```
>libpisp.so.1 => /usr/local/lib/aarch64-linux-gnu/libpisp.so.1 (0x0000007f80880000)
>```
This is the problem right here.  There is a custom/manual build of ``libpisp`` that is being linked to libcamera.  I'm not sure how this got there, but somebody with access to your system might have manually built and installed an older version of ``libpisp`` into your system, and this now conflicts with the new ``libpisp`` library API that we are using.

naushir, 2024-09-09:
Here are some instructions to possibly fix your mismatch.  But just a warning - you may render your system unworkable.

1) Ensure you have the system libpisp packages installed with ``ls -la /usr/local/aarch64-linux-gnu/libpisp*``.  You should see something like:
```shell
lrwxrwxrwx 1 root root       12 Aug  5 11:21 /usr/local/lib/aarch64-linux-gnu/libpisp.so -> libpisp.so.1
lrwxrwxrwx 1 root root       16 Aug  5 11:21 /usr/local/lib/aarch64-linux-gnu/libpisp.so.1 -> libpisp.so.1.0.7
```

2) If you see the above, then remove the local install of libpisp:
```shell
sudo mv /usr/local/lib/aarch64-linux-gnu/libpisp.so  /usr/local/lib/aarch64-linux-gnu/libpisp.so_backup
```

3) Reboot and see if that works.

axeljerabek, 2024-09-09:
root@pi:/home/axel# ls -la /usr/local/lib/aarch64-linux-gnu/libpisp*
lrwxrwxrwx 1 root root     12 21. Feb 2024  /usr/local/lib/aarch64-linux-gnu/libpisp.so -> libpisp.so.1
lrwxrwxrwx 1 root root     16 21. Feb 2024  /usr/local/lib/aarch64-linux-gnu/libpisp.so.1 -> libpisp.so.1.0.4
-rwxr-xr-x 1 root root 562416 21. Feb 2024  /usr/local/lib/aarch64-linux-gnu/libpisp.so.1.0.4

Those I have. moved the /usr/local/lib/aarch64-linux-gnu/libpisp.so and rebooted.

Still have the same error:

root@pi:/home/axel# rpicam-hello
rpicam-hello: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp7BackEnd7SetCropEjRK19pisp_be_crop_config
root@pi:/home/axel# cd /usr/local/lib/aarch64-linux-gnu/

root@pi:/usr/local/lib/aarch64-linux-gnu# ll libpisp*
-rwxr-xr-x 1 root root 550K 21. Feb 2024  libpisp.so.1.0.4
lrwxrwxrwx 1 root root   16 21. Feb 2024  libpisp.so.1 -> libpisp.so.1.0.4

naushir, 2024-09-09:
You should be using ``libpisp.so.1.0.7``, which should be present in ``/usr/lib/aarch64-linux-gnu``.

Try removing/moving all traces of ``libpisp`` from ``/usr/local/lib/aarch64-linux-gnu/`` completely.

axeljerabek, 2024-09-09:
Ok, thanks man! That fixed it. I had a libpisp.so.1.0.4 under /usr/local/lib/aarch64-linux-gnu and a libpisp.so.1.0.7 under /usr/lib/aarch64-linux-gnu. Removed the 1.0.4 ones and now it works again! THANKS! :)

naushir, 2024-09-09:
Great, glad it's working.  The system can get very confused if packages are built/installed manually and through apt as well.
Closing now...

closed 2024-09-09 by naushir