raspberrypi/libcamera #182: IMX296 IMX 334 support

ychwang, 2024-09-28:
can libcamera support IMX296 or IMX334 sonser

naushir, 2024-09-28:
IMX296 is natively supported by the Raspberry Pi libcamera implementation. 

IMX334 is not.  You will need a kernel driver and additional support in libcamera (camera helper and tuning files) to get it working.

ychwang, 2024-09-28:
> IMX296 is natively supported by the Raspberry Pi libcamera implementation.
> 
> IMX334 is not. You will need a kernel driver and additional support in libcamera (camera helper and tuning files) to get it working.

> IMX296 is natively supported by the Raspberry Pi libcamera implementation.
> 
> IMX334 is not. You will need a kernel driver and additional support in libcamera (camera helper and tuning files) to get it working.

Thanks

closed 2024-09-28 by naushir

6by9, 2024-09-29:
It does already have a kernel driver, but with the full res mode running at a link frequency of 891MHz / 1.782Gbit/s it exceeds the spec for Pi5 (1.5Gbit/s). It may work, but no guarantees. Reworking clock structures is possible, but generally needs the datasheet.

ychwang, 2024-09-30:
I get it.Imx296 is my target cmos. Imx334  the image is to large,my program deal is not fit my soulution.So,Imx296 is my best choose.Thank you very much.