raspberrypi/libcamera #183: Cross compiled camera application error

ivan-ushakov, 2024-09-28:
Hello,

I use Docker cross compilation. I follow [rpicam_app.cpp](https://github.com/raspberrypi/rpicam-apps/blob/main/core/rpicam_app.cpp) (and also [C++ Guide](https://libcamera.org/guides/application-developer.html) but for some reason I'm getting `ERROR Camera camera.cpp:587 Removing camera while still in use` error and my callback never called.

My system on Raspberry Pi Zero:

```
Linux raspberrypi 6.6.31+rpt-rpi-v6 #1 Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv6l GNU/Linux
```

Raspberry Pi Zero libcamera version:

```
rpicam-apps build: 49344f2a8d18 17-06-2024 (12:19:15)
libcamera build: v0.3.0+65-6ddd79b5
```

Debug output from my test application:

```
pi@raspberrypi:~ $ LIBCAMERA_LOG_LEVELS=*:0 LIBCAMERA_RPI_CONFIG_FILE=/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml ./camera-bot settings.json
[1:12:05.047862075] [1636] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rkisp1.so is signed
[1:12:05.050606017] [1636] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rkisp1.so'
[1:12:05.053322959] [1636] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_pisp.so is signed
[1:12:05.053945945] [1636] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_pisp.so'
[1:12:05.056012901] [1636] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_vc4.so is signed
[1:12:05.057511869] [1636] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_vc4.so'
[1:12:05.059319831] [1636] DEBUG IPAModule ipa_module.cpp:334 ipa_soft_simple.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_soft_simple.so is signed
[1:12:05.060857798] [1636] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_soft_simple.so'
[1:12:05.061693780] [1636] DEBUG IPAModule ipa_module.cpp:334 ipa_vimc.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so is signed
[1:12:05.063142749] [1636] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so'
[1:12:05.064567719] [1636]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+65-6ddd79b5
[1:12:05.066356681] [1639] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[1:12:05.148681927] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-codec" created from /dev/media2
[1:12:05.150762883] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 5 missing dependencies
[1:12:05.164564588] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[1:12:05.166564546] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: bcm2835-codec
[1:12:05.170970452] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media0
[1:12:05.172961410] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 4 missing dependencies
[1:12:05.177402315] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "bcm2835-isp" created from /dev/media1
[1:12:05.178717287] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 4 missing dependencies
[1:12:05.188689074] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[1:12:05.190814029] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: bcm2835-isp
[1:12:05.202301784] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[1:12:05.202585778] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: bcm2835-isp
[1:12:05.206088704] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "unicam" created from /dev/media3
[1:12:05.210231615] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[1:12:05.214751519] [1639] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[1:12:05.216736477] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: unicam
[1:12:05.219701414] [1639] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'imx8-isi'
[1:12:05.221842368] [1639] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'mali-c55'
[1:12:05.224398314] [1639] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rkisp1'
[1:12:05.225113298] [1639] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rpi/pisp'
[1:12:05.225853283] [1639] DEBUG RPI pisp.cpp:865 Unable to acquire a CFE instance
[1:12:05.227159255] [1639] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rpi/vc4'
[1:12:05.227646244] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "unicam"
[1:12:05.229161212] [1639] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "bcm2835-isp"
[1:12:05.231423164] [1639] DEBUG DmaBufAllocator dma_buf_allocator.cpp:109 Using /dev/dma_heap/linux,cma
[1:12:05.232069150] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: White Balance, Automatic (0x0098090c)
[1:12:05.232526140] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Exposure (0x00980911)
[1:12:05.234195105] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Gain, Automatic (0x00980912)
[1:12:05.234636095] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Horizontal Flip (0x00980914)
[1:12:05.235018087] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Vertical Flip (0x00980915)
[1:12:05.236657052] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Auto Exposure (0x009a0901)
[1:12:05.237146042] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Camera Orientation (0x009a0922)
[1:12:05.238734008] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Camera Sensor Rotation (0x009a0923)
[1:12:05.239177999] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Vertical Blanking (0x009e0901)
[1:12:05.239458993] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Horizontal Blanking (0x009e0902)
[1:12:05.239928983] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Analogue Gain (0x009e0903)
[1:12:05.242504928] [1639] DEBUG V4L2 v4l2_device.cpp:636 'ov5647 10-0036': Control: Pixel Rate (0x009f0902)
[... 84 lines cut ...]
cpp = 'armv6-rpi-linux-gnueabihf-g++'
ar = 'armv6-rpi-linux-gnueabihf-ar'
strip = 'armv6-rpi-linux-gnueabihf-strip'

[host_machine]
system = 'linux'
cpu_family = 'x86_64'
cpu = 'x86_64'
endian = 'little'
EOF

meson setup build --cross-file x86_64-w64-armv6.txt  --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=disabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=disabled

ninja -C build install
```

I use CMake for building:

```
find_package(PkgConfig)
pkg_check_modules(LIBCAMERA REQUIRED IMPORTED_TARGET libcamera)
pkg_check_modules(LIBPISP REQUIRED IMPORTED_TARGET libpisp)

target_link_libraries(${PROJECT_NAME} PRIVATE PkgConfig::LIBPISP PkgConfig::LIBCAMERA)
```

naushir, 2024-09-30:
What is the eventual rpicam-apps command line you are trying to run?  Does this problem also happen with the system pacakges?

ivan-ushakov, 2024-09-30:
`rpicam-still` works fine on my Zero. Error happens in my cross compiled C++ application and I don't understand why. I used the same source code for the libcamera as Zero has

naushir, 2024-10-01:
Just to confirm, you are cross compiling ``rpicam-apps``, and it does not work, but you build ``rpicam-apps`` natively on the Pi and it works.  Is that correct? If so, what command line are you using to invoke the ``rpicam-apps`` application?

ivan-ushakov, 2024-10-01:
Sorry, I mean I have my C++ application (not `rpicam-apps`) where I use libcamera API. I cross compile this application with `ghcr.io/tttapa/docker-arm-cross-toolchain:armv6-rpi-linux-gnueabihf` toolchain and I try to run this application on my Raspberry Pi Zero device

naushir, 2024-10-03:
Ah I see.  Can you share the source code (or create a minimal application) that shows the error for us to look at?

naushir, 2024-10-03:
When you cross compile, are you using the exact same version of the ``libcamera.so`` file on the host PC and the Raspberry Pi?

ivan-ushakov, 2024-10-03:
No, I use cross compiled version of libcamera as I mentioned in my first message

naushir, 2024-10-07:
I don't really know what more to suggest here.  Perhaps you can try running the [simple camera app](https://git.libcamera.org/libcamera/simple-cam.git/tree/) through your cross compile method and see if that works?

ivan-ushakov, 2024-10-09:
I decided to go with `sbuild` and to have the same versions of libraries that my device has but for some reason libcamera version is too old compared to device.

I use this command to get Raspberry Pi Bookworm file system:

```bash
mk-sbuild --arch=armhf bookworm --debootstrap-mirror=http://raspbian.raspberrypi.org/raspbian --name=rpizero-bookworm --debootstrap-keyring "$HOME/.gnupg/pubring.kbx --merged-usr" --skip-proposed --skip-updates --skip-security
```

Here is library version:

```bash
$ ls /var/lib/schroot/chroots/rpizero-bookworm-armhf/usr/lib/arm-linux-gnueabihf/libcame*
/var/lib/schroot/chroots/rpizero-bookworm-armhf/usr/lib/arm-linux-gnueabihf/libcamera-base.so        /var/lib/schroot/chroots/rpizero-bookworm-armhf/usr/lib/arm-linux-gnueabihf/libcamera.so
/var/lib/schroot/chroots/rpizero-bookworm-armhf/usr/lib/arm-linux-gnueabihf/libcamera-base.so.0.0.3  /var/lib/schroot/chroots/rpizero-bookworm-armhf/usr/lib/arm-linux-gnueabihf/libcamera.so.0.0.3
```

But device has another versions:

```bash
pi@raspberrypi:~ $ ls /usr/lib/arm-linux-gnueabihf/libcam*
/usr/lib/arm-linux-gnueabihf/libcamera-base.so.0.3  /usr/lib/arm-linux-gnueabihf/libcamera-base.so.0.3.0  /usr/lib/arm-linux-gnueabihf/libcamera.so.0.3  /usr/lib/arm-linux-gnueabihf/libcamera.so.0.3.0  /usr/lib/arm-linux-gnueabihf/libcamera_app.so
```

I don't understand how this is possible

closed 2024-10-16 by ivan-ushakov

ivan-ushakov, 2024-10-23:
Maybe someone will be interested. I found that problem was in cross compile toolchain (https://github.com/tttapa/RPi-Cross-Cpp-Development/issues/4) and `_GLIBCXX_HAVE_ATOMIC_LOCK_POLICY ` compiler flag. For some reason Raspbian OS compiled with this flag disabled and `shared_ptr` uses mutex for counter protection. In toolchain this flag was enabled so `shared_ptr` uses atomic for counter. This leads to different memory layout of `shared_ptr` and some other problems and at the end `shared_ptr` correct behaviour.