raspberrypi/libcamera #185: rpicam-still does not save an image

hinxx, 2024-10-04:
Trying to capture a single PNG from the camera and no file is created.

```
hinxx@raspberrypi:~ $ rpicam-still --timeout 0 --nopreview  --encoding png --output test.png
[0:15:23.472518593] [2137]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:15:23.480105158] [2140]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:15:23.497327766] [2140]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:15:23.498154543] [2137]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-GBRG_PISP_COMP1
[0:15:23.498285135] [2140]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected CFE format: 1296x972-PC1g
^C
hinxx@raspberrypi:~ $ ls 
Bookshelf  Desktop  Documents  Downloads  Music  Pictures  Public  Templates  Videos
```

naushir, 2024-10-04:
Your command line sets a timeout of 0, which means wait forever. Change the timeout setting to something like 5000 for a 5 second wait before capture.

hinxx, 2024-10-04:
Works!
Hmm, I thought that 0 would make it capture the image with no timeout.

closed 2024-10-04 by hinxx