raspberrypi/libcamera #193: IMX296LLR

ychwang, 2024-10-16:
my cmos is IMX296LLR ,it is a gray cmos, how i can use libcamera to get gray img.The same time  my fps is only 10fps,how can i improve the captrue fps.

naushir, 2024-10-16:
What application are you trying to run here?  Have you used rpicam-apps for the capture?

ychwang, 2024-10-18:
> What application are you trying to run here? Have you used rpicam-apps for the capture?

i used libcamera app,i can not find rpicam-apps  SDK .

ychwang, 2024-10-18:
it is not easy to  use SDK ,not like Gige protcol easy to understand

naushir, 2024-10-18:
libcamera-app and rpicam-apps are the same library. Please could you give me the exact command line you are trying to use to capture an image?

ychwang, 2024-10-18:
> libcamera-app and rpicam-apps are the same library. Please could you give me the exact command line you are trying to use to capture an image?

I want use capture the image  in my QT program,i want use a class to control ,get image and tranform the images to my alogrithm

naushir, 2024-10-18:
Can I suggest you start with the basics.  See if you can capture an image with

```
rpicam-still -t 5s -o test.jpg
```

ychwang, 2024-10-18:
> Can I suggest you start with the basics. See if you can capture an image with
> 
> ```
> rpicam-still -t 5s -o test.jpg
> ```

i means  i write a qt program,this program need capture the images per 30ms, my alogrithm  process the images the camera capture.

naushir, 2024-10-18:
> Can I suggest you start with the basics. See if you can capture an image with
> 
> ```
> rpicam-still -t 5s -o test.jpg
> ```

Please start with the above.

ychwang, 2024-10-19:
> > Can I suggest you start with the basics. See if you can capture an image with
> > ```
> > rpicam-still -t 5s -o test.jpg
> > ```
> 
> Please start with the above.

Maybe I did not explain my needs, I hope that I can use the SDK to write a program to get your own video stream, can get the raw data of the video stream, and then use the image processing method to process the collected video stream.

naushir, 2024-10-21:
You can look at examples such as [simple-cam](https://git.libcamera.org/libcamera/simple-cam.git) which is purely libcamera based, or [rpicam-hello](https://github.com/raspberrypi/rpicam-apps/blob/main/apps/rpicam_hello.cpp) which is based on our rpicam-apps framework.

I'm afraid I can't really advise any more than that

closed 2024-10-21 by naushir