raspberrypi/libcamera #194: Failed to start media pipeline: -32 on pi5

pbaetens, 2024-10-16:
Hello,

Pi5 / latest self-compiled bookworm with custom image sensor driver.
This driver worked fine on bullseye before.

When starting the stream, I get this error in DMESG.

any idea how to debug this?

[ 1662.265013] rp1-cfe 1f00110000.csi: Failed to start media pipeline: -32

pbaetens, 2024-10-16:
2 sensors are connected. 
mira016 does not work, mira050 does work

```
pi@raspberry5:~ $ rpicam-hello --list-cameras
Available cameras
-----------------
0 : mira016 [400x400 12-bit GRBG] (/base/axi/pcie@120000/rp1/i2c@88000/mira016@36)
    Modes: 'SGRBG10_CSI2P' : 400x400 [357.14 fps - (0, 0)/400x400 crop]
           'SGRBG12_CSI2P' : 400x400 [200.00 fps - (0, 0)/400x400 crop]
           'SGRBG8' : 400x400 [357.14 fps - (0, 0)/400x400 crop]

1 : mira050 [576x768 12-bit GRBG] (/base/axi/pcie@120000/rp1/i2c@80000/mira050@36)
    Modes: 'SGRBG10_CSI2P' : 576x768 [120.05 fps - (0, 0)/576x768 crop]
           'SGRBG12_CSI2P' : 576x768 [59.95 fps - (0, 0)/576x768 crop]
           'SGRBG8' : 576x768 [120.05 fps - (0, 0)/576x768 crop]
```

```
Media controller API version 6.6.47

Media device information
------------------------
driver          rp1-cfe
model           rp1-cfe
serial          
bus info        platform:1f00110000.csi
hw revision     0x114666
driver version  6.6.47

Device topology
- entity 1: csi2 (8 pads, 9 links)
            type V4L2 subdev subtype Unknown flags 0
            device node name /dev/v4l-subdev0
	pad0: Sink
		[fmt:SGRBG8_1X8/400x400 field:none colorspace:raw]
		<- "mira016 6-0036":0 [ENABLED,IMMUTABLE]
	pad1: Sink
		[fmt:unknown/16384x1 field:none]
		<- "mira016 6-0036":1 [ENABLED,IMMUTABLE]
	pad2: Sink
		[fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
	pad3: Sink
		[fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
	pad4: Source
		[fmt:SGRBG16_1X16/400x400 field:none colorspace:raw]
		-> "rp1-cfe-csi2_ch0":0 []
		-> "pisp-fe":0 [ENABLED]
	pad5: Source
		[fmt:unknown/16384x1 field:none]
		-> "rp1-cfe-embedded":0 []
	pad6: Source
		[fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
		-> "rp1-cfe-csi2_ch2":0 []
		-> "pisp-fe":0 []
	pad7: Source
		[fmt:SRGGB10_1X10/640x480 field:none colorspace:raw xfer:none ycbcr:601 quantization:full-range]
		-> "rp1-cfe-csi2_ch3":0 []
		-> "pisp-fe":0 []

- entity 10: pisp-fe (5 pads, 7 links)
             type V4L2 subdev subtype Unknown flags 0
             device node name /dev/v4l-subdev1
	pad0: Sink
		[fmt:SGRBG16_1X16/400x400 field:none colorspace:raw]
		<- "csi2":4 [ENABLED]
		<- "csi2":6 []
		<- "csi2":7 []
	pad1: Sink
		[fmt:FIXED/16384x1 field:none]
[... 176 lines cut ...]
- entity 35: rp1-cfe-fe_image0 (1 pad, 1 link)
             type Node subtype V4L flags 1
             device node name /dev/video12
	pad0: Sink
		<- "pisp-fe":2 [ENABLED]

- entity 39: rp1-cfe-fe_image1 (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video13
	pad0: Sink
		<- "pisp-fe":3 []

- entity 43: rp1-cfe-fe_stats (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video14
	pad0: Sink
		<- "pisp-fe":4 [ENABLED]

- entity 47: rp1-cfe-fe_config (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video15
	pad0: Source
		-> "pisp-fe":1 [ENABLED]

```

naushir, 2024-10-16:
I can't see anything obvious that's wrong with that setup. What command line are you using to drive the camera?

naushir, 2024-10-22:
Any update on this?

naushir, 2024-10-31:
Closing now.  If you have any update please do re-open.

closed 2024-10-31 by naushir

pbaetens, 2024-10-31:
found the issue, was a bug in the sensor driver. ticket can be kept closed.

dassicity, 2025-05-16:
Hey @pbaetens , would you please like to communicate what was the bug in the sensor driver ?