raspberrypi/libcamera #195: libcamera-still image quality much better than my c++ code

muttistefano, 2024-10-16:
Hi all,
I am developing a C++ app with a quadroscopic 1MP global shutter camera board on RPI5.
I noticed that by default, libcamera-still quality is much better than the picture I acquired with my code.
Does libcamera-still use some configuration by default?
I am not currently modifying any control parameters in my code.
Is there a way to get libcamera-still default parameters to understand how to tune mine? 
thanks

naushir, 2024-10-16:
Can you post some example images showing the differences?

muttistefano, 2024-10-16:
Hi, 
this is the one captured from the quadrascopic camera arducam using "libcamera-still -t 500 -n -o test.jpg"
[image]

while these ones are captured using the a c++ app made using libcamera API:
[image]
[image]
[image]
[image]

I tried tweaking with auto exposure, flickering control, and brightness, but I can achieve the same quality.
Plus, it seems like there is some tearing in the one I acquired with the API.
Thanks

naushir, 2024-10-17:
Which libcamera library are you using?  Is it supplied by Arducam or the Raspberry Pi supplied one?

muttistefano, 2024-10-17:
I am using the one patched by arducam.

naushir, 2024-10-17:
Have you tried the Raspberry Pi libcamera packages instead?  I'm afraid I don't know what Arducam do in their version of libcamera, so it would be best to ask them for assistance.

muttistefano, 2024-10-17:
Well I tought the normal libcamera won't work with their module because they provide a board that merges 4 cameras (https://www.arducam.com/product/arducam-1mp4-quadrascopic-camera-bundle-kit-for-raspberry-pi-nvidia-jetson-nano-xavier-nx-four-ov9782-global-shutter-color-camera-modules-and-camarray-camera-hat-b0331/)
I am also in contact with them.
Thanks

6by9, 2024-10-17:
> quadrascopic camera arducam

Presumably the quad OV9281 https://www.uctronics.com/arducam-1mp-4-quadrascopic-camera-bundle-kit-for-raspberry-pi-nvidia-jetson-nano-xavier-nx-four-ov9281-global-shutter-monochrome-camera-modules-and-camarray-camera-hat.html
It's not supported by the standard  libcamera.

The Arducam board is doing some significant data manipulation to combine the 4 images into one, so support needs to come from them. There's very little that Raspberry Pi can do to support it.

The individual images you've captured just look under-exposed. Have you given AE time to adapt?

naushir, 2024-10-17:
Agree, it looks like the images are simply underexposed.  But I can't tell what/how the statistics are generated and used on a 4x combined image like this.  This is definitely something you need to speak to Arducam about.

muttistefano, 2024-10-17:
I will post here for reference if arducam comes back with news.
Thanks all.

closed 2024-10-17 by muttistefano