raspberrypi/libcamera #198: [ERROR] Camera Timeout Error with Raspberry Pi Camera Module V3 after Switching to SSD

Vaskozzz, 2024-10-18:
Hello everyone,

I would like to share a current issue I'm facing with my Raspberry Pi Camera Module V3 (wide) while trying to use it. Here's some context:

I am powering my Raspberry Pi with a 12V, 20A supply, using a step-down converter set to 5.1V and 8A, which is at its maximum setting to provide the necessary current for the Raspberry Pi. Previously, I connected my Raspberry Pi using this step-down converter along with my peripherals and camera without any issues. I could run my programs normally, and I did not encounter any low voltage warnings.

However, I recently switched from using a microSD card with my operating system (as recommended by the Pi Imager) to an SSD with a 52PI HAT. Initially, everything worked smoothly after connecting the SSD. Unfortunately, later that night, when I attempted to power off the Raspberry Pi, the camera feed started displaying pink vertical lines. The following day, when I tried to execute the camera commands again, I received the terminal error shown below:

`rpi@raspberrypi:~ $ libcamera-hello
[0:00:18.072787691] [2108]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:00:18.081116006] [2114]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:00:18.101522414] [2114]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:00:18.944975283] [2108]  INFO Camera camera.cpp:1197 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR_PISP_COMP1
[0:00:18.945147580] [2114]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[0:00:20.092635487] [2114]  WARN V4L2 v4l2_videodevice.cpp:2095 /dev/video4[17:cap]: Dequeue timer of 1000000.00us has expired!
[0:00:20.092686227] [2114] ERROR RPI pipeline_base.cpp:1364 Camera frontend has timed out!
[0:00:20.092691727] [2114] ERROR RPI pipeline_base.cpp:1365 Please check that your camera sensor connector is attached securely.
[0:00:20.092696968] [2114] ERROR RPI pipeline_base.cpp:1366 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!`

I have already checked and replaced the cable, tried different ports, and even swapped out the camera, as I have two identical ones. Unfortunately, the issue persists. I then reverted to the microSD card, formatting it with a clean operating system from the Pi Imager, but the problem remains.

Interestingly, when I connect a USB camera, it works without any issues. I would greatly appreciate any insights on the origin of this problem and any potential solutions you might suggest.

Thank you in advance for your help!

naushir, 2024-10-18:
I suspect this is indeed a power supply issue.  Can I suggest trying an official Raspberry Pi 5 power supply connected to the USB-C port as an experiment to see if it goes wrong?

naushir, 2024-10-18:
One other question - how did you connect your supply to the Pi 5 board?  Did it go through the USB-C connector?

naushir, 2024-10-22:
Any update on this?  If not I'll close this down shortly.

naushir, 2024-10-31:
Closing now.

closed 2024-10-31 by naushir