raspberrypi/libcamera #199: Problem compiling libcamera on Pi Zero 2W (armv7l)

LeBirlante, 2024-10-20:
Hi there,

I was trying to compile libcamera for my Zero 2W, so I basically dumped my SD card to an IMG file, mounted it on my desktop linux and chrooted it, allowing QEMU to handle the armv7l architecture.

Things were going along nicely, until this:

-----------------------------------------------------
[6/11] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o
FAILED: src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o 
c++ -Isrc/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p -Isrc/ipa/rpi/cam_helper -I../src/ipa/rpi/cam_helper -Isrc/ipa/rpi -I../src/ipa/rpi -Iinclude -I../include -Iinclude/libcamera/ipa -Iinclude/libcamera -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wno-redundant-move -Wno-psabi -Wmissing-declarations -Wshadow -include /home/varet/libcamera/build/config.h -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o -MF src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o.d -o src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o -c ../src/ipa/rpi/cam_helper/cam_helper_imx500.cpp
../src/ipa/rpi/cam_helper/cam_helper_imx500.cpp: In member function 'void CamHelperImx500::parseInferenceData(libcamera::Span<const unsigned char>, RPiController::Metadata&)':

../src/ipa/rpi/cam_helper/cam_helper_imx500.cpp:259:40: error: 'char* strncpy(char*, const char*, size_t)' specified bound 64 equals destination size [-Werror=stringop-truncation]

  259 |                                 strncpy(exported.networkName, inputTensorInfo.networkName.c_str(),
      |                                 ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  260 |                                         sizeof(exported.networkName));
      |                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
----------------------------------------------------

my C++ days are long gone, but would it be that the crux of the problem lies at the "specified bound 64 equals destination size", triggered due to using the -Werror=stringop-truncation compile option?

thanks for your time,

naushir, 2024-10-21:
That looks like the compiler being overly picky about the code.  It seems to be complaining that the ``strncpy`` will possibly overwrite the ``\0`` nul terminator at the end of the string.  However, on the very next line we add the terminator back:

```c++
				strncpy(exported.networkName, inputTensorInfo.networkName.c_str(),
					sizeof(exported.networkName));
				exported.networkName[sizeof(exported.networkName) - 1] = '\0';
```

I think you can either ignore this warning, or update the line to 

```c++
				strncpy(exported.networkName, inputTensorInfo.networkName.c_str(),
					sizeof(exported.networkName) - 1);
```

LeBirlante, 2024-10-23:
I modified the code as per your suggestion, and sure enough, compile was successful. Now I'm trying to find out why rpicam-apps keeps complaining about not finding libcamera dependencies (even when I installed it via "ninja -C build install"). When libcamera was installed via my distro packaging tools, this wasn't a problem, but now that I've installed libcamera manually, it seems rpicam-apps meson build it's missing some paths. I'll give it a look and see what can I get out of that...

thanks for your time!

naushir, 2024-10-31:
Thanks for the update.  Closing this particular issue now.  If you have further problems feel free to open a new issue.

closed 2024-10-31 by naushir