raspberrypi/libcamera #201: GStreamer framerate support

superware, 2024-11-01:
Hello,

On the following setup:

* Raspberry PI 4B
* Bookworm, 6.6.51+rpt-rpi-v8
* GStreamer 1.22.0-2
* libcamera 0.3.2+rpt20240927-1 arm64
* [RPi Camera (D) OV5647](https://www.waveshare.com/wiki/RPi_Camera_(D))

The following (specifically `framerate=50/1`) halts the pipeline:

```
gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=1280,height=720,framerate=50/1,interlace-mode=(string)progressive,colorimetry=bt709' ! fakesink
```

While `framerate=30/1` works, which means libcamera is not ignoring the requested frame rate. Is there any way to force libcamera to a specific framerate?

Thanks in advance.

naushir, 2024-11-06:
I don't think gstreamer fully supports camera controls (like framerate) yet.  Is there a specific reason you are using gstreamer here?    Could you instead use picamera2 or rpicam-apps?

kbingham, 2024-11-06:
gstreamer does support framerate and that's been known/expected to work on RPi for quite some time.

naushir, 2024-11-06:
@superware, does ``rpicam-hello -t 0 --framerate 50`` give you the expected result?

superware, 2024-11-06:
@naushir well I'm on RPI4 with Raspberry Pi OS Lite (no desktop):

```
$ rpicam-hello -t 0 --framerate 25
[0:50:41.767093850] [2392]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:50:41.808974321] [2395]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:50:41.811503018] [2395]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media2
[0:50:41.811590091] [2395]  INFO RPI pipeline_base.cpp:1126 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 1296:972:12:P(25)
    SGBRG10_CSI2P,640x480/58.924 - Score: 3296
    SGBRG10_CSI2P,1296x972/43.247 - Score: 1000
    SGBRG10_CSI2P,1920x1080/30.6185 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/15.6335 - Score: 20299.9
Stream configuration adjusted
[0:50:41.819030295] [2392]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:50:41.819445698] [2395]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
#0 (0.00 fps) exp 27650.00 ag 2.00 dg 1.00
#1 (12.01 fps) exp 27650.00 ag 2.00 dg 1.00
#2 (12.01 fps) exp 27604.00 ag 2.00 dg 1.00
#3 (12.01 fps) exp 27627.00 ag 2.00 dg 1.00
#4 (12.01 fps) exp 27604.00 ag 2.00 dg 1.00
#5 (12.01 fps) exp 27604.00 ag 2.00 dg 1.00
#6 (12.01 fps) exp 27604.00 ag 2.00 dg 1.00
#7 (12.01 fps) exp 27604.00 ag 2.00 dg 1.00
#8 (12.01 fps) exp 27604.00 ag 2.00 dg 1.00
```

Why 12 fps? While the following halts:

```
rpicam-hello -t 0 --framerate 50
[0:53:51.648558198] [2447]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:53:51.681518782] [2450]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:53:51.684122220] [2450]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media2
[0:53:51.684204330] [2450]  INFO RPI pipeline_base.cpp:1126 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 1296:972:12:P(50)
    SGBRG10_CSI2P,640x480/58.924 - Score: 3296
    SGBRG10_CSI2P,1296x972/43.247 - Score: 14506
    SGBRG10_CSI2P,1920x1080/30.6185 - Score: 40112.7
    SGBRG10_CSI2P,2592x1944/15.6335 - Score: 70299.9
Stream configuration adjusted
[0:53:51.691610795] [2447]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 640x480-SGBRG10_CSI2P
[0:53:51.691920903] [2450]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
#0 (0.00 fps) exp 13031.00 ag 2.00 dg 1.00
#1 (25.49 fps) exp 15927.00 ag 2.00 dg 2.13
```

naushir, 2024-11-06:
I've tried this locally and my OV5647 behaves as expected at 25 and 50fps.  Few questions:

- Do you have the Waveshare camera with IR illumination?
- What kernel are you running?  Can you post the output of ``uname -a``.

superware, 2024-11-06:
* Looks like a [RPi Camera (D)](https://www.waveshare.com/rpi-camera-d.htm) with "Raspberry PI Camera rev 2.1" on the PCB.
* `Linux rpi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`

naushir, 2024-11-06:
You seem to have the correct software environment setup, so I'm not sure why your framerate values not as expected.

Just to clarify that's not an official Raspberry Pi product, and likely not a true Omnivision sensor.  Do you happen to have another camera to test with (preferably an official Raspberry Pi camera)?

superware, 2024-11-06:
Unfortunately I don't have an official one, do you think this can actually be a fake?

[image]
[image]

naushir, 2024-11-06:
Omnivision stopped producing the OV5647 sensor many many years ago.  Anything recently advertised as a Raspberry Pi Camera v1 or OV5647 sensor is a clone device made by an unknown vendor.  This almost certainly include the Waveshare device that you have.

6by9, 2024-11-06:
Last time buy for OV5647 was announced was 2014.
Either Omnivision changed their minds and decided to keep making it, or somehow a third party are making an identical looking product.

Performance also depends on the external components such as the oscillator on the camera board. On the Pi camera boards that was 25MHz, and the software assumes that to be the case. Put some other oscillator on the board and it won't go at the rate expected.

I wouldn't normally suggest this, but you could install Bullseye and enable the legacy camera stack to determine if that produces the correct frame rate. If it doesn't, then it's definitely a module issue and nothing to do with libcamera itself.

superware, 2024-11-06:
> install Bullseye and enable the legacy camera stack to determine if that produces the correct frame rate

Ok, since many others can also benefit from this, I'm trying with Bullseye.

superware, 2024-11-06:
So same RPI4 and OV5647, legacy Bullseye 64bit standard (not lite/full) image, enabled `Legacy Camera` with `raspi-config`, added `dtoverlay=ov5647` to `/boot/config.txt` below `[all]`, rebooted, and `libcamera-hello -t 0 --framerate 30` shows normal video but at 14.41 fps. Is this the proof?

6by9, 2024-11-06:
No as that is still using libcamera, not the legacy stack.

I'm having to try and remember as it has been legacy and deprecated for a fair while. The legacy stack via raspivid etc is also not supported on 64 bit OSes.

Remove the "dtoverlay=ov5647" line. Reboot. I would hope `v4l2-ctl --list-devices` gives you "MMAL service" or similar as /dev/video0.
```
v4l2-ctl -v width=1280,height=720,pixelformat=YU12
v4l2-ctl -p 30
v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null
```
should stream image data, displays a `<` for each frame, and an averaged frame rate for the stream.
If that gives 30fps, then something very funny is going on in libcamera, otherwise it looks like it's that module.

superware, 2024-11-06:
So removed `dtoverlay=ov5647`, made sure `Legacy Camera` is enabled in `raspi-config`, rebooted and:

```
$ v4l2-ctl -v width=1280,height=720,pixelformat=YU12
$ v4l2-ctl -p 30
Frame rate set to 30.000 fps
$ v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null
<<<<<<<<<<<<<< 11.53 fps
<<<<<<<<<<<<<<< 12.92 fps
<<<<<<<<<<<<<<<< 13.43 fps
<<<<<<<<<<<<<<< 13.66 fps
<<<<<<<<<<<<<<<< 13.82 fps
<<<<<<<<<<<<<<< 13.91 fps
<<<<<<<<<<<<<<< 13.98 fps
<<<<<<<<<<<<<<<< 14.04 fps
<<<<<<<<<<<<<<< 14.08 fps
...
```

superware, 2024-11-06:
Confirmed, `-V` shows 1280q720 YUV12 and `-P` show 30.000 fps. So if I understand correctly, this is indeed an OV5647 sensor that comes in a Waveshare box, but the supporting board is a clone using low grade components and hence requires a slower oscillator delivering about half the fps?

6by9, 2024-11-06:
IMHO it is a faulty product that Waveshare have supplied you. It is not compatible with the v1 camera module that Raspberry Pi sold from approx 2012-2015.

I'm guessing that the oscillator is incorrect (you may be able to read the markings on it - the silver rectangular component on the back side), and therefore the sensor isn't running at the rates expected.
It could be a one off, it could be a batch at fault, or it could be a design issue. I have no idea on that, but it's not a situation that is worth trying to fix up in the software.

superware, 2024-11-06:
I've just contacted Waveshare support, looking forward to their professional response. Will report back and take it from there.

naushir, 2024-11-08:
Closing this now since it's a HW related issue.

closed 2024-11-08 by naushir