raspberrypi/libcamera #203: OV5647 not functioning with Raspberry Pi 4

Aly-Abdel-Motaleb, 2024-11-06:
I have a raspberry pi 4 with the ov5647 camera the problem is that I have followed the instructions in the site [[https://docs.arducam.com/Raspberry-Pi-C ... MP-OV5647/](https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/5MP-OV5647/)
then the raspberry pi output this I have tried with 3 connectors and 2 different cameras

```
Twins@raspberrypi:~ $ libcamera-hello
[0:12:49.676451259] [2050]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+51-28bff115-dirty (2024-10-08T05:59:20BST)
[0:12:49.726434336] [2053]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:49.728666913] [2053]  INFO RPI vc4.cpp:455 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[0:12:49.728744987] [2053]  INFO RPI pipeline_base.cpp:1151 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:12:51.042246669] [2050]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:12:51.043193014] [2053]  INFO RPI vc4.cpp:630 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:12:52.179284961] [2053]  WARN V4L2 v4l2_videodevice.cpp:2095 /dev/video0[13:cap]: Dequeue timer of 1000000.00us has expired!
[0:12:52.179482830] [2053] ERROR RPI pipeline_base.cpp:1412 Camera frontend has timed out!
[0:12:52.179519033] [2053] ERROR RPI pipeline_base.cpp:1413 Please check that your camera sensor connector is attached securely.
[0:12:52.179550792] [2053] ERROR RPI pipeline_base.cpp:1414 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
```

```
Twins@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

**Configuration file**

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtoverlay=ov5647

```

naushir, 2024-11-06:
Firstly, what libcamera package are your using:

>```
>[0:12:49.676451259] [2050]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+51-28bff115-dirty (2024-10 08T05:59:20BST)
>```

The above suggests a local build with source code modifications. Is that correct?  If so, please reinstall the Raspberry Pi libcamera packages.  Also, please do not follow the Arducam instructions, instead refer to the following forum [post](https://forums.raspberrypi.com/viewtopic.php?t=362707&sid=1186f1b1669f615e32ed5af1b40bc807).

Aly-Abdel-Motaleb, 2024-11-06:
```
Twins@raspberrypi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [46.34 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [32.81 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

Twins@raspberrypi:~ $ libcamera-hello
[0:57:42.511549808] [8545]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+51-28bff115-dirty (2024-10-08T05:59:20BST)
[0:57:42.562256647] [8550]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:57:42.564813245] [8550]  INFO RPI vc4.cpp:455 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:57:42.564952059] [8550]  INFO RPI pipeline_base.cpp:1151 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:57:43.853248666] [8545]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:57:43.856270355] [8550]  INFO RPI vc4.cpp:630 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:57:44.998085269] [8550]  WARN V4L2 v4l2_videodevice.cpp:2095 /dev/video0[13:cap]: Dequeue timer of 1000000.00us has expired!
[0:57:44.998173620] [8550] ERROR RPI pipeline_base.cpp:1412 Camera frontend has timed out!
[0:57:44.998187528] [8550] ERROR RPI pipeline_base.cpp:1413 Please check that your camera sensor connector is attached securely.
[0:57:44.998200009] [8550] ERROR RPI pipeline_base.cpp:1414 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
```
I have tried to remove and reinstall libcamera and yet the same problem the camera is detected but can't capture a photo

naushir, 2024-11-06:
Can you tell me the exact steps you used to install/re-installed libcamera?  I don't recognise the version id string.

```
[0:57:44.998173620] [8550] ERROR RPI pipeline_base.cpp:1412 Camera frontend has timed out!
[0:57:44.998187528] [8550] ERROR RPI pipeline_base.cpp:1413 Please check that your camera sensor connector is attached securely.
[0:57:44.998200009] [8550] ERROR RPI pipeline_base.cpp:1414 Alternatively, try another cable and/or sensor.
```

This suggests that it's very likely due to a hardware problem.  Either the camera board, the connector cable, or the connector on the Raspberry Pi might be faulty.  But without knowing what changes you (or somebody) else has made to the libcamera libraries, I cannot say for certain.

Aly-Abdel-Motaleb, 2024-11-06:
I have not made any change I just removed it and re downloaded it using these commands 
`sudo apt remove -y libcamera-apps`
`sudo apt install -y libcamera-apps`

naushir, 2024-11-06:
I expect you have installed Arducam's version of libcamera based on the version string shown.  This may not be the issue itself though.   I think your camera hardware might not be functional.  I suggest reaching out to your supplier to get a replacement.

Aly-Abdel-Motaleb, 2024-11-06:
I have 2 different cameras and 3 different connectors all of which give the same error. I don't think the problem is hardware problem as it identify the camera itself in the list cameras command.

naushir, 2024-11-06:
The timeout typically indicates a hardware issue, regardless of whether the camera is detected - that happens across a different bus interface.  It could be a power supply issue as well.  I very much doubt this is a software problem, but it's hard to say as you don't seem to be running Raspberry Pi supplied libcamera libraries.

Another potential issue could be the Raspberry Pi itself.  Do you have access to an official Raspberry Pi camera that you can try out?

Aly-Abdel-Motaleb, 2024-11-06:
The problem that I am in Egypt so there in no access for the official raspberry pi camera as it is expensive to be sold here.
I am so frustrated as I have captured a photo with it after I have updated the OS , out of the blue it has stopped giving that error.

naushir, 2024-11-06:
Please get in touch with Arducam support - I don't know what changes they have made to their libcamera packages. Unfortunately, because of this I cannot help on this issue any more.   If it's not a hardware issue, it's likely that their software libraries are broken.

closed 2024-11-06 by naushir

Aly-Abdel-Motaleb, 2024-11-06:
I have downloaded the raspberry pi libcamera so what is the problem ?

naushir, 2024-11-06:
Your version string suggests that you are not running a Raspberry Pi built library.  Either way, to repeat myself, this is very unlikely to be a software issue.  This is why I am unable to help any further.  Sorry about that.

noviceiii, 2024-12-22:
I may step in with the same problem. Further, I was able to test it with another raspii, cable and camera with the same result: its rather not an hardware defect. I tryed with libcamera-apps and ibcamera-apps-lite.

```
uname -a
Linux cam 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

```
libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [46.34 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [32.81 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

```

```
libcamera-hello -n
[0:16:11.842183882] [793]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[0:16:11.891451632] [796]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:16:11.894297186] [796]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[0:16:11.894423313] [796]  INFO RPI pipeline_base.cpp:1120 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:16:11.895594316] [793]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:16:11.895959774] [796]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:16:13.000141374] [796]  WARN V4L2 v4l2_videodevice.cpp:2127 /dev/video0[13:cap]: Dequeue timer of 1000000.00us has expired!
[0:16:13.000219355] [796] ERROR RPI pipeline_base.cpp:1366 Camera frontend has timed out!
[0:16:13.000237021] [796] ERROR RPI pipeline_base.cpp:1367 Please check that your camera sensor connector is attached securely.
[0:16:13.000252187] [796] ERROR RPI pipeline_base.cpp:1368 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
```

```
uname -a
Linux cam 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

nandan-06, 2025-03-13:
I am trying to connect my RPi cam with RPi CM4 and waveshare nano baseboard. I updated the config file:

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
# camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtoverlay=ov5647,cam0
gpu_mem=128
start_x=1
```

Libcamera commands are working perfectly but the ffmpeg commands are giving errors.
I listed my camera devices and /dev/video0 is listed under unicam. Here it is:
```
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media3

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media1
        /dev/media2

unicam (platform:fe800000.csi):
        /dev/video0
        /dev/media4

rpivid (platform:rpivid):
        /dev/video19
        /dev/media0
```

Here is the ffmpeg error:
```
[video4linux2,v4l2 @ 0x558c9e5490] ioctl(VIDIOC_G_PARM): Inappropriate ioctl for device
[video4linux2,v4l2 @ 0x558c9e5490] Time per frame unknown
[video4linux2,v4l2 @ 0x558c9e5490] ioctl(VIDIOC_STREAMON): Invalid argument
/dev/video0: Invalid argument
```
please let me know if you need any other config details to help me debug

6by9, 2025-03-13:
> I am trying to connect my RPi cam with RPi CM4 and waveshare nano baseboard. I updated the config file:

Unrelated to the original issue which was on a Pi4, and the issue is closed which generally means no further action will be taken on it. 

> Libcamera commands are working perfectly but the ffmpeg commands are giving errors. I listed my camera devices and /dev/video0 is listed under unicam. Here it is:
>
> ```
> bcm2835-codec-decode [...]

FFmpeg doesn't support libcamera and is trying to use the raw image sensor to get processed images.