raspberrypi/libcamera #206: libcamera c++ api not working with raspberry pi 5 and pi camera

AAleka, 2024-11-10:
I have a raspberry pi 5 and a pi camera (IMX296). I cannot get them to work with libcamera c++ api (I installed it with apt install libcamera-dev). It shows one frame and then I get the following errors:

[1:38:18.970107448] [2487]  WARN V4L2 v4l2_videodevice.cpp:2095 /dev/video4[15:cap]: Dequeue timer of 1000000.00us has expired!
[1:38:18.970175948] [2487] ERROR RPI pipeline_base.cpp:1364 Camera frontend has timed out!
[1:38:18.970181319] [2487] ERROR RPI pipeline_base.cpp:1365 Please check that your camera sensor connector is attached securely.
[1:38:18.970186652] [2487] ERROR RPI pipeline_base.cpp:1366 Alternatively, try another cable and/or sensor.

However, I tested the cable, camera and my code on a raspberry pi zero and it works fine. Also I tested my pi 5 and a camera with libcamera-apps and it works fine. I am using the original pi 5 power supply, I tried changing the /boot/firmware/config.txt (camera_auto_detect=0/1, dtoverlay=imx296,cam0) but it still does not work.

naushir, 2024-11-11:
Do you happen to have any other camera that you can test with your Pi 5?  Can you provide the exact command line you are running?  Also, can you run ``camera-bug-report`` and paste the output file generated here?

AAleka, 2024-11-11:
@naushir so, when I run libcamera-apps (```libcamera-hello```) in the terminal it works. But when I try to use the C++ API it shouw the first frame and throws the errors above. Unfortunately, I don't have any other Pi camera but I have the Pi Zero and I successfully tested the API on it, so it can't be the cable or the camera issue. The output of ```camera-bug-report```:

```
--------------------------------------------------------------------------------
rpicam-apps Bug Report
--------------------------------------------------------------------------------
Date: 11-11-2024 (17:12:31)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
rpi5

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 1

Revision	: c04170
Serial		: e46803cb3ed37867
Model		: Raspberry Pi 5 Model B Rev 1.0

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/firmware/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=05a4a276-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=TR
** cat /boot/firmware/config.txt **
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
[... 1116 lines cut ...]
--------------------------------------------------------------------------------
** uname -a **
Linux rpi5 6.6.59-v8-16k+ #1812 SMP PREEMPT Fri Nov  8 15:40:46 GMT 2024 aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
2024/11/05 12:38:12 
Copyright (c) 2012 Broadcom
version 3c4fc886 (release) (embedded)

** rpicam-hello --version **
rpicam-apps build: v1.5.2 a8ccf9f3cd9d 03-10-2024 (22:33:44)
libcamera build: v0.3.2+27-7330f29b

```

naushir, 2024-11-11:
> @naushir so, when I run libcamera-apps (`libcamera-hello`) in the terminal it works. But when I try to use the C++ API it shouw the first frame and throws the errors above.

Could you clarify this please - so if you run ``rpicam-hello``/``libcamera-hello`` on a Pi 5, everything works as expected with the camera?  When you mean the C++ API, is this an application you have written yourself? Can you share some source code?

AAleka, 2024-11-11:
Yes, running ```rpicam-hello```/```libcamera-hello``` from the terminal works as expected. But the C++ code that I wrote myself (from the libcamera's documentation) does not work. Here is my code:
```
#include <opencv2/opencv.hpp>
#include <libcamera/libcamera.h>

using namespace libcamera;

struct OPTIONS
{
    unsigned int width = 1440;    // frame width
    unsigned int height = 1080;    // frame height
    float        fps = 60.0f;   // frames per second
    float        shutter_speed = 8000.0f; // exposure time (milliseconds)
    float        threshold_value = 180.0f;  // threshold value
    float        color_gains = 0.5f;    // white balance
    int          autofocus = 0;       // enable/disable autofocus 
    int          stride = 0;       // stride value
    double       penalty = 10.0;    // dynamic programming penalty
} options;

static std::shared_ptr<Camera> camera;
static std::condition_variable condition_variable;
static std::mutex mutex;
static std::atomic<bool> is_running(true);

static void requestComplete(Request *request);
void signalHandler(int signum);

int main(int argc, char **argv) 
{
    std::unique_ptr<CameraManager> cm = std::make_unique<CameraManager>();
    cm->start();

    if (cm->cameras().empty()) 
    {
        std::cerr << "No cameras available\n";
        cm->stop();
        return EXIT_FAILURE;
    }

    camera = cm->get(cm->cameras()[0]->id());
    if (!camera) 
    {
        std::cerr << "Failed to get the camera\n";
        cm->stop();
        return EXIT_FAILURE;
    }

    camera->acquire();
    std::unique_ptr<CameraConfiguration> config = camera->generateConfiguration({ StreamRole::VideoRecording });

    if (!config || config->size() != 1) 
    {
        std::cerr << "Failed to generate configuration\n";
        return -1;
    }

    StreamConfiguration &streamConfig = config->at(0);
    streamConfig.size.width = options.width;
    streamConfig.size.height = options.height;
    
    config->validate();
    std::cout << "Validated viewfinder configuration is: " << streamConfig.toString() << '\n';

    if (camera->configure(config.get()) < 0) 
    {
        std::cerr << "Failed to configure the camera\n";
        return -1;
    }

[... 115 lines cut ...]
    if (!is_running)
    {
        condition_variable.notify_one();
        return;
    }

    if (!request)
    {
        std::cerr << "Request is invalid\n";
        return;
    }

    request->reuse(Request::ReuseBuffers);
    camera->queueRequest(request);
}

void signalHandler(int signum)
{
    std::cout << "Interrupt signal (" << signum << ") received.\n";
    is_running = false;

    std::unique_lock<std::mutex> lock(mutex);
    condition_variable.notify_one();
}
```

naushir, 2024-11-11:
One thing I noticed is that you are updating your preview windows from the requestComplete handler.  This is probably not advisable as it holds the thread up for a long time.  Could you try commenting out the preview update and see if that helps.

Another thing to look at is the simple camera app: https://git.libcamera.org/libcamera/simple-cam.git which could be used as a skeleton application.

AAleka, 2024-11-11:
@naushir Thank you for your help, after removing the preview window it doesn't crash anymore. But I am curious why this happens, because it doesn't happen on the Pi Zero. Anyways, I guess it might be related to opencv then.

naushir, 2024-11-12:
It could be some timing bug and things get sequenced sufficiently different on the two platforms.  I would suggest you start with the simple camera app as a template and work from there.

naushir, 2024-11-21:
Closing now as we are not actively working on this issue.