raspberrypi/libcamera #208: bug: Segmentation fault error while running.

kiki-kanri, 2024-11-14:
Build with the latest main commit.

Kernel version used: 6.6.47+rpt-rpi-2712

Installed kernels:
```
linux-headers-6.6.47+rpt-rpi-2712/stable,now 1:6.6.47-1+rpt1 arm64 [installed]
linux-headers-6.6.51+rpt-rpi-2712/stable,now 1:6.6.51-1+rpt3 arm64 [installed,automatic]
linux-headers-rpi-2712/stable,now 1:6.6.51-1+rpt3 arm64 [installed]
linux-image-6.6.47+rpt-rpi-2712-dbg/stable,now 1:6.6.47-1+rpt1 arm64 [installed]
linux-image-6.6.47+rpt-rpi-2712/stable,now 1:6.6.47-1+rpt1 arm64 [installed]
linux-image-6.6.51+rpt-rpi-2712/stable,now 1:6.6.51-1+rpt3 arm64 [installed,automatic]
linux-image-rpi-2712/now 1:6.6.47-1+rpt1 arm64 [installed,upgradable to: 1:6.6.51-1+rpt3]
```

Error content:
```
root@raspberrypi:~/rpi-os/scripts/build-and-install-packages# gdb rpicam-hello -n
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rpicam-hello...
(No debugging symbols found in rpicam-hello)
(gdb) run
Starting program: /usr/local/bin/rpicam-hello
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[0:58:41.489866829] [75938]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+98-75fe515a
[New Thread 0x7ffff430eec0 (LWP 75940)]
[0:58:41.497490125] [75940]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 14-11-2024 (15:48:37)
[New Thread 0x7ffff3a3eec0 (LWP 75941)]
[New Thread 0x7ffff322eec0 (LWP 75942)]
[0:58:41.607838295] [75940]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_D0
Preview window unavailable
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:58:41.609430702] [75938]  INFO Camera camera.cpp:1197 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR_PISP_COMP1
[0:58:41.609572554] [75940]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[New Thread 0x7ffff0d8eec0 (LWP 75943)]
[New Thread 0x7fffebffeec0 (LWP 75944)]
[New Thread 0x7fffeac82ec0 (LWP 75945)]

Thread 3 "rpicam-hello" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff3a3eec0 (LWP 75941)]
0x00007ffff3ab3660 in libcamera::ipa::Pwl::eval(double, int*, bool) const () from /usr/local/lib/libcamera/ipa_rpi_pisp.so
(gdb) bt
#0  0x00007ffff3ab3660 in libcamera::ipa::Pwl::eval(double, int*, bool) const () from /usr/local/lib/libcamera/ipa_rpi_pisp.so
#1  0x00007ffff3aa3a10 in RPiController::Awb::prepareStats() () from /usr/local/lib/libcamera/ipa_rpi_pisp.so
#2  0x00007ffff3aa5320 in RPiController::Awb::doAwb() () from /usr/local/lib/libcamera/ipa_rpi_pisp.so
#3  0x00007ffff3aa5598 in RPiController::Awb::asyncFunc() () from /usr/local/lib/libcamera/ipa_rpi_pisp.so
#4  0x00007ffff7d7e9dc in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7b2ee90 in start_thread (arg=0x7ffff430cc27) at ./nptl/pthread_create.c:442
#6  0x00007ffff7b97b1c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
```

kiki-kanri, 2024-11-14:
Using tag [v0.3.2+rpt20240927](https://github.com/raspberrypi/libcamera/releases/tag/v0.3.2%2Brpt20240927) build works fine!

naushir, 2024-11-14:
I've not been able to reproduce this in my brief testing. I'll try again tomorrow and report back.

naushir, 2024-11-15:
@kiki-kanri are you using a custom/modified json tuning file?  I still cannot seem to reproduce this bug.

kiki-kanri, 2024-11-15:
> @kiki-kanri are you using a custom/modified json tuning file? I still cannot seem to reproduce this bug.

I don't use json tuning files, but I do set up the compilation and linking parameters, I'll test it on a clean system sometime.

Here's the program in order and compiled by me:

- openssl
- x264
- ffmpeg
- nlohmann-json
- libpisp
- libcamera
- rpicam-apps

naushir, 2024-11-15:
What compiler have you setup?

kiki-kanri, 2024-11-15:
> What compiler have you setup?

Here's the compilation and linking parameters that work for me on v0.3.2+rpt20240927.

cflags:

```
-fipa-pta
-floop-block
-floop-interchange
-floop-strip-mine
-flto=4
-fomit-frame-pointer
-funroll-loops
-I/opt/openssl3/include
-march=armv8.2-a
-mtune=cortex-a76
-O3
-pthread
```

ldflags: `-L/opt/openssl3/lib`

tom-ard, 2024-11-19:
I found that when using the new libcamera, changing the awb algorithm in the tuning data to a grayscale world and not setting "ct_curve", a "Segmentation fault" error will occur.

naushir, 2024-11-19:
@tom-ard thanks, I've reproduced this now!  Will debug and issue a fix shortly.

naushir, 2024-11-19:
Commit https://github.com/raspberrypi/libcamera/commit/ad9e3b917935bb96e921a83f9a80f560d9e00a3b ought to solve the crash.

kiki-kanri, 2024-11-19:
Tested and operated normally.

closed 2024-11-19 by kiki-kanri