raspberrypi/libcamera #209: IMX290 WDR support and low light.

DangerD1024, 2024-11-21:
Tried latest version of libcamera, WDR is not working, 
`libcamera-hello --metering average`
Looks a bit better but still not okay
`--hdr=auto` - WARN IPARPI ipa_base.cpp:1191 No HDR algorithm available

Also nothing is seen at low light enviroment, however videos on youtube shows that it should work fine at low light + WDR...

Is there any way to enable it or this functionality is not available?

naushir, 2024-11-21:
The sensor does not support HDR output, so you won't be able to enable HDR on a Pi 4 and CM 4.

On Pi 5, you can use the ISP HDR, but it seems like the software packages provided by your supplier don't have the HDR algorithm enabled.  You can do this manually by adding it into the tuning file, simply copy the below ``"hdr"`` block into the json tuning file:

```json
                "hdr":
                {
                    "sdn":
                    {
                        "deviation": 1.6,
                        "strength": 0.5,
                        "deviation2": 3.2,
                        "deviation_no_tdn": 3.2,
                        "strength_no_tdn": 0.75
                    },
                    "cdn":
                    {
                        "deviation": 200,
                        "strength": 0.3
                    },
                    "tdn":
                    {
                        "deviation": 1.3,
                        "threshold": 0.1
                    }
                },
```

It's probably better to ask your camera supplier to add this directly in their software.

closed 2024-11-21 by naushir

6by9, 2024-11-21:
The IMX290 driver has no HDR support.
Whilst the datasheet does mention multiple exposure and digital overlay WDR readout modes, I have no information on them. You'd need postprocessing to combine the images anyway.

The only sensor I'm aware of that has HDR in a manner that is supported is IMX708 (v3 camera module), and that is done on-sensor.

naushir, 2024-11-25:
> IMX290 can output 2 frames so they should be merged on RPI (60fps on 2 lanes, to make 30FPS HDR) Currently driver supports merging images on RPI side?

You can try enabling the ``"MultiExposure"`` mode as given in the above configuration example.  Please get in touch with your sensor vendor on enabling this feature, we will be happy to accept a PR and add it into our tree.

DangerD1024, 2024-11-25:
```
libcamera-hello --hdr --tuning-file /usr/local/share/libcamera/ipa/rpi/vc4/imx290.json
[0:05:36.932073718] [3099]  WARN RPiAgc agc.cpp:70 AGC channel 1 not available
[0:05:36.932301382] [3099]  WARN RPiHdr hdr.cpp:165 Channel 0 not found in mode SingleExposure
[0:05:37.399760939] [3099]  WARN RPiHdr hdr.cpp:186 Unexpected HDR mode
```
Looks like need to enable it somehow...

6by9, 2024-11-25:
> **On Pi 5,** you can use the ISP HDR, but it seems like the software packages provided by your supplier don't have the HDR algorithm enabled. You can do this manually by adding it into the tuning file, simply copy the below `"hdr"` block into the json tuning file:

> ```
> libcamera-hello --hdr --tuning-file /usr/local/share/libcamera/ipa/rpi/vc4/imx290.json
> [0:05:36.932073718] [3099]  WARN RPiAgc agc.cpp:70 AGC channel 1 not available
> [0:05:36.932301382] [3099] [...]

Use of vc4 would imply you're on Pi4 or below.

Please read https://forums.raspberrypi.com/viewtopic.php?t=372520 for a summary on 3rd party camera module support offered.
IMX290 is in category 2 - it has a mainline Linux driver and libcamera support, but support is not provided by Raspberry Pi for it. AFAIK no one has done a full image quality tuning for the sensor, and certainly HDR is out of scope.

DangerD1024, 2024-12-12:
I can enable HGC which improves quality in low-light:
`i2ctransfer -f -y 6 w3@0x1a 0x30 0x09 0x11`
This enables 2 Frame WDR:
`i2ctransfer -f -y 6 w3@0x1a 0x30 0x09 0x02`
But after that image became corruptes as driver not understand what sensor sends...
Is there any way i can merge 2 frames with libcamera?