raspberrypi/libcamera #211: dma-buf PAGE_SIZE Alignment?

schoolpost, 2024-11-22:
Raspberry Pi 5 is using 16K PAGE_SIZE. AFAIK the buffers generated in libcamera adhere to the minimum alignment specs of the PISP but do not necessary meet this 16K alignment; I believe it should be possible to get both aligned ( at the cost of some extreme padding in some cases. )  

How can I alter the padding that the PISP is adding to buffers such that it adheres to a 16K alignment? Specifically I would like to alter the RAW buffer to match this alignment. 

In the case of my UHD/4K image sensor this is going to mean the width/stride will align to 4096px wide @ 16-bit unpacked / 8192 byte stride. 

Without this alignment it is causing issues with other system components / hardware like the GPU which can also use imported dma-buf's.

schoolpost, 2024-11-29:
Adjusting buffer line stride worked perfect as you described. 

An extension of this topic, is it possible to request the RAW buffer with some extra rows of padding at the bottom?

Such example would be a buffer that is 4096x2160 ( 8192 stride ), I could request a buffer that is instead 4096x2176 ( adding 16 rows of empty padding )

naushir, 2024-12-02:
I think this should be possible.  In the allocation code, you should be able to provide any buffer size you want.  So you can add 16 lines to the calculated frame size and it should over-allocate the buffer.

https://github.com/raspberrypi/rpicam-apps/blob/6318ac74e44ad973d5170540c3818016a31bb94c/core/rpicam_app.cpp#L1015

closed 2024-12-02 by schoolpost