raspberrypi/libcamera #214: PISP_COMPRESSION_MODE_2 Support

schoolpost, 2024-11-28:
It would be great to have the simple companding mode supported in libcamera. 

I haven't extensively investigated this, but I did see some of the building blocks are set up for that to be switched on looking through the various source files? not sure what would be required on the rpicam-apps side of things either.

naushir, 2024-11-28:
@njhollinghurst for info.

What specific use case are you targeting here?  PISP_COMPRESSION_MODE_2 is probably not what you are after, but maybe something else in our pipeline might be?

njhollinghurst, 2024-11-28:
IIRC we use compression mode 1 exclusively. Mode 2 is a simple "sqrt-like" companding scheme.
I'm not sure why mode 2 is mentioned so often in the code (or why mode 3 is missing).

schoolpost, 2024-11-28:
> @njhollinghurst for info.
> 
> What specific use case are you targeting here? PISP_COMPRESSION_MODE_2 is probably not what you are after, but maybe something else in our pipeline might be?

Mostly for testing at the moment. ( image quality and general performance characteristics. ) 

Of course this can be done in software, but where there is an option to get this for "free" in hardware; I always like that. 

16-but unpacked RAW is quite a memory hog, especially at higher frame rates and higher resolutions; especially with the software based lossless compression I am applying to it; so it's twofold: I can halve the bandwidth requirements ( process 2x the number of pixels ) by getting 8-bit companded and also companding itself can lend to better compression ratios for the lossless compression. 

Most importantly the pixel format is left untouched. The only good working container for RAW that has any software interoperability in Adobe DNG, so I always need to conform to the standards described in their spec. I would love to just drop PISP COMP MODE 1 into a DNG file, but that would be non-standard and no software would be able to do anything with it. 

Adobe DNG ( TIFF ) has an option for a linearization table that can be included as a metadata tag to perform the inverse de-companding by the RAW reader. 

Now it remains to be seen if the PISP implementation ( in particular if the chosen intervals ) can perform to a acceptable quality standard ( @njhollinghurst seems to suggest otherwise ) but the concept itself is very well regarded and used in very high end imaging pipelines / applications. ( Many high end professional / prosumer camera's that can record RAW video include some form of companding to improve compression ratios and reduce storage requirements. ) In those cases usually it is companding in the form of 16-bit linear -> 12 bit log , or 12-bit linear to 10-bit log, etc... so again without having tested it's unclear whether 8-bit companding is still going to produce an acceptable image. It would be great if this companding could be user configurable ( set your own intervals )

njhollinghurst, 2024-11-29:
Unfortunately it's not possible to change the curve, and the compression block, when enabled, _always_ gives 8bpp output.

Early in development, the same curve was used for Mode 2 and Mode 3, and the flags were orthogonal. Later, Mode 3 was changed to a flatter curve (so as not to waste bits on values below the black level), but Mode 2 was retained. (There's also a Mode 0 which does nothing but truncate to 8bpp, only when the Compress block is enabled.)

The quality concern with this kind of "sqrt-like" curve is that the brightest values only have 7 effective bits, and middling values have 8 effective bits. Darker pixels have more precision (but not much more).

schoolpost, 2024-11-29:
That's too bad, oh well. I wouldn't mind trying MODE_2 as is either way; 

I can see the references here to it, and I've done some of my own mucking around changing flags where it currently sets MODE1 to MODE2, but I quickly see that there are missing references to the format/packing in many other source files; and in any of my attempts I can't get MODE2 output. 
https://github.com/raspberrypi/libcamera/blob/1230f78d2f6d38d812568fd27f7b985b9ff19e39/src/libcamera/pipeline/rpi/pisp/pisp.cpp#L140C1-L157C3

schoolpost, 2024-12-13:
Closing this issue, but would otherwise be interested to see if MODE2 could in one way or another be enabled for testing purposes.

closed 2024-12-13 by schoolpost