raspberrypi/libcamera #215: Camera Configuration Issues with Raspberry Pi 5 After Updates

codermery, 2024-11-28:
Hello Libcamera Team,

I am encountering configuration issues with the IMX708 camera module on a Raspberry Pi 5 running Ubuntu 24.04. Previously, the camera worked seamlessly with libcamera and camera_ros, but after a recent system reset and updates, the configuration fails consistently.

Key Details:
- Hardware: Raspberry Pi 5 + IMX708 Camera Module (2 units)
- Software: Ubuntu 24.04, ROS 2 Jazzy, Latest libcamera and camera_ros
- Errors:
  - `Can't configure camera with invalid configuration`
  - `failed to configure streams`

Despite these errors, QCam works perfectly, which suggests that libcamera is functional. However, when using ROS 2 or camera_ros, it fails to configure the camera streams.

Steps I've Tried:
1. Reinstalled all dependencies.
2. Verified camera functionality with QCam.
3. Tested multiple formats and resolutions, including `1640x1232-YUYV`.
4. Reverted to previous versions of libcamera and camera_ros, but the issue persists.

Questions:
1. Are there any compatibility issues with the recent updates to libcamera for Raspberry Pi 5 and the IMX708 module?
2. Is there a debugging process to identify and resolve the stream configuration errors?

Thank you for your support!

Best regards,

naushir, 2024-11-29:
There's nothing in this update that I can think of that should break this.  Can you provide some more information please:

1) Can you post the output of ``rpicam-hello --version``
2) Run your application with the following environment variable set: ``export LIBCAMERA_LOG_LEVELS=RPI:0`` and post the output log.

codermery, 2024-11-29:
# Issue with Libcamera and IMX708 Camera Module

Hello,

I would like to provide additional details about the challenges I'm facing with the IMX708 camera module and libcamera. Regardless of how or which version of libcamera I install, I have never been able to run **`libcamera-hello`** or **`rpicam-hello`**. Every time I attempt to use these commands, I receive the following error:

```text
command not found
```
Interestingly, the QCam application works perfectly, and I can capture images without any issues. This suggests that the hardware and basic libcamera functionality are operational. However, when I attempt to use camera_ros or other ROS 2 integrations, I encounter errors related to stream configuration.

Interestingly, the **QCam** application works perfectly, and I can capture images without any issues. This suggests that the hardware and basic libcamera functionality are operational. However, when I attempt to use `camera_ros` or other ROS 2 integrations, I encounter errors related to **stream configuration**.

## Summary of the Problem

1. **Libcamera Test Applications:**
   - `libcamera-hello` and `rpicam-hello` have never worked.
   - I consistently see the error **`command not found`**.

2. **QCam Functionality:**
   - The camera works flawlessly with QCam, indicating correct hardware setup.

3. **ROS 2 and camera_ros Issues:**
   - While using `camera_ros`, I encounter **stream configuration** errors.

## My Questions

- Without access to `libcamera-hello` or `rpicam-hello`, is there another way to validate compatibility between libcamera and `camera_ros`, considering QCam is working?
- Are there any specific steps I should follow to resolve this issue?
- Could this issue be related to a missing package or configuration for libcamera on Ubuntu 24.04 with Raspberry Pi 5?

I appreciate any guidance or suggestions you can provide.

Best regards,  
[Your Name]

naushir, 2024-11-29:
This sounds like you are not using Raspberry Pi provided packages for libcamera.  In such cases, maybe your package vendor removed support for the IMX708?  I would probably start by contacting the maintainers of the camera_ros packages and see where they obtained the packages from.

codermery, 2024-11-29:
# Directly Downloading Libcamera Sources from Raspberry Pi

Hello @naushir,

Thank you for your response and guidance regarding this issue. I would like to clarify a few points:

- I downloaded the libcamera sources directly from the [official Raspberry Pi GitHub repository](https://github.com/raspberrypi/libcamera).
- Additionally, I built libcamera using the following commands:
  ```bash
  git clone https://github.com/raspberrypi/libcamera.git
  cd libcamera
  meson build
  ninja -C build
  sudo ninja -C build install

naushir, 2024-11-29:
Have you setup the build parameters correctly?  From our [camera documentation](https://www.raspberrypi.com/documentation/computers/camera_software.html), you need to use this meson setup command:

```
meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled
```

codermery, 2024-11-29:
Now I tried the `meson setup` command as outlined in the camera documentation. But the result did not change. Here are the steps I followed:

## What I did:
1. I rebuilt libcamera with the following command:
```bash
meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled
ninja -C build
sudo ninja -C build install

naushir, 2024-11-29:
Have you purged the system installed libcamera libraries?  

Also can you your application with the following environment variable set: ``export LIBCAMERA_LOG_LEVELS=*:0`` and post the output log.

codermery, 2024-11-30:
LIBCANERA_LOG_LEVELS=*:DEBUG cam -l
```
[1:11:25.540318062] [5901] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[1:11:25.540429339] [5901] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[1:11:25.540470987] [5901]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0
[1:11:25.540642060] [5904] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[1:11:25.568831078] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[1:11:25.570020219] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 10 missing dependencies
[1:11:25.573099571] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[1:11:25.573152033] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[1:11:25.573346588] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[1:11:25.573655327] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[1:11:25.575633537] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[1:11:25.575648370] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[1:11:25.575769092] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[1:11:25.575779444] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[1:11:25.576069646] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[1:11:25.576092034] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[1:11:25.576450699] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[1:11:25.576489069] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 9 missing dependencies
[1:11:25.576858363] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media4
[1:11:25.576922789] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media4 due to 9 missing dependencies
[1:11:25.579778716] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[1:11:25.579868216] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[1:11:25.583751767] [5904] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media4 found
[1:11:25.583789174] [5904] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media4: pispbe
[1:11:25.584485355] [5904] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'rpi/vc4'
[1:11:25.584546762] [5904] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
[1:11:25.584554836] [5904] DEBUG RPI vc4.cpp:178 Unable to acquire a Unicam instance
Available cameras:
```

naushir, 2024-12-02:
I suspect there may be something wrong with your kernel drivers.  libcamera is reporting that it cannot register the Pi 5 pipeline handler.  My output for the same command is show below.

```
[65:16:45.605809460] [5761] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_pisp.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so is signed
[65:16:45.605918016] [5761] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
[65:16:45.606021609] [5761] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so is signed
[65:16:45.606051850] [5761] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
[65:16:45.606063405] [5761]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2
[65:16:45.606204998] [5762] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[65:16:45.613689842] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media1
[65:16:45.613895491] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 10 missing dependencies
[65:16:45.615056715] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[65:16:45.615068289] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: rp1-cfe
[65:16:45.615146196] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media3
[65:16:45.615155530] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[65:16:45.615278400] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[65:16:45.615286900] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: rpivid
[65:16:45.615376123] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media0
[65:16:45.615386826] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 8 missing dependencies
[65:16:45.615472604] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media2
[65:16:45.615482808] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 8 missing dependencies
[65:16:45.616379032] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[65:16:45.616403235] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: pispbe
[65:16:45.617272385] [5762] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[65:16:45.617281644] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: pispbe
[65:16:45.617616052] [5762] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'rpi/vc4'
[65:16:45.617651145] [5762] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[65:16:45.617659496] [5762] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[65:16:45.617671552] [5762] DEBUG Camera camera_manager.cpp:138 Found registered pipeline handler 'rpi/pisp'
[65:16:45.617686737] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "rp1-cfe"
[65:16:45.617705589] [5762] DEBUG DeviceEnumerator device_enumerator.cpp:318 Successful match for media device "pispbe"
```

You seem to be missing the ``Camera camera_manager.cpp:138 Found registered pipeline handler 'rpi/pisp'`` line.  I assume since you are using Ubuntu, you don't have the Raspberry Pi kernel installed?  If so, you won't be able to use the camera system unfortunately.

naushir, 2024-12-16:
Any updates on this issue?  If not, I'll close this shortly.

codermery, 2024-12-16:
Hi,
I’m currently not available to provide a detailed update. Please keep this thread open for now; I will follow up as soon as I can.

Thanks for understanding.

daniel-lopez-puig, 2025-02-12:
Hi, I have the same issue and in fact we are both using Ubuntu 24.04 (server in my case) @naushir . Is there a way to make it work with this OS? Camera module 2 is fully compatible but module 3 not? Do you know if it will be at some point?
Thanks!

naushir, 2025-02-13:
We don't directly provide any packages/libraries to Canonical for packaging in their Ubuntu release.  So unfortunately I cannot help here at all.  It's best to get in touch with their support channels to discuss how to proceed.

Closing this one down now...

codermery, 2026-01-17:
@Reyudo I abandoned that project. I'm no longer using Raspberry Pi 5, but rather NVIDIA Jetson for another project, and I'm not encountering such issues anymore. However, I've heard that the problem still persists, and I've received confirmation of this from many sources.

kbingham, 2026-01-17:
But if the problem persists in Ubuntu (Not Raspberry Pi OS) - someone needs to take the issue up with Ubuntu.

Has anyone reported this to Ubuntu ? If so - where ?