raspberrypi/libcamera #216: [BUG] libcamera has no Python support in Raspberry Pi OS Lite (any)

neuron-whisperer, 2024-12-05:
The canonical way to access libcamera in Python is through picamera2.

The GitHub repo for picamera2 includes alternate instructions for installing "without GUI packages" such as OpenGL and Qt, which is essential for running picamera2 in any version of Raspberry Pi Lite OS. Unfortunately, those instructions are completely wrong - the install packages depend upon PyQt5, and if it is unavailable and cannot be installed, then picamera2 fails to install. This problem has persisted for over a year - [https://github.com/raspberrypi/picamera2/issues/822] - and the package maintainers are uninterested in fixing it

As a result, libcamera is inaccessible in Python in any Raspberry Pi OS Lite, as well as any follow-on products or packages that depend on picamera2, including the new Sony IMX500 AI Camera.

naushir, 2024-12-06:
>Unfortunately, those instructions are completely wrong - the install packages depend upon PyQt5, and if it is unavailable and cannot be installed, then picamera2 fails to install. This problem has persisted for over a year - [https://github.com/[raspberrypi/picamera2/issues/822](https://github.com/raspberrypi/picamera2/issues/822)] - and the package maintainers are apparently uninterested in fixing it.

Firstly, please be aware that there are at most 2 or 3 engineers working on the *entire* imaging system for Raspberry Pi.  This includes hardware (sensor modules, ISP), software (libcamera, kernel drivers), applications (picamera2, rpicam-apps) and documentation.  As such, it may take us longer to act on some issues, please don't take that is us being "uninterested".

Now onto your issue.  Picamera2 is extremely featureful and as it has grown, there have been additional external dependencies added. I've not looked into the exact dependency chain yet, but to get you working immediately, is there a specific reason you are using Raspberry Pi OS Lite?  Have you considered just using a standard Raspberry Pi OS image and booting to the command line?  With this setup you should have no dependency issues, and have a fully functional imaging system on the command line.

davidplowman, 2024-12-06:
The Picamera2 situation changed with the Pi 5 because it caused the installation package to become much larger, so that it's no longer included in the Lite images by default (as it used to be). By and large this is not reflected correctly in the documentation - so I'll see if I can get that updated.

Beyond that, the recommended way to install Picamera2 on a Lite image - `sudo apt install python3-picamera2` or `sudo apt install python3-picamera2 --no-install-recommends` both work correctly so far as I am aware across all platforms.

We do not recommend installing Picamera2 using pip because libcamera has no pip package and it's very easy to install a version of Picamera2 that doesn't match your libcamera installation. Moreover, in my experience, installing PyQt5 through pip usually fails, and it's very easy to end up in version hell there too. More on all this [here](https://forums.raspberrypi.com/viewtopic.php?t=361758). Anyway, all the more reason to install from apt - you get stuff that's been tested and confirmed as working together.

If you have any specific things that you've tried, and which didn't work, please do report the precise set of commands that will reproduce it, and we can take a look.

neuron-whisperer, 2024-12-08:
> Is there a specific reason you are using Raspberry Pi OS Lite?

Yes, plenty of reasons.

The projects that I undertake tend to be "kiosk-style" applications: one application running full-screen in Python and nothing else. In that context:

1) Many of my projects run on low-spec devices such as the Raspberry Pi Zero/Zero 2. On these devices, the overhead of a GUI-based OS considerably slows down the device - booting takes a lot longer, and anything that the GUI-based OS is doing in the background slows down or stalls the device for literally no gain.

2) Some of my other projects run on the Raspberry Pi 5 and involve machine learning, and, as such, are maximally using all onboard device resources - core, memory, and caches. In those contexts, the GUI-based OS diminishes performance - the OS is constantly context-switching some cores away from the machine learning tasks. In addition to the immediate and frequent hitches, the context switching causes cache invalidation and shuffling data between RAM and swap memory. The result is significant performance drain for literally no gain.

3) Running a full-screen, single-app process in a GUI-based OS is, frankly, a pain in the ass. You need to configure it as a service that's run on startup. You need to grant it all kinds of permissions in the OS. And you need to deal with all kinds of GUI-based OS events like a background process popping up a message *on top of* your fullscreen app (e.g.: "A new version of Java is available!") None of that occurs in Raspberry Pi OS Lite.

4) GUI-based OSes also want to provide all kinds of "help" that just screw up the application - for instance: automatically updating system packages in the background, in a way that spontaneously breaks functionality. GUI-based OSes also mess with touch input in ways that overcomplicate simple tasks like coordinate detection.

I'm not the only user who has reasons to use Raspberry Pi OS Lite. For instance, [RetroPie is based on Raspberry Pi OS Lite](https://retropie.org.uk/docs/Manual-Installation/), presumably for the same reasons since RetroPie is commonly executed in arcade cabinets as a kiosk-style application.

Hopefully this explanation has persuaded you that people have legitimate reasons to run in Raspberry Pi OS Lite without the useless complexity and resource diversion of a GUI-based OS. Please let us know whether picamera2 will restore Raspberry Pi OS Lite support, or whether the plan is to abandon Lite and exclusively support Xwindows/Wayland-based environments.

neuron-whisperer, 2024-12-08:
> Beyond that, the recommended way to install Picamera2 on a Lite image - `sudo apt install python3-picamera2` or `sudo apt install python3-picamera2 --no-install-recommends` both work correctly so far as I am aware across all platforms.

apt can only install picamera2 in the system Python deployment. So all of these instructions drag in the requirement of using the system-wide version of Python, either directly or in a virtual environment.

It appears that picamera2 does not work with any version of Python except the system deployment. Is that correct?

That seems like an unreasonable requirement. Lots of people have reasons to install and use versions of Python other than the system install version. Maybe they need the feature set of a different version of Python than the system version; or maybe they need to use different packages than are installed for the system version.

I'm not aware of any other package with the requirement that it can _only_ be installed in the system Python deployment.

naushir, 2024-12-08:
> > Is there a specific reason you are using Raspberry Pi OS Lite?
> 
> Yes, plenty of reasons.
> 

And all the things you mentioned become non-issues if you set your standard Raspberry Pi OS to [boot to the console](https://www.raspberrypi.com/documentation/computers/configuration.html#boot-options) as I mentioned in my first response.  When you do this, there is no GUI running, and no resources taken up for the GUI.

naushir, 2024-12-08:
> It appears that picamera2 does not work with any version of Python except the system deployment. Is that correct?
>
> That seems like an unreasonable requirement. Lots of people have reasons to install [...]

Picamera2 uses the native libcamera python bindings.  These are built with pybind11 and linked with the system installed python version at build time.  It would be unfeasible for Raspberry Pi to package multiple builds of the libcamera python bindings for different versions of python.  If users have strict requirements for particular python versions, I assume they can build libcamera themselves for their particular environment.

What version of python are you running, why is this affecting you?  Or is this purely a hypothetical question?

neuron-whisperer, 2024-12-09:
> And all the things you mentioned become non-issues if you set your standard Raspberry Pi OS to [boot to the [...]

Okay, look - I didn't open this GitHub issue to debate the relative merits of Raspberry Pi OS Lite vs. Full, or to solicit advice on which one I should choose.

I came here for two reasons:

(1) To alert the devs (i.e., you) that the instructions on the home page of the picamera2 GitHub repository are objectively incorrect *and have been for over a year* in a way that is causing trouble for users, and

(2) To ask why picamera2 can't run in Raspberry Pi OS Lite without dragging in resources from Raspberry Pi OS Full, for which the answer appears to be: because one of the two developers has very strong personal opinions against Raspberry Pi OS Lite and would rather urge users to migrate to Raspberry Pi OS Full than to address serious compatibility issues with Raspberry Pi OS Lite *that picamera2 purportedly supports.*

If you're committed to the choice of abandoning Raspberry Pi OS Lite for this project, then that's totally your choice since it's your repo. But at least be open and honest with your users about this requirement - as well as the rest of the Raspberry Pi OS team, who might also have strong feelings about the ongoing viability of key infrastructure componens of Raspberry Pi OS Lite.

naushir, 2024-12-09:
> If you're committed to abandoning Raspberry Pi OS Lite for this project, then that's totally your choice since it's your repo. But at least be open and honest with your users about this requirement - as well as the rest of the Raspberry Pi OS team, who might also have strong feelings about the future of key infrastructure components of Raspberry Pi OS Lite.

Why do you insist on repeating this false statement?  Across your Reddit post and multiple forums posts users have stated that they have been running cameras fine in Raspberry Pi OS lite, and via the command line without a GUI.  In this issue you've been given muliple ways of running the camera without a GUI, *including* installing Picamera2 on Lite without the additional dependencies.  Yet you choose to ignore these, I'm not sure what your agenda is!

I'm done with engaging with you now.

closed 2024-12-09 by naushir