raspberrypi/libcamera #220: Issues with libcamera on pi5

fishrockz, 2024-12-12:
I tried building the current next branch of libcamera

this is how i configured it https://gitlab.com/gorse/heather-linux/-/blob/willsalmon/libcamera-0.4.0/elements/libcamera.grd?ref_type=heads#L159

the pi5 had a v3 and a v3 noir attached

im building a pretty recent kenel https://gitlab.com/gorse/heather-linux/-/blob/willsalmon/libcamera-0.4.0/elements/kernel_pi5.grd?ref_type=heads#L36 with pertty much the instuctions from https://www.raspberrypi.com/documentation/computers/linux_kernel.html and pertty upto date firmware https://gitlab.com/gorse/heather-linux/-/blob/willsalmon/libcamera-0.4.0/elements/firmware_pi5.grd?ref_type=heads#L15

LIBCAMERA_LOG_LEVELS=*:0 cam --list

gives:

```bash
 [470] DEBUG IPAModule ipa_module.cpp:333 ipa_rpi_vc4.so: IPA module /usr/lib/libcamera/ipa_rpi_vc4.so is signed
[0:06:32.501846210] [470] DEBUG IPAManager ipa_manager.cpp:235 Loaded IPA module '/usr/lib/libcamera/ipa_rpi_vc4.so'
[0:06:32.501859358] [470]  INFO Camera camera_manager.cpp:327 libcamera v0.3.2
[0:06:32.501984988] [471] DEBUG Camera camera_manager.cpp:73 Starting camera manager
[0:06:32.502091414] [471] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rpivid" created from /dev/media4
[0:06:32.502196803] [471] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media4: rpivid
[0:06:32.502253914] [471] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media2
[0:06:32.502468970] [471] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media2: rp1-cfe
[0:06:32.502515192] [471] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "rp1-cfe" created from /dev/media0
[0:06:32.502729303] [471] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: rp1-cfe
[0:06:32.502768173] [471] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media3
[0:06:32.502905007] [471] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media3: pispbe
[0:06:32.502939488] [471] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "pispbe" created from /dev/media1
[0:06:32.503076173] [471] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: pispbe
[0:06:32.503087451] [471] DEBUG Camera camera_manager.cpp:140 Found registered pipeline handler 'rpi/vc4'
[0:06:32.503102710] [471] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
[0:06:32.503108636] [471] DEBUG RPI vc4.cpp:189 Unable to acquire a Unicam instance
Available cameras:
```

Do you have any advice on how to debug why the cameras are not showing up?

for reference i am able to get pi4 and pizeros to work with a older libcamera eg https://gitlab.com/gorse/pi-camera

kbingham, 2024-12-12:
-Dpipelines=rpi/vc4 -Dipas=rpi/vc4 

You have not enabled the rpi/pisp IPA and Pipeline handler
I think you could just leave -Dpipelines alone and let it run as auto ? Otherwise you need to add the pi5 pipeline handler too.

fishrockz, 2024-12-12:
Thanks @kbingham 

So Adding rpi/pisp to ipas and pipeline did the trick!

But i tried `-Dpipelines=auto` however because i didn't have libpisp it silently dropped rpi/pisp, setting -Dpipelines=rpi/vc4,rpi/pisp triggered the missing lib error letting me know i had to add it.

naushir, 2024-12-13:
``libpisp`` should be downloaded and built as part of a subproject if the library is not found on the system.  Would you be able to share the output of a fresh ``meson configure`` step please?

fishrockz, 2024-12-13:
> `libpisp` should be downloaded and built as part of a subproject if the library is not found on the system. Would you be able to share the output of a fresh `meson configure` step please?

So letting meson download things from the internet makes any attempt at meaning full:
* License checking
* Software bil of materials
* reproducible builds

Very hard as you have to try and work out what meson did in addition to your own recorded sources.

> this is how i configured it https://gitlab.com/gorse/heather-linux/-/blob/willsalmon/libcamera-0.4.0/elements/libcamera.grd?ref_type=heads#L159

The sources section of this file would be pretty meaning less if i let meson just download more stuff

There for when building [heather linux](https://gitlab.com/gorse/heather-linux) I use [girderstream](https://gitlab.com/girderstream/girderstream) which blocks internet access when building (much like any other serious OS does)

If i was just building libcamera for development then im sure the feature might be helpful..

naushir, 2024-12-13:
Fair enough.  I assume this issue can now be resolved then.  If not, feel free to re-open.

closed 2024-12-13 by naushir