raspberrypi/libcamera #223: IMX708 framerate drops at 120fps

xhaakon, 2024-12-17:
* [Arducam IMX708](https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/12MP-IMX708)
* Raspberry Pi Compute Module 4
* libcamera v0.3.2+99-1230f78d

The camera should be capable of delivering 120 fps, however I'm observing the framerate dipping as low as 60 fps with `rpicam-vid`. The framerate dips are accompanied with "Dropping unmatched input frame" in libcamera logs.

```
LIBCAMERA_LOG_LEVELS=RPI:DEBUG rpicam-vid --width=1280 --height=720 --framerate=120 |& grep -e "Dropping unmatched" -e " exp "
[97:57:35.956132354] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[97:57:35.971381980] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[97:57:35.996355608] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#0 (0.00 fps) exp 7882.00 ag 5.99 dg 1.03
#1 (120.11 fps) exp 7882.00 ag 5.99 dg 1.03
#2 (120.13 fps) exp 7882.00 ag 5.99 dg 1.03
#3 (120.12 fps) exp 7882.00 ag 5.99 dg 1.03
#4 (120.11 fps) exp 7882.00 ag 5.99 dg 1.03
#5 (120.13 fps) exp 7882.00 ag 5.99 dg 1.03[97:57:36.096134883] [66332] DEBUG RPISTREAM rpi_stream.cpp:271 Queuing buffer 0 for ISP Input
#6 (120.11 fps) exp 7882.00 ag 5.99 dg 1.03
[97:57:36.118958618] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#7 (120.12 fps) exp 7882.00 ag 5.99 dg 1.03
#8 (60.06 fps) exp 7882.00 ag 5.99 dg 1.03
[97:57:36.143184313] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#9 (120.08 fps) exp 7882.00 ag 5.99 dg 1.03
#10 (60.07 fps) exp 7882.00 ag 5.99 dg 1.03
#11 (120.12 fps) exp 7882.00 ag 5.99 dg 1.03
[97:57:36.179084205] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#12 (120.13 fps) exp 7882.00 ag 5.99 dg 1.03
#13 (60.05 fps) exp 7882.00 ag 5.99 dg 1.03
[97:57:36.203201273] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#14 (120.18 fps) exp 7882.00 ag 5.99 dg 1.03
#15 (60.06 fps) exp 7882.00 ag 5.99 dg 1.03
[97:57:36.226155079] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#16 (120.11 fps) exp 7882.00 ag 5.99 dg 1.03
#17 (60.06 fps) exp 7882.00 ag 5.99 dg 1.03
[97:57:36.251890678] [66332] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
#18 (120.13 fps) exp 7882.00 ag 5.99 dg 1.03
#19 (60.06 fps) exp 7882.00 ag 5.99 dg 1.03
```

The same is happening with GStreamer:

```
[79:38:16.975247529] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[79:38:17.002330521] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3623, dropped: 0, current: 113.94, average: 107.01
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3684, dropped: 0, current: 120.53, average: 107.21
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3745, dropped: 0, current: 120.14, average: 107.40
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3806, dropped: 0, current: 120.13, average: 107.58
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3867, dropped: 0, current: 119.69, average: 107.76
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3928, dropped: 0, current: 120.55, average: 107.93
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 3989, dropped: 0, current: 120.12, average: 108.10
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 4050, dropped: 0, current: 120.06, average: 108.26
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 4111, dropped: 0, current: 119.75, average: 108.42
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 4172, dropped: 0, current: 120.55, average: 108.58
[79:38:22.117172325] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[79:38:22.177505796] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[79:38:22.206952106] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[79:38:22.256746706] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[79:38:22.324490573] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 4225, dropped: 0, current: 104.81, average: 108.53
[79:38:22.382861756] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
[79:38:22.421178924] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
...
79:38:22.847203001] [54076] DEBUG RPI vc4.cpp:1047 Dropping unmatched input frame in stream Unicam Embedded
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 4276, dropped: 0, current: 100.94, average: 108.43
```

These drops don't happen if I lower the framerate to 60fps. With rpicam-vid, I was able to prolong the initial period of consistent 120fps frame delivery by increasing --buffer-count up to 80, but it helped only for a few seconds before fps drops reappeared.

Throughout the testing CPU utilization stayed around 30%.

Please let me know what kind of information and logs can I provide to help investigate the issue.

naushir, 2024-12-17:
First thing I would suggest is to look at the options and settings needed for high frame rate capture on VC4 based platforms: https://www.raspberrypi.com/documentation/computers/camera_software.html#capture-high-framerate-video

naushir, 2025-01-06:
Any update on this?  If not, I'll close this down shortly.

xhaakon, 2025-01-06:
Thanks, changing the CPU frequency scaling governor from ondemand to performance seems to fix the framerate issue. We can close this.

closed 2025-01-06 by xhaakon