raspberrypi/libcamera #226: compiler warnings introduced recently.

adminy, 2025-01-02:
### Suppressing with: `-Wno-sign-compare`
```
       > [136/219] Compiling C++ object src/libcamera/libcamera.so.0.3.2.p/pipeline_rpi_pisp_pisp.cpp.o
       > [137/219] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx708.cpp.o
       > [138/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/algorithm.cpp.o
       > [139/219] Compiling C++ object src/libcamera/libcamera.so.0.3.2.p/v4l2_subdevice.cpp.o
       > [140/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/controller.cpp.o
       > [141/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_black_level.cpp.o
       > [142/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_cac.cpp.o
       > [143/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_af.cpp.o
       > [144/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_agc.cpp.o
       > [145/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_alsc.cpp.o
       > [146/219] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/imx500_tensor_parser_imx500_tensor_parser.cpp.o
       > ../src/ipa/rpi/cam_helper/imx500_tensor_parser/imx500_tensor_parser.cpp: In function 'int {anonymous}::parseInputTensorBody(RPiController::IMX500InputTensorInfo&, const uint8_t*, const InputTensorApParams&, const DnnHeader&)':
       > ../src/ipa/rpi/cam_helper/imx500_tensor_parser/imx500_tensor_parser.cpp:589:42: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
       >   589 |                         if (((pixelIndex == inputApParams.width * inputApParams.height) ||
       >       |                               ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       > ../src/ipa/rpi/cam_helper/imx500_tensor_parser/imx500_tensor_parser.cpp:590:42: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
       >   590 |                              (pixelIndex == inputApParams.width * inputApParams.height * 2) ||
       >       |                               ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       > ../src/ipa/rpi/cam_helper/imx500_tensor_parser/imx500_tensor_parser.cpp:591:42: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
       >   591 |                              (pixelIndex == inputApParams.width * inputApParams.height * 3))) {
       >       |                               ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       > [147/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_awb.cpp.o
       > [148/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_agc_channel.cpp.o
       > [149/219] Compiling C++ object src/ipa/rpi/common/librpi_ipa_common.a.p/ipa_base.cpp.o
       > ninja: build stopped: subcommand failed.
```
### Suppressing with `-Wno-stringop-truncation`
```
       > [133/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/device_status.cpp.o
       > [134/219] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o
       > FAILED: src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o
       > g++ -Isrc/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p -Isrc/ipa/rpi/cam_helper -I../src/ipa/rpi/cam_helper -Isrc/ipa/rpi -I../src/ipa/rpi -Iinclude -I../include -Iinclude/libcamera/ipa -Iinclude/libcamera -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -O3 -Wno-redundant-move -Wmissing-declarations -Wshadow -include /build/source/build/config.h -Wno-sign-compare -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o -MF src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o.d -o src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o -c ../src/ipa/rpi/cam_helper/cam_helper_imx500.cpp
       > ../src/ipa/rpi/cam_helper/cam_helper_imx500.cpp: In member function 'void CamHelperImx500::parseInferenceData(libcamera::Span<const unsigned char>, RPiController::Metadata&)':
       > ../src/ipa/rpi/cam_helper/cam_helper_imx500.cpp:259:40: error: 'char* __builtin_strncpy(char*, const char*, long unsigned int)' specified bound 64 equals destination size [-Werror=stringop-truncation]
       >   259 |                                 strncpy(exported.networkName, inputTensorInfo.networkName.c_str(),
       >       |                                        ^
       > cc1plus: all warnings being treated as errors
       > [135/219] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx708.cpp.o
       > [136/219] Compiling C++ object src/libcamera/libcamera.so.0.3.2.p/pipeline_rpi_pisp_pisp.cpp.o
       > [137/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/algorithm.cpp.o
       > [138/219] Compiling C++ object src/libcamera/libcamera.so.0.3.2.p/v4l2_subdevice.cpp.o
       > [139/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/controller.cpp.o
       > [140/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_black_level.cpp.o
       > [141/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_cac.cpp.o
       > [142/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_af.cpp.o
       > [143/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_ccm.cpp.o
       > [144/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_agc.cpp.o
       > [145/219] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/imx500_tensor_parser_imx500_tensor_parser.cpp.o
       > [146/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_alsc.cpp.o
       > [147/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_awb.cpp.o
       > [148/219] Compiling C++ object src/ipa/rpi/controller/librpi_ipa_controller.a.p/rpi_agc_channel.cpp.o
       > [149/219] Compiling C++ object src/ipa/rpi/common/librpi_ipa_common.a.p/ipa_base.cpp.o
       > ninja: build stopped: subcommand failed.
```
Rather than suppressing the warnings lets fix the actual potential issues.

naushir, 2025-01-06:
As far as I know we don't suppress any warnings in the libcamera build.  I also don't see these warnings in my local gcc/clang compilers.  Can you tell me what compiler you are using?

adminy, 2025-01-08:
gcc (GCC) 14.2.1 20241116

solsticedhiver, 2025-04-23:
I can confirm I am seeing the problem on archlinuxarm (armv7, well a systemd-nspawn container if you want to know), with gcc 14.2.1 20250207:

```
[167/535] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o
FAILED: src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o 
c++ -Isrc/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p -Isrc/ipa/rpi/cam_helper -I../libcamera/src/ipa/rpi/cam_helper -Isrc/ipa/rpi -I../libcamera/src/ipa/rpi -Iinclude -I../libcamera/include -Iinclude/libcamera/ipa -Iinclude/libcamera -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c++17 -Wno-redundant-move -Wno-psabi -Wmissing-declarations -Wshadow -include /home/alarm/pkg/PKGBUILDs/extra/libcamera/src/build/config.h -march=armv7-a -mfloat-abi=hard -mfpu=neon -O2 -pipe -fstack-protector-strong -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fno-omit-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -fPIC -DLIBCAMERA_BASE_PRIVATE -MD -MQ src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o -MF src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o.d -o src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx500.cpp.o -c ../libcamera/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp
distcc[161274] (dcc_build_somewhere) Warning: failed to distribute, running locally instead
../libcamera/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp: Dans la fonction membre « void CamHelperImx500::parseInferenceData(libcamera::Span<const unsigned char>, RPiController::Metadata&) »:
../libcamera/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp:248:40: erreur: « char* __builtin_strncpy(char*, const char*, unsigned int) » la limite spécifiée 64 est égale à la taille de la destination [-Werror=stringop-truncation]
  248 |                                 strncpy(exported.networkName, inputTensorInfo.networkName.c_str(),
      |                                        ^
cc1plus : tous les avertissements sont traités comme des erreurs
distcc[161274] ERROR: compile ../libcamera/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp on localhost failed
[168/535] Compiling C++ object src/ipa/rpi/cam_helper/librpi_ipa_cam_helper.a.p/cam_helper_imx519.cpp.o

```
I had to use the CXXFLAGS given above

naushir, 2025-04-24:
The following patch should fix the ``-Wno-stringop-truncation`` error

```diff
diff --git a/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp b/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp
index 9a5623f14ac5..6b12f3021afd 100644
--- a/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp
+++ b/src/ipa/rpi/cam_helper/cam_helper_imx500.cpp
@@ -246,7 +246,7 @@ void CamHelperImx500::parseInferenceData(libcamera::Span<const uint8_t> buffer,
                                exported.height = inputTensorInfo.height;
                                exported.numChannels = inputTensorInfo.channels;
                                strncpy(exported.networkName, inputTensorInfo.networkName.c_str(),
-                                       sizeof(exported.networkName));
+                                       sizeof(exported.networkName) - 1);
                                exported.networkName[sizeof(exported.networkName) - 1] = '\0';
                                metadata.set("cnn.input_tensor_info", exported);
                                metadata.set("cnn.input_tensor", std::move(inputTensorInfo.data));
@@ -279,7 +279,7 @@ void CamHelperImx500::parseInferenceData(libcamera::Span<const uint8_t> buffer,
                                        << "IMX500 output tensor info export failed, numTensors > MaxNumTensors";
                        }
                        strncpy(exported.networkName, outputTensorInfo.networkName.c_str(),
-                               sizeof(exported.networkName));
+                               sizeof(exported.networkName) - 1);
                        exported.networkName[sizeof(exported.networkName) - 1] = '\0';
                        metadata.set("cnn.output_tensor_info", exported);
                        metadata.set("cnn.output_tensor", std::move(outputTensorInfo.data));
```

However, I'm not sure about what's going on with the ``-Wno-sign-compare`` error.  As far as I can tell, both LHS and RHS are unsigned values, so everything should be ok.  Perhaps the code has changed since the error was first reported?

@solsticedhiver can you test the above patch and report it it fixes your compiler warning?

solsticedhiver, 2025-04-24:
that worked fine.

naushir, 2025-04-25:
Change has now been merged (https://github.com/raspberrypi/libcamera/commit/2993e7953e2f3ce374fd1d5a5adde15ce6918692) and will make its way into the next release.