raspberrypi/libcamera #227: Raspberrypi 5 / Picamera3 - Module Not Found Error on Python 3.12 (OS: Ubuntu 24.04)

alperrkilic, 2025-01-02:
Hi,

I'm trying to run picamera3 on a Raspberry Pi 5 running Ubuntu 24.04. I've manually downloaded and installed the necessary documentation according to the instructions on this page: https://libcamera.org/getting-started.html. I've also followed the instructions on the official raspberrypi page (https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-building-libcamera), but I couldn't solve the problem.

libcamera is installed on /home/sentinel/Desktop/libcamera, I can run it with the "libcamera-hello" command, but I can't use it in Python.

Error Message: 
-----------------
Traceback (most recent call last):
  File "/home/sentinel/Desktop/main.py", line 1, in <module>
    from picamera2 import Picamera2
  File "/home/sentinel/.local/lib/python3.12/site-packages/picamera2/_init_.py", line 6, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/sentinel/.local/lib/python3.12/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/sentinel/.local/lib/python3.12/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Rectangle, Size
ImportError: cannot import name 'ControlType' from 'libcamera' (unknown location)

Code:
----------------

`from picamera2 import Picamera2`
`import time`

`picam2 = Picamera2()`
`picam2.start()`
`picam2.stop()`

`print("Image captured and saved as image.jpg")`

If you need more information I can provide it. Thank you in advance for your help.

naushir, 2025-01-06:
It looks to be some kind of mismatch between the libcamera installation and the python binding (or picamera2).  Can I ask how you installed picamera2 on your Ubuntu environment?  Also, do you need to use Ubuntu?  This should "just work" on Raspberry Pi OS if you are able to use that instead.

naushir, 2025-01-16:
Any update on this?  If not I'll close down shortly.

alperrkilic, 2025-01-16:
Thanks for your help. I wasn't able to solve it with Python. But, solved the issue by using the library written in C++.

closed 2025-01-16 by naushir