raspberrypi/libcamera #233: Flash light

magni-optitog, 2025-01-31:
Hi, maybe this is a feature I am not aware of but would like some pointers. 

I have a ov2311 camera and on it there is an FSTROBE pinout, also XVS, SIOC, SIOD, GPO. 

Does anyone know how to connect a flashing light that activates when the camera is taking a photo? I would like to connect a LED to it to act as a flash light, only lighting up when the camera is taking a photo.

Would appreciate all help.

Thanks.

6by9, 2025-01-31:
The V4L2 subsystem hasn't devised a common way of configuring any form of GPIO or synchronisation connection from image sensors at present.

OV2311 was also a driver I wrote to work with an Arducam module I had. However that module has died, and Arducam now only make a Pivariety version of the module. PiVariety is very definitely in Arducam's court to deal with all support requests.

naushir, 2025-01-31:
Need to discuss this with you sensor vendor unfortunately.

closed 2025-01-31 by naushir