raspberrypi/libcamera #234: GStreamer IMX296 GS camera problem with resolution 1456x1088

rafalmaciag, 2025-02-01:
Hi,

Ive tested **GS IMX296** camera with the latest gstreamer plugin (the one with added controls). It seems that:
1) If we let gstreamer to pick the resolution by itself, it defaults to 1280x1080. 

```bash
gst-launch-1.0 libcamerasrc ! video/x-raw,format=I420,framerate=30/1 ! jpegenc !  tcpserversink host=0.0.0.0 port=5000 sync=false
```
I'm not against picking a different resolution, although I would expect to pick native: 1456x1088

2) When you specify width and height:

```
gst-launch-1.0 libcamerasrc ! video/x-raw,format=I420,width=1456,height=1088,framerate=30/1 ! jpegenc !  tcpserversink host=0.0.0.0 port=5000 sync=false
```

It starts the pipeline successfully, but the image is corrupted - lots of horizontal lines (under a little angle) - it looks like a problem with padding or metadata. (might be somewhere gst_libcamera_src_negotiate and GstVideoFormat ???)

Here's what the camera list look like:
```
libcamera-vid --list-cameras
Available cameras
-----------------
0 : imx296 [1456x1088 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx296@1a)
    Modes: 'SRGGB10_CSI2P' : 1456x1088 [60.38 fps - (0, 0)/1456x1088 crop]
```

I'd like the native resolution 1456x1088 to work, any help is appreciated :)

rafalmaciag, 2025-02-02:
I seems I see the issue:

```
 *  - Add GstVideoMeta support (strides and offsets)
```
Looks like something on a roadmap :/

naushir, 2025-02-03:
That's correct, gstreamer needs to know about alternate strides.  Can you expand on what you are using gstreamer for?  Perhaps you can achieve the same thing in python/Picamera2 until gstreamer has been updated to support this?

rafalmaciag, 2025-02-03:
I found a workaround with gstrawparse; it seems a bit more complicated than I had thought;

closed 2025-02-04 by naushir

fayezsalka, 2025-07-18:
> gstrawparse

Could you please share the gstreamer pipeline with the gstrawparse fix?