raspberrypi/libcamera #235: libcamerify does'nt work any more with libcamera v0.4.0 on my system.

LaurentChemla, 2025-02-03:
When I try to launch some v4l2 command with libcamerify, I get this error:

# /usr/local/bin/libcamerify v4l2-ctl --list-devices
[0:44:06.330196744] [4969]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
[0:44:06.343119541] [4970]  INFO RPI pisp.cpp:720 libpisp version v1.0.7 28196ed6edcf 03-02-2025 (07:28:51)
[0:44:06.441986454] [4970]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
v4l2-ctl: ../include/libcamera/controls.h:189: T libcamera::ControlValue::get() const [with T = long int; typename std::enable_if<((! libcamera::details::is_span<U>::value) && (! std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value)), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == details::control_type<std::remove_cv_t<T>>::value' failed.
Aborted

The same command work flawlessly with v0.3.2.

Thanks for all your work, guys.

L.

naushir, 2025-02-03:
@kbingham FYI

naushir, 2025-02-03:
Are you able to attach a debugger and provide a callback when this error occurs?

kbingham, 2025-02-04:
Also similarly reported from MotionProject : https://bugs.libcamera.org/show_bug.cgi?id=253

naushir, 2025-02-04:
Here's a backtrace of the error:

```
(gdb) bt
#0  __pthread_kill_implementation (threadid=549621485632, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at ./nptl/pthread_kill.c:44
#1  0x0000007ff7ad0aa4 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000007ff7a8a72c in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000007ff7a7747c in __GI_abort () at ./stdlib/abort.c:79
#4  0x0000007ff7a841a4 in __assert_fail_base (fmt=0x7ff7b9ef38 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n",
    assertion=assertion@entry=0x7ff7f813d8 "type_ == details::control_type<std::remove_cv_t<T>>::value",
    file=file@entry=0x7ff7f81413 "../include/libcamera/controls.h", line=line@entry=189,
    function=function@entry=0x7ff7f81433 "T libcamera::ControlValue::get() const [T = long]") at ./assert/assert.c:92
#5  0x0000007ff7a8420c in __GI___assert_fail (
    assertion=0x7ff7f813d8 "type_ == details::control_type<std::remove_cv_t<T>>::value",
    file=0x7ff7f81413 "../include/libcamera/controls.h", line=189,
    function=0x7ff7f81433 "T libcamera::ControlValue::get() const [T = long]") at ./assert/assert.c:101
#6  0x0000007ff7f71640 in libcamera::ControlValue::get<long, (decltype(nullptr))0>() const (this=0x7ff003a9e0)
    at ../include/libcamera/controls.h:189
#7  0x0000007ff7f6d09c in V4L2CameraProxy::setFmtFromConfig (this=0x5555624e40, streamConfig=...)
    at ../src/v4l2/v4l2_camera_proxy.cpp:203
#8  0x0000007ff7f6cf0c in V4L2CameraProxy::open (this=0x5555624e40, file=0x55556248f0)
    at ../src/v4l2/v4l2_camera_proxy.cpp:78
#9  0x0000007ff7f6bedc in V4L2CameraFile::V4L2CameraFile (this=0x55556248f0, dirfd=-100,
    path=0x555562c250 "/dev/video0", efd=3, nonBlocking=false, proxy=0x5555624e40)
    at ../src/v4l2/v4l2_camera_file.cpp:25
#10 0x0000007ff7f7e458 in std::_Construct<V4L2CameraFile, int&, char const*&, int&, int, V4L2CameraProxy*&> (
    __p=0x55556248f0, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/stl_construct.h:119
#11 0x0000007ff7f7e23c in std::allocator_traits<std::allocator<void> >::construct<V4L2CameraFile, int&, char const*&, int&, int, V4L2CameraProxy*&> (__p=0x55556248f0, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/alloc_traits.h:635
#12 0x0000007ff7f7df6c in std::_Sp_counted_ptr_inplace<V4L2CameraFile, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<int&, char const*&, int&, int, V4L2CameraProxy*&> (this=0x55556248e0, __a=...,
--Type <RET> for more, q to quit, c to continue without paging--c
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/shared_ptr_base.h:604
#13 0x0000007ff7f7dd70 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<V4L2CameraFile, std::allocator<void>, int&, char const*&, int&, int, V4L2CameraProxy*&> (this=0x7fffffe338, __p=@0x7fffffe330: 0x0, __a=...,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/shared_ptr_base.h:972
#14 0x0000007ff7f7dc94 in std::__shared_ptr<V4L2CameraFile, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<void>, int&, char const*&, int&, int, V4L2CameraProxy*&> (this=0x7fffffe330, __tag=...,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/shared_ptr_base.h:1712
#15 0x0000007ff7f7dbe8 in std::shared_ptr<V4L2CameraFile>::shared_ptr<std::allocator<void>, int&, char const*&, int&, int, V4L2CameraProxy*&> (this=0x7fffffe330, __tag=..., __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/shared_ptr.h:464
#16 0x0000007ff7f7a698 in std::make_shared<V4L2CameraFile, int&, char const*&, int&, int, V4L2CameraProxy*&> (
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40,
    __args=@0x7fffffe340: 0x5555624e40, __args=@0x7fffffe340: 0x5555624e40)
    at /usr/bin/../lib/gcc/aarch64-linux-gnu/12/../../../../include/c++/12/bits/shared_ptr.h:1009
#17 0x0000007ff7f797c0 in V4L2CompatManager::openat (this=0x7ff7fb0df8 <V4L2CompatManager::instance()::instance>,
    dirfd=-100, path=0x555562c250 "/dev/video0", oflag=2, mode=0) at ../src/v4l2/v4l2_compat_manager.cpp:180
#18 0x0000007ff7f782c0 in open (path=0x555562c250 "/dev/video0", oflag=2) at ../src/v4l2/v4l2_compat.cpp:54
#19 0x00000055555634e4 in ?? ()
[... cut ...]
203                     const int64_t duration = it->second.def().get<int64_t>();
(gdb) p it
$1 = (const std::__detail::_Node_const_iterator<std::pair<libcamera::ControlId const* const, libcamera::ControlInfo>, false, false> &) @0x7fffffd7b0: {<std::__detail::_Node_iterator_base<std::pair<libcamera::ControlId const* const, libcamera::ControlInfo>, false>> = {_M_cur = 0x7ff003a9b0}, <No data fields>}
```

Looks to be pointing to https://github.com/raspberrypi/libcamera/blob/83cb81014528a66f81131a4067a6de29550e3667/src/v4l2/v4l2_camera_proxy.cpp#L203

kbingham, 2025-02-04:
I was also able to reproduce this here too : Some how it seems the FrameDurationLimit 'default' has a ControlTypeNone ?

naushir, 2025-02-04:
I think it might be complaining about FrameDurationLimit not having a default?

kbingham, 2025-02-04:
Yes, next question is - what changed - or did we never have a default ?

kbingham, 2025-02-04:
We should probably also make that code in the v4l2 layer a bit more safe by checking the default is actually given though! I suspect we've added an assertion to protect the control types, and that's now catching the fact that before we were essentially accessing undefined / unintialised memory for the default value.

kbingham, 2025-02-04:
I think there are some controls that don't 'have' a default so a default isn't mandatory...
Probably something that should be validated by lc-compliance on a pipeline handler though to make sure implementations are correct.

ekstremedia, 2025-02-07:
Im getting the exact same error as [LaurentChemla](https://github.com/LaurentChemla), when trying to run libcamerify motion after i just did an apt upgrade on my pi5 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1. Is there any workaround i can do for now just to get it up and running again?

voidmain99, 2025-02-08:
> Im getting the exact same error as [LaurentChemla](https://github.com/LaurentChemla), when trying to run libcamerify motion after i just did an apt upgrade on my pi5 6.6.74+rpt-rpi-2712 [...]

I updated all my cameras the other day and they all stopped. I downgraded a few of the packages to get them working again. I should have tested on one before upgrading all of them. Here's a script I created to downgrade them:

`#!/bin/bash

mkdir -p /tmp/fm
cd /tmp/fm

ARCH="`dpkg --print-architecture`"

wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-tools_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-ipa_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-v4l2_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/python3-libcamera_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp-common_1.0.7-1_all.deb
wget http://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp1_1.0.7-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/r/rpicam-apps/rpicam-apps_1.5.3-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/p/picamera2/python3-picamera2_0.3.23-1_all.deb

echo "Press Enter to Continue"
read
ls *deb
echo "Press Enter to install above files"
read
apt install ./*deb
echo "Press Enter to restart motion"
read
sudo service motion restart
sleep 5
sudo service motion status
echo "apt-mark hold libcamera-ipa libcamera-tools libcamera-v4l2 libpisp-common libpisp1 python3-libcamera python3-picamera2 rpicam-apps"
echo "Press Enter to put on above upgrade hold"
read
apt-mark hold libcamera-ipa libcamera-tools libcamera-v4l2 libpisp-common libpisp1 python3-libcamera python3-picamera2 rpicam-apps
`

Matthew1471, 2025-02-08:
> Im getting the exact same error as [LaurentChemla](https://github.com/LaurentChemla), when trying to run libcamerify motion after i just did an apt upgrade on my pi5 6.6.74+rpt-rpi-2712 [...]

I listed my steps here https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2628977611:

```
apt-get purge libpisp-common
wget https://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp-common_1.0.7-1_all.deb
wget https://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp1_1.0.7-1_arm64.deb
wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-ipa_0.3.2+rpt20241119-1_arm64.deb
wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera0.3_0.3.2+rpt20241119-1_arm64.deb
wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-tools_0.3.2+rpt20241119-1_arm64.deb
wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-v4l2_0.3.2+rpt20241119-1_arm64.deb
dpkg -i ./libpisp-common_1.0.7-1_all.deb ./libpisp1_1.0.7-1_arm64.deb ./libcamera-ipa_0.3.2+rpt20241119-1_arm64.deb ./libcamera0.3_0.3.2+rpt20241119-1_arm64.deb ./libcamera-tools_0.3.2+rpt20241119-1_arm64.deb ./libcamera-v4l2_0.3.2+rpt20241119-1_arm64.deb
apt-mark hold libpisp-common
```

It'll put a hold on a shared library that 0.4 requires so won't auto-update again (you'll want to remove the hold when the issue gets fixed)

ekstremedia, 2025-02-08:
> > Im getting the exact same error as [LaurentChemla](https://github.com/LaurentChemla), when trying to run libcamerify motion after i just did an apt upgrade on my pi5 6.6.74+rpt-rpi-2712 [#1](https://github.com/raspberrypi/libcamera/issues/1) SMP PREEMPT Debian 1:6.6.74-1+rpt1. Is there any workaround i can do for now just to get it up and running again?
> 
> I listed my steps here [motioneye-project/motioneye#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2628977611) (it assumes you're using a 64-bit Raspberry Pi - other person's script above is longer but more comprehensive):
> 
> ```
> apt-get purge libpisp-common
> wget https://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp-common_1.0.7-1_all.deb
> wget https://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp1_1.0.7-1_arm64.deb
> wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-ipa_0.3.2+rpt20241119-1_arm64.deb
> wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera0.3_0.3.2+rpt20241119-1_arm64.deb
> wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-tools_0.3.2+rpt20241119-1_arm64.deb
> wget https://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-v4l2_0.3.2+rpt20241119-1_arm64.deb
> dpkg -i ./libpisp-common_1.0.7-1_all.deb ./libpisp1_1.0.7-1_arm64.deb ./libcamera-ipa_0.3.2+rpt20241119-1_arm64.deb ./libcamera0.3_0.3.2+rpt20241119-1_arm64.deb ./libcamera-tools_0.3.2+rpt20241119-1_arm64.deb ./libcamera-v4l2_0.3.2+rpt20241119-1_arm64.deb
> apt-mark hold libpisp-common
> ```
> 
> It'll put a hold on updating a dependency that 0.4 requires so it won't auto-update again (you'll want to remove the hold when the issue gets fixed)

jph76, 2025-02-09:
Thanks, @voidmain99 , for sharing your workaround with us!

In case you're running Raspberry Pi OS Lite like me, one needs to replace `rpicam-apps` with `rpicam-apps-lite`:

```
#!/bin/bash

mkdir -p /tmp/fm
cd /tmp/fm

ARCH="`dpkg --print-architecture`"

wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-tools_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-ipa_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/libcamera-v4l2_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libc/libcamera/python3-libcamera_0.3.2+rpt20241119-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp-common_1.0.7-1_all.deb
wget http://archive.raspberrypi.com/debian/pool/main/libp/libpisp/libpisp1_1.0.7-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/r/rpicam-apps/rpicam-apps-lite_1.5.3-1_${ARCH}.deb
wget http://archive.raspberrypi.com/debian/pool/main/p/picamera2/python3-picamera2_0.3.23-1_all.deb

echo "Press Enter to Continue"
read
ls *deb
echo "Press Enter to install above files"
read
sudo apt install ./*deb
echo "Press Enter to restart motion"
read
sudo service motion restart
sleep 5
sudo service motion status
echo "apt-mark hold libcamera-ipa libcamera-tools libcamera-v4l2 libpisp-common libpisp1 python3-libcamera python3-picamera2 rpicam-apps-lite"
echo "Press Enter to put on above upgrade hold"
read
sudo apt-mark hold libcamera-ipa libcamera-tools libcamera-v4l2 libpisp-common libpisp1 python3-libcamera python3-picamera2 rpicam-apps-lite
```

naushir, 2025-02-10:
We have patched the upstream libcamera tree with a fix, and a new apt package is now available.  Hopefully this should fix this bug.

closed 2025-02-10 by naushir

Mr-Dave, 2025-02-11:
FYI:  I have updated to the latest versions today and the problem persists.  The following is a snippet of the update log indicating the packages that were just installed.  
```
---snip---
Do you want to continue? [Y/n] y 
Get:1 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera-dev armhf 0.4.0+rpt20250210-1 [37.1 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera-ipa armhf 0.4.0+rpt20250210-1 [838 kB]
Get:2 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf libtasn1-doc all 4.19.0-2+deb12u1 [328 kB]
Get:6 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera0.4 armhf 0.4.0+rpt20250210-1 [651 kB]
Get:7 http://archive.raspberrypi.com/debian bookworm/main armhf rpicam-apps-lite armhf 1.6.0-2 [475 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera-apps-lite all 1.6.0-2 [3,584 B]
Get:9 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera-tools armhf 0.4.0+rpt20250210-1 [315 kB]
Get:10 http://archive.raspberrypi.com/debian bookworm/main armhf libcamera-v4l2 armhf 0.4.0+rpt20250210-1 [53.1 kB]
Get:11 http://archive.raspberrypi.com/debian bookworm/main armhf raspi-config all 20250210 [36.8 kB]
Get:4 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf libtasn1-6-dev armhf 4.19.0-2+deb12u1 [89.7 kB]
Get:5 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf libtasn1-6 armhf 4.19.0-2+deb12u1 [42.6 kB]
Fetched 2,869 kB in 3s (1,130 kB/s)  
---snip---
```
libcamerify usage output
```
---snip---
motion: ../include/libcamera/controls.h:189: T libcamera::ControlValue::get() const [with T = long long int; typename std::enable_if<((! libcamera::details::is_span<U>::value) && (! std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value)), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == details::control_type<std::remove_cv_t<T>>::value' failed.
Aborted
---snip---
```

degola, 2025-02-12:
Can confirm @Mr-Dave report, I also updated to 0.4.0+rpt20250210-1 and same issue persist:
```
# libcamerify v4l2-ctl --all
[0:01:35.478012890] [583] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:01:35.478169665] [583] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:01:35.478287534] [583]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+51-ca36c77f
[0:01:35.570958185] [586]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:01:35.578066837] [586]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media0
v4l2-ctl: ../include/libcamera/controls.h:189: T libcamera::ControlValue::get() const [with T = long int; typename std::enable_if<((! libcamera::details::is_span<U>::value) && (! std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value)), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == details::control_type<std::remove_cv_t<T>>::value' failed.
Aborted
```

naushir, 2025-02-12:
Unfortunately, I missed an issue with my fix.  The packages have now been updated (I think!) and if you update again, it should now work correctly.

Camilochiang, 2025-02-12:
> Unfortunately, I missed an issue with my fix. The packages have now been updated (I think!) and if you update again, it should now work correctly.

I just give it a try
```
sudo apt purge libcamera*
sudo apt install libcamera0.4 libcamera-dev libcamera-tools libcamera-ipa libcamera-v4l2 python3-libcamera libpisp-common libpisp1 rpicam-apps-lite python3-picamera2 
```

 but the problem persist
```
(AGS_311_venv) @AGS-RPI-0004:~$ uname -a
Linux AGS-RPI-0004 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

(AGS_311_venv) @AGS-RPI-0004:~$ libcamera-hello
[0:01:23.242753930] [811] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:01:23.242802797] [811] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:01:23.242827435] [811]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+51-ca36c77f
Preview window unavailable
ERROR: *** no cameras available ***

(AGS_311_venv) @AGS-RPI-0004:~$ apt list -a libcam* -a i | grep "installed"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libcamera-dev/stable,now 0.4.0+rpt20250210-1 arm64 [installed]
libcamera-ipa/stable,now 0.4.0+rpt20250210-1 arm64 [installed]
libcamera-tools/stable,now 0.4.0+rpt20250210-1 arm64 [installed]
libcamera-v4l2/stable,now 0.4.0+rpt20250210-1 arm64 [installed]
libcamera0.4/stable,now 0.4.0+rpt20250210-1 arm64 [installed]

(AGS_311_venv) @AGS-RPI-0004:~$ dpkg --get-selections | grep 'libcam'
libcamera-dev:arm64                             install
libcamera-ipa:arm64                             install
libcamera-tools                                 install
libcamera-v4l2:arm64                            install
libcamera0.4:arm64                              install
python3-libcamera:arm64                         install
(AGS_311_venv) @AGS-RPI-0004:~$

```
So not sure if you push it to the apt repository?

naushir, 2025-02-12:
Oops, it did not make it to the public apt repo.  Pushed now, try again in 5-10 minutes.

degola, 2025-02-12:
🙏 works perfectly now, thank you very much!

jph76, 2025-02-13:
Can also confirm it works again. Thank you!

Script to revert the workaround from https://github.com/raspberrypi/libcamera/issues/235#issuecomment-2644396318 (works with either `rpicam-apps` and `rpicam-apps-lite`):

```
#!/bin/bash
sudo apt-mark unhold libcamera-ipa libcamera-tools libcamera-v4l2 libpisp-common libpisp1 python3-libcamera python3-picamera2 rpicam-apps rpicam-apps-lite
sudo apt update
sudo apt upgrade
sudo systemctl restart motion
sleep 5
systemctl status --no-pager motion
```