raspberrypi/libcamera #236: Remove glare with libcamera

gasparramoa, 2025-02-10:
Hello all,

## Problem

I'm trying to reduce glare from LEDs using **exposure** and **analogue gain** with Libcamera, but so far, I haven't been able to eliminate it. I've tried to use low exposure and high gain, but even using that I still have "some" glare...

[image] (AnalogueGain=10.0, ExposureTime=1000)

[image] (AnalogueGain=1.0, ExposureTime=3000)

I'm unsure if this is even possible. Ideally, a camera parameter would limit the amount of light captured to prevent overexposure. However, after searching online, I haven’t found any solutions that directly address this through camera settings.

I've heard that a polarizing filter might help. Has anyone tried using one with a Raspberry Pi camera?
Is this the only way to mitigate glare, aside from repositioning the camera or adjusting external lighting?

Thanks in advance!

## System

- OS Bullseye LITE
- Arducam IMX219
- Rbpi 4

naushir, 2025-02-27:
Sorry for the late reply.  Using an ND filter might help here. Do you have one you can try out?

gasparramoa, 2025-03-03:
Thank you for the reply. I was able to solve my problem by using another time of material that does not glow (matte). Nevertheless, I will give it a try to an ND filter and publish the results here once I have that. I am closing since the problem is fixed for me.

closed 2025-03-03 by gasparramoa