raspberrypi/libcamera #237: undefined default control values

christianrauch, 2025-02-10:
Inspecting the default values for controls, I found some controls (`AeEnable`, `ExposureTime`, `AwbEnable`, amongst others) are missing default values (`<ValueType Error>`). It would make sense to have `ExposureTime` undefined if auto exposure is enabled by default, but having `AeEnable` undefined brings the camera in a weird state. The camera surely must either operate with a fixed exposure time or with auto exposure.

How does it come that some controls are undefined, and what is the meaning of this? How is someone using the libcamera API in this fork supposed to interpret these undefined states? E.g. how do I know if "auto exposure" and "auto white balance" is enabled or not?

naushir, 2025-02-11:
There's a bit of a discussion going on with the libcamera team about default values of controls.  I'm a bit unclear on how they should be set on initial startup without configuring the camera.  Also, like you mentioned, some controls don't have defaults depending on the mode of operation, so I'm also unsure about this.

I can leave this open until we work things out, but right now I don't expect things to change with this. 

> g.g. how do I know if "auto exposure" and "auto white balance" is enabled or not?

You can assume these are enabled on startup by default.  If you need to be explicit, I suggest you set AWB/AGC enable controls when starting up the camera in your application.

naushir, 2025-02-26:
This is now resolved.

closed 2025-02-26 by naushir